espressif/vscode-esp-idf-extension #1407: [Feature Request]: 需要让flasher_args可手动配置分区文件是否需要烧录

shishirong, 2025-01-21:
**Is your feature request related to a problem? Please describe.**
使用场景：我的工程分区表如下：
```csv
nvs,data,nvs,0x9000,16K,
otadata,data,ota,0xd000,8K,
ota_0,app,ota_0,0x10000,1M,
ota_1,app,ota_1,0x110000,1M,
storage,data,littlefs,0x210000,1984K,
```
调试过程中，会经常烧录应用程序，目前默认行为是会烧录所有分区文件，但是平时调试时只需要烧录ota_0 而其他分区需要保持不变，vscode-esp-idf-extension并没有找到相关配置可以控制这个默认烧录行为

**Describe the solution you'd like**
允许用户手动配置分区文件是否需要烧录

SinglWolf, 2025-01-30:
It would be great if the file build functions for the partition and the upload of this file into the partition looked like this (as in PlatformIO):

[image]

brianignacio5, 2025-02-13:
This feature is already available in the extension but unfortunately not documented.

1) In the Activity bar click the Espressif icon and expand the `Device Partition Explorer` section.

[image]

2) Click the Refresh Partition Table button that appear next to `Device Partition Explorer` title. Select the partition table offset to use (read from sdkconfig file or pass the value yourself like 0x8000)

[image]

3) The list of partitions will be read from the device (check that the serial port shown in the status bar is correct). When you click a partition you can now: read the current partition and save as bin in the current project or select a bin to flash in the current partition.

4) You can also right click a .bin file and select `Flash binary to partition...`. A dropdown will appear to select one of current device partitions or manually type offset to flash to.

[image]

SinglWolf, 2025-02-14:
> PTAL [@shishirong](https://github.com/shishirong) [@SinglWolf](https://github.com/SinglWolf) and let me know what you think.

The `Device Partition Explorer` feature is interesting, but it is too complicated to use and unsafe for practical use - all partitions are writable for any binary files. In addition, the `Device Partition Explorer` does not have the mode you described for downloading a partition to file. This would be useful for subtypes `coredump`, `nvs`, `fat`, `spiffs`, `littlefs` (`littlefs` is not supported in the `Partition Table Editor`). There is also no easy way to download only `bootloader.bin` to the device partition.

shishirong, 2025-02-14:
> This feature is already available in the extension but unfortunately not documented.

这个确实可以实现烧录部分分区文件的功能，只不过操作起来还是繁琐了一点，因为开发与调试过程中，通常都是直接使用 `Build, Flash and Monitor`:  [image]  来进行烧录的

brianignacio5, 2025-02-14:
> In addition, the Device Partition Explorer does not have the mode you described for downloading a partition to file. 
You are right I'm not sure why I thought it was there. I could add this feature is not very hard.

> There is also no easy way to download only bootloader.bin to the device partition.
You can right click `bootloader.bin` and select the offset but I understand because you would prefer to see these partitions listed in the Device Partition Explorer above. 

I've just noticed that `gen_esp32part.py` doesn't really return all partitions from current device.

So let me summarize the required changes, to my understanding, for the feature request implementation here:

1) Find a way to obtain all partitions from current serial port device.
2) Add read partition and save to file option
3) Add a command to build and another to flash specific partition

brianignacio5, 2025-02-21:
Hello guys  @shishirong @SinglWolf !

I've created a PR to fix issues we discussed in here #1436 

Please take a look.

bootloader and partition table information come from flasher_args.json and bootloader size from binaries in build directory.
Now we can read a partition from device and saved as bin in current project.
Also added commands to build and flash just `App`, `bootloader` and `partition table` listed as `ESP-IDF: Build App`, `ESP-IDF: Flash App only`, `ESP-IDF: Build Bootloader`, `ESP-IDF: Flash Bootloader Only`, `ESP-IDF: Build Partition Table`, `ESP-IDF: Flash Partition Table Only`

You can test these changes by installing this VSIX by click menu **View** -> **Command Palette...**, type **Install from VSIX** and then select downloaded **esp-idf-extension.vsix** file to install the extension.
* [esp-idf-extension.vsix.zip](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/2629570389.zip)

brianignacio5, 2025-02-21:
About the filesystem Image and OTA, I think is very similar to provide functionality with the previous comment.

It will depend that your current project does have a partition table with related partitions (ota, spiffs).

About building these filesystems please review this:

for FATFS: https://github.com/espressif/esp-idf/blob/c71d74e2f853b1135c63f47e349f2a08f63f3e01/examples/storage/fatfs/fatfsgen/main/CMakeLists.txt#L18
for SPIFFS: https://github.com/espressif/esp-idf/blob/c71d74e2f853b1135c63f47e349f2a08f63f3e01/examples/storage/spiffsgen/main/CMakeLists.txt#L9
for LittleFS: https://github.com/espressif/esp-idf/blob/c71d74e2f853b1135c63f47e349f2a08f63f3e01/examples/storage/littlefs/main/CMakeLists.txt#L7
for NVS: https://github.com/espressif/esp-idf/blob/c71d74e2f853b1135c63f47e349f2a08f63f3e01/examples/storage/nvsgen/main/CMakeLists.txt#L9

Since a filesystem existence in your project depends on configuration is hard to determine an script to trigger this from IDE without more information.

I try to discuss it with ESP-IDF Storage team to see if there sometime we can about it.

SinglWolf, 2025-02-21:
> You can test these changes by installing this VSIX by click menu **View** -> **Command Palette...**, type **Install from VSIX** and then select downloaded **esp-idf-extension.vsix** file to install the extension.

Unfortunately, I don't have a spare computer to check. I'll wait for you to approve my PR https://github.com/espressif/vscode-esp-idf-extension/pull/1409.

shishirong, 2025-03-09:
> > This feature is already available in the extension but unfortunately not documented.
> 
> 这个确实可以实现烧录部分分区文件的功能，只不过操作起来还是繁琐了一点，因为开发与调试过程中，通常都是直接使用 `Build, Flash and Monitor`: [image] 来进行烧录的

@brianignacio5 可以考虑一下这个提议吗？

brianignacio5, 2025-03-10:
Sorry @shishirong I don't understand what you want to achieve. build flash monitor function is already there. What is missing ?

shishirong, 2025-03-10:
@brianignacio5 "Build Flash Monitor " default is flash all partition , but Just want to  "Build + flash<app partition> + Monitor" at development usually.  Although the `ESP-IDF: Flash App only` command has been added now, it is still not as convenient to use as `Build Flash Monitor`.

brianignacio5, 2025-03-10:
So you would like to have something like Build <app partition> + Flash <app partition> + Monitor ? or just Build  + Flash<App partition> + Monitor

shishirong, 2025-03-10:
yes,  add option for config build and flash partition  in some args.json file (flasher_args.json will change after every build). if add a gui page similarly  "ESP-IDF Partition Table Editor" for edit It couldn't be better.

shishirong, 2025-03-26:
@brianignacio5 the feature is not in plan?

brianignacio5, 2025-03-26:
Yes it is planned. we just need to think how this would be implemented.

brianignacio5, 2025-04-17:
I've just added the `idf.flashPartitionToUse` to allow you to define which partition is flashed in Build flash monitor and flash command (only UART). This allows you to run build flash monitor to flash only single partition from app bootloader or partition table.

Please take a look @shishirong with this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/2962384773.zip)

PR #1436 (2025-04-21): [VSC-1572] add app bootloader partition table build flash cmds
## Description

Add bootloader and partition table to Device Partition Explorer. Use current project files and minimize reading from device. 
Using `<buildDirPath>/flasher_args.json` to get bootloader and partition table offset. 
Move Open Partition Table editor action to Device partition Explorer as icon. (Next to refresh button)

Add read partition action in Device Partition Explorer.

Add `ESP-IDF: Build App`, `ESP-IDF: Flash App only`, `ESP-IDF: Build Bootloader`, `ESP-IDF: Flash Bootloader Only`, `ESP-IDF: Build Partition Tabl [... cut]
closed 2025-04-17 by shishirong