espressif/vscode-esp-idf-extension #1410: extension does not install/configure correctly

melbnz, 2025-01-23:
### OS

Windows

### Operating System version

windows 11 (arm processor) 23H2 build 22631.4751

### Visual Studio Code version

1.96.4

### ESP-IDF version

1.9.0

### Python version

2024.22.2 (microsoft VSC)

### Doctor command output

There is no data provider registered that can provide view data.

### Extension

no log
not installed

### Description

commands in extension are blank and inoperable

[image]

### Debug Message

commands are blank and inoperable

### Other Steps to Reproduce

uninstalled re-installed
Other extensions install and run fine
I followed all instruction in prerequisites and install

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

melbnz, 2025-01-28:
Thanks for taking the time to engage, when i run command palette i do not see an explicit global setting.  if i open local user settings there's just a legacy ssh link

[image]

i did follow rest of yoru process (as i've tried before) with no change and no Commands showing post re-installation.

I am running on ARM processor, but this hasn't been an issue so far.

[image]

melbnz, 2025-01-29:
we'll start with logs as there are errors on install...
2025-01-30 10:23:14.530 [error] Activating extension espressif.esp-idf-extension failed due to an error:
2025-01-30 10:23:14.530 [error] Error: No native build was found for platform=win32 arch=arm64 runtime=electron abi=128 uv=1 armv=8 libc=glibc node=20.18.1 electron=32.2.6 webpack=true
    loaded from: c:\Users\melbu\.vscode\extensions\espressif.esp-idf-extension-1.9.0

	at g.resolve.g.path (c:\Users\melbu\.vscode\extensions\espressif.esp-idf-extension-1.9.0\dist\extension.js:2:867672)
	at g (c:\Users\melbu\.vscode\extensions\espressif.esp-idf-extension-1.9.0\dist\extension.js:2:865767)
	at 6237 (c:\Users\melbu\.vscode\extensions\espressif.esp-idf-extension-1.9.0\dist\extension.js:2:14548)
	at __webpack_require__ (c:\Users\melbu\.vscode\extensions\espressif.esp-idf-extension-1.9.0\dist\extension.js:2:1369211)
	at 6269 (c:\Users\melbu\.vscode\extensions\espressif.esp-idf-extension-1.9.0\dist\extension.js:2:5793)
	at __webpack_require__ (c:\Users\melbu\.vscode\extensions\espressif.esp-idf-extension-1.9.0\dist\extension.js:2:1369211)
	at 176 (c:\Users\melbu\.vscode\extensions\espressif.esp-idf-extension-1.9.0\dist\extension.js:2:9378)
	at __webpack_require__ (c:\Users\melbu\.vscode\extensions\espressif.esp-idf-extension-1.9.0\dist\extension.js:2:1369211)
	at 9076 (c:\Users\melbu\.vscode\extensions\espressif.esp-idf-extension-1.9.0\dist\extension.js:2:835587)
	at __webpack_require__ (c:\Users\melbu\.vscode\extensions\espressif.esp-idf-extension-1.9.0\dist\extension.js:2:1369211)
	at 4987 (c:\Users\melbu\.vscode\extensions\espressif.esp-idf-extension-1.9.0\dist\extension.js:2:835159)
	at __webpack_require__ (c:\Users\melbu\.vscode\extensions\espressif.esp-idf-extension-1.9.0\dist\extension.js:2:1369211)
	at c:\Users\melbu\.vscode\extensions\espressif.esp-idf-extension-1.9.0\dist\extension.js:2:1567348
	at c:\Users\melbu\.vscode\extensions\espressif.esp-idf-extension-1.9.0\dist\extension.js:2:1921790
	at Object.<anonymous> (c:\Users\melbu\.vscode\extensions\espressif.esp-idf-extension-1.9.0\dist\extension.js:2:1921829)
	at Module._compile (node:internal/modules/cjs/loader:1484:14)
	at Module._extensions..js (node:internal/modules/cjs/loader:1564:10)
	at Module.load (node:internal/modules/cjs/loader:1295:32)
	at Module._load (node:internal/modules/cjs/loader:1111:12)
	at c._load (node:electron/js2c/node_init:2:16955)
	at e._load (file:///c:/Users/melbu/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:175:10896)
	at t._load (file:///c:/Users/melbu/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:167:22534)
	at i._load (file:///c:/Users/melbu/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:159:21138)
	at Module.require (node:internal/modules/cjs/loader:1318:19)
	at require (node:internal/modules/helpers:179:18)
	at zz.xb (file:///c:/Users/melbu/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:167:23854)
	at async Promise.all (index 0)

wpalm, 2025-02-11:
> we'll start with logs as there are errors on install... 2025-01-30 10:23:14.530 [error] Activating extension espressif.esp-idf-extension failed due to an error: 2025-01-30 10:23:14.530 [error] [...]

wpalm, 2025-02-11:
I'm having the same problem, I get the same error message, on a Surface Pro 11. I guess there is no support for ARM64. A workaround is to install the x64 variant of VSCode.

melbnz, 2025-02-11:
@wpalm did you try the x64 version?  Solve the issue?  I’m getting so frustrated with arm incompatibility on various apps even with new prism 24h2 emulator.

wpalm, 2025-02-12:
> @wpalm did you try the x64 version?  Solve the issue?  I’m getting so frustrated with arm incompatibility on various apps even with new prism 24h2 emulator. 

Yes, it works great!

brianignacio5, 2025-02-13:
The thing is that we use node serial port for debugging. They haven't release Windows arm64 yet. See https://github.com/serialport/node-serialport/issues/2811

JHubi1, 2025-03-01:
This is still an issue

ronbreukers, 2025-03-10:
I have exacly the same issue with windows 11 ARM on a HP Omnibook 14. I use the X86 version of VS Code, but it gives every 10 minutes or so an annoing popup that the version is not for my windows. For all the rest this X86 version works fine.

lsdlsd88, 2025-03-19:
still happening.

pilotto8, 2025-05-14:
For now still an issue

PR #1585 (2025-06-11): update serialport latest for win arm support
## Description

Update serial port to 13.0.0 to support Windows Arm 64 prebuild binding/cpp

Fixes #1410

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Try to get serial ports or start a debug session.

- Expected behaviour:

Getting serial ports and debug should work on Windows ARM

- Expected output:

## How has this been tested?

Try to get serial ports or start a debug session.

**Test Configuration**:
* ESP-IDF Version:
* OS (Windows,Linu [... cut]
closed 2025-06-11 by brianignacio5