espressif/vscode-esp-idf-extension #1414: Automatically Detect Project Files to Activate ESP-IDF Plugin

Xinyu-Sheng, 2025-01-29:
When developing projects based on ESP-IDF using VS Code, it is currently necessary to manually activate the ESP-IDF plugin each time to load the corresponding environment and tools, which is not very convenient. In contrast, PlatformIO can automatically activate its plugin by detecting specific ini files, greatly simplifying the development process. I hope that the ESP-IDF plugin could offer similar functionality, where it would automatically activate upon detecting certain project files (such as sdkconfig or other ESP-IDF-related configuration files) in the current directory, thereby enhancing work efficiency.

I would like to see an implementation where, when a user opens a folder containing an ESP-IDF project, VS Code can automatically recognize the project and activate the ESP-IDF plugin without manual intervention. Specifically, this could be triggered by monitoring the presence of specific files (e.g., sdkconfig, particular content within CMakeLists.txt, etc.) as conditions for activating the plugin.

Currently, one alternative is to manually activate the ESP-IDF plugin through the command palette every time, which is not ideal for developers frequently switching between projects. Another possibility is writing custom scripts to check for these files and activate the plugin, but this adds extra learning and maintenance costs.

Providing such functionality would make the development experience smoother, especially important for those managing multiple different types of projects (including non-ESP-IDF projects) simultaneously. It is hoped that this improvement will be seen in future versions. Additionally, if there are suggestions or discussions on how to better integrate this feature, I am also very willing to participate and contribute.

Xinyu-Sheng, 2025-01-31:
@radurentea 
I would like to have a stricter detection mechanism because my other projects may currently activate the ESP-IDF extension as they meet the existing rules. To avoid this,could we implement more stringent activation conditions? For example, requiring the presence of multiple specific files unique to ESP-IDF projects, or introducing an additional project-specific identifier that must be present in the workspace for the extension to activate. This would help ensure that the ESP-IDF extension only activates for intended ESP-IDF projects and not for others inadvertently.

Ishan-Sharma-17, 2025-06-03:
Hello, the change works perfectly now and I no longer see the extension active when opening an STM32 project.

Thank you!