espressif/vscode-esp-idf-extension #1415: [Feature Request]: RTOS thread view to show thread names

karlp, 2025-01-29:
**Is your feature request related to a problem? Please describe.**
When using GDB and OpenOCD from the command line, `info threads` shows me thread names as well as thread ids.
When using VSCode, the Callstack window only shows thread ids, and I have to oepn them and browse the callstacks to select the correct thread.

**Describe the solution you'd like**
Would be nice to see the provided thread names :)

**Describe alternatives you've considered**
living with it? :)

**Additional context**
VS Code extension version 1.9.0 plus the fix in https://github.com/espressif/vscode-esp-idf-extension/pull/1413)
Screenshot of VSCode:

[image]

Vs screenshot from console GDB.  Same openocd behind the scenes: v0.12.0-esp32-20230921

[image]

Vs screenshot from a non-esp32 embedded freertos project in vscode:

[image]

bryghtlabs-richard, 2025-01-30:
I'm using 1.9.0, with a project build with ESP-IDF 5.1.2, and thread names do work for me:

[image]

brianignacio5, 2025-02-07:
The question is why launch configuration are you using ? Are you using our fork of Eclipse CDT debug adapter or another debugger ?

karlp, 2025-02-07:
Cannot reproduce anymore :|  my computer restarted, vscode got autoupdated to 1.97.  and restarted. Works like any other project now, and like yours. ==> closing.  Thanks for looking at it!

closed 2025-02-07 by karlp