espressif/vscode-esp-idf-extension #1416: [Feature Request]: Automatic platform detection.

SinglWolf, 2025-01-30:
Greetings!
There are a lot of similar lines in the `vscode-esp-idf-extension` settings that confuse beginners and greatly annoy more experienced users (e.g., me)
[image]
[image]
[image]

Using the example of `Python`, I propose a solution to the problem with different delimiters in the path, depending on the platform on which `vscode-esp-idf-extension` is running.
``` python
from sys import platform

if platform == "linux" or platform == "darwin":
    # linux or OS X
    SLASH = '/'
elif platform == "win32":
    # Windows...
    SLASH = '\\'
```
This will reduce the number of items  `settings` and reduce the entry threshold in the `vscode-esp-idf-extension`.

SinglWolf, 2025-02-03:
> * Only show relevant platform-specific settings in the VS Code settings UI and automatically handle path delimiter conversion internally (/ vs ) without requiring separate settings for each OS

An acceptable option

> * Or to have different descriptions, so that the user would understand what's the difference between **"idf: Build Path"** and **"idf: Build Path Win"** for example?

This only leads novice users into a stupor from the huge number of almost identical lines. And others will probably wonder why, if everything can be done automatically, without user intervention.

brianignacio5, 2026-09-17:
This is implemented with https://github.com/espressif/vscode-esp-idf-extension/pull/1868

closed 2026-09-17 by brianignacio5