espressif/vscode-esp-idf-extension #1418: [Setup] Unable to write to User Settings because idf.pythonInstallPath is not a registered configuration

bryghtlabs-richard, 2025-01-30:
### OS

Windows

### Operating System version

Windows 11

### Visual Studio Code version

1.96.4

### ESP-IDF version

5.1.2

### Python version

idf-python 3.11.2

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS undefined undefined undefined 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 undefined 
System environment variable PYTHON 
 undefined 
Visual Studio Code version undefined 
Visual Studio Code language undefined 
Visual Studio Code shell undefined 
ESP-IDF Extension version undefined 
Workspace folder No workspace folder is open 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) undefined
ESP-IDF Path (idf.espIdfPath) undefined
ESP-MDF Path (idf.espMdfPath) undefined
ESP-Matter Path (idf.espMatterPath) undefined
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) undefined
Custom extra paths undefined
System python Path (idf.pythonInstallPath) undefined
Virtual environment Python path (computed) undefined
Serial port (idf.port) undefined
OpenOCD Configs (idf.openOcdConfigs) undefined
ESP-IDF Tools Path (idf.toolsPath) undefined
Git Path (idf.gitPath) undefined
Notification Mode (idf.notificationMode) undefined
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) undefined
Access to ESP-IDF Path (idf.espIdfPath) undefined
Access to ESP-MDF Path (idf.espMdfPath) undefined
Access to ESP-Matter Path (idf.espMatterPath) undefined
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) undefined
Access to ESP-IDF Custom extra paths
Access to System python Path (idf.pythonInstallPath) undefined
Access to Virtual environment Python path (computed) undefined
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) undefined
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path undefined
Spaces in ESP-ADF Path (idf.espAdfPath) undefined
Spaces in ESP-IDF Path (idf.espIdfPath) undefined
Spaces in ESP-MDF Path (idf.espMdfPath) undefined
Spaces in ESP-Matter Path (idf.espMatterPath) undefined
Spaces in ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) undefined
Spaces in ESP-IDF Custom extra paths
Spaces in System python Path (idf.pythonInstallPath) undefined
[... 70 lines cut ...]
{"user":true,"message":"Unable to write to User Settings because idf.pythonInstallPath is not a registered configuration.","stack":"CodeExpectedError: Unable to write to User Settings because idf.pythonInstallPath is not a registered configuration.\n    at Aot.y (vscode-file://vscode-app/c:/Users/RichardAllen/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/workbench.desktop.main.js:3108:6643)\n    at Aot.E (vscode-file://vscode-app/c:/Users/RichardAllen/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/workbench.desktop.main.js:3108:8930)\n    at Aot.n (vscode-file://vscode-app/c:/Users/RichardAllen/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/workbench.desktop.main.js:3108:3863)\n    at Object.factory (vscode-file://vscode-app/c:/Users/RichardAllen/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/workbench.desktop.main.js:3108:3763)\n    at ww.j (vscode-file://vscode-app/c:/Users/RichardAllen/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/workbench.desktop.main.js:31:74345)\n    at vscode-file://vscode-app/c:/Users/RichardAllen/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/workbench.desktop.main.js:31:74257\n    at new Promise (<anonymous>)\n    at ww.queue (vscode-file://vscode-app/c:/Users/RichardAllen/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/workbench.desktop.main.js:31:74199)\n    at Aot.writeConfiguration (vscode-file://vscode-app/c:/Users/RichardAllen/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/workbench.desktop.main.js:3108:3732)\n    at nOs.Hb (vscode-file://vscode-app/c:/Users/RichardAllen/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/workbench.desktop.main.js:3108:42111)","level":"error","timestamp":"2025-01-30T16:13:34.596Z"}
{"message":"The \"file\" argument must be of type string. Received undefined","stack":"TypeError: The \"file\" argument must be of type string. Received undefined\n\tat normalizeSpawnArguments (node:child_process:549:3)\n\tat spawn (node:child_process:789:13)\n\tat Object.execFile (node:child_process:352:17)\n\tat t.<computed> [as execFile] (node:electron/js2c/node_init:2:2955)\n\tat \\dist\\extension.js:2:1435460\n\tat new Promise (<anonymous>)\n\tat Je (\\dist\\extension.js:2:1435437)\n\tat Ee (\\dist\\extension.js:2:1422366)\n\tat \\dist\\extension.js:2:1665553\n\tat async Ca (\\dist\\extension.js:2:1665152)\n\tat async \\dist\\extension.js:2:1890460","level":"error","timestamp":"2025-01-30T16:23:24.260Z"}
{"user":true,"level":"warn","message":"Extension configuration report has been copied to clipboard with errors","timestamp":"2025-01-30T16:23:24.260Z"}

### Description

I'm trying to setup VSCode with ESP-IDF Extension. We forked ESP-IDF 5.1.2 to fix a few bugs in C, so I would like to use my existing copy of ESP-IDF. However, in ESP-IDF Setup, which I click "Install", it fails with a popup: "Unable to write to User Settings because idf.pythonInstallPath is not a registered configuration.", like this:

[image]

Is there a way to set idf.pythonInstallPath manually? Or any other step I should do first?

I tried "ESP-IDF: Install ESP-IDF Extension Python Packages", but that fails with "Open a folder first" popup.

### Debug Message

I don't think I'm getting this far in installation.

### Other Steps to Reproduce

I believe I installed ESP-IDF 5.1.2. about a year ago.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

bryghtlabs-richard, 2025-01-30:
After closing and reopening VSCode, I was able to re-run ESP-IDF plugin configuration with my existing ESP-IDF 5.1.2, and this time it installed some python packages, and appears to have worked.

brianignacio5, 2025-02-05:
It is a weird error because the the `idf.pythonInstallPath` is an extension setting that is used by extension itself. I'm not sure how to reproduce the issue itself.

We will be removing the setup wizard and this setting in favor of our new [ESP-IDF Install Manager (EIM)](https://github.com/espressif/idf-im-ui) anyway.

closed 2025-02-12 by brianignacio5