espressif/vscode-esp-idf-extension #1420: [Feature Request]: ability to resolve structure global structure members on mouse-hover

bryghtlabs-richard, 2025-01-30:
**Is your feature request related to a problem? Please describe.**
Yes - when I hover the mouse over a structure's member, it shows me the type(which is nice), but often I actually want to see the data inside the structure.

Example structure definition in header:

```
typedef struct {
  uint8_t AIFrom;
  uint8_t AITo;
}moveVar;
```

Example usage site:
```
extern moveVar moveStatus;

void func(){
  clearChessLEDs();
  convertSendLEDData();
  sequenceAILights(moveStatus.AIFrom, moveStatus.AITo);
}
```

Hovering on moveStatus.AIFrom shows:

[image]

**Describe the solution you'd like**
I'd like ESP-IDF VSCode to display the structure member variable's value instead, or in addition to, the type. Ideally, the same way that ESP-IDF VSCode displays structure member variables values when allocated on the stack.

**Describe alternatives you've considered**
I can add each global variable to the watch window, but it's quite time consuming.

brianignacio5, 2025-02-05:
Unfortunately this a Language extension feature and our extension doesn't implement this yet. In our documentation we suggest users either Microsoft C/C++ extension or Clang extension for language support. It might be best to post a feature request in their respective GitHub repositories.

bryghtlabs-richard, 2025-02-13:
Thanks @brianignacio5 , I've mentioned it in the Microsoft C/C++ extension repo. We can close this if preferred.

brianignacio5, 2025-02-14:
Some follow up questions about your use case @bryghtlabs-richard 

1) Which debug configuration are you using ? I mean how does `launch.json` looks like?

2) Does this happens only when you are debugging or also when you just hovering on the source code file ?

I just want to know since a user replied to you in the cpptools repository that is related to debugging, but it is my understanding that hover definitions are handled by the language extension so I'm not sure how is related to debugging.

bryghtlabs-richard, 2025-03-24:
ping

bryghtlabs-richard, 2025-04-16:
Here are a few examples of hovering while debugging and only getting the type info rather than the value:

[image]

[image]

But sometimes it works:

[image]

brianignacio5, 2025-04-24:
Hi @bryghtlabs-richard 

I've added a bugfix in https://github.com/espressif/vscode-esp-idf-extension/pull/1521

Please take a look testing the vsix from the PR or I add the link here:

You can test these changes by installing this VSIX by click menu **View** -> **Command Palette...**, type **Install from VSIX** and then select downloaded **esp-idf-extension.vsix** file to install the extension.
* [esp-idf-extension.vsix.zip](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/2993376447.zip)

bryghtlabs-richard, 2025-04-24:
I think it installed, but it didn't work.

So I uninstalled ESP-IDF VSCode plugin, then reinstalled from VSIX.

But now OpenOCD reports:`❌ Error: Debug Adapter has to be specified, see "adapter driver" command`, but doctor says `OpenOCD Configs (idf.openOcdConfigs) board/esp32s3-builtin.cfg` which includes `source [find interface/esp_usb_jtag.cfg]` which sets `adapter driver esp_usb_jtag` correctly, so I'm not sure what's going on.

I tried uninstalling the VSIX and switching back to release version but the issue pertains. I think I need to reconfigure the OCD plugin but I can't seem to find how.

brianignacio5, 2025-04-25:
I don't think this issue is related to the vsix or extension. Consider checking [openOCD troubleshooting FAQ](https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ) or reach out in https://github.com/espressif/openocd-esp32 which they could help better that me regarding openOCD.

bryghtlabs-richard, 2025-04-25:
>I don't think this issue is related to the vsix or extension. 

Agreed. I just mean, I'm sorry I can't test for you for now.

bryghtlabs-richard, 2025-05-01:
Somehow my "idf.openOcdConfigs" in settings.json had changed. I restored it "board/esp32s3-builtin.cfg" and can debug again.

PR #1521 (2025-05-09): [VSC-1654 | 1584 | 1583] add evaluate on hover, view vars as hex, set data breakpoints
## Description

1. Add the ability to show variables values on hover (it will run a evaluateRequest to the debug adapter)
2. Add a right click option to show the integer value parsed as Hex and binary. Values will be  show in `ESP-IDF: Hex View` in debug tab. Variables values are updated during debug session if it changes.
3. Add Data breakpoints request. Just right click a variable and select `Break on Value Read`, `Break on Value Read` or `Break on Value Read`.

[image]

Fixes #1511 
Fixes #1420
Fixes #141 [... cut]
closed 2025-05-09 by brianignacio5