espressif/vscode-esp-idf-extension #1421: [Feature Request]: Options in the esp_idf_project_configuration.json file

SinglWolf, 2025-02-03:
Please shorten the file name. For example, `project_config.json`.
Add a section named `common` or `default` for the entire project.
Do not generate strings with empty parameters. This clutters up the showing, especially relevant in a multiproject.
Add the most frequently used options, for example, in my multiproject I use different `partitions.csv` files, and the sdkconfig file is the same for all configurations.
``` json
{
"partitionFilePath": "${workspaceFolder}/8M.scv"
}
```
Add the path to the directory for creating the binary file for the upload to partition, e.g.
``` json
{
"imageFolderPath": "${workspaceFolder}/sd-card"
}
```
Add the option to select the file system to create the image, e.g.
``` json
{
"imageFS": "ffat"
}
```
Add the `no-reset` flag for the monitor, e.g.
``` json
{
"monitorNoReset": true
}
```
Remove the `${workspaceFolder}` variable from all settings. Consider the absence of a slash at the beginning of the path as `${workspaceFolder}`
Delete the option
``` json
{
"sdkconfigFilePath": "${workspaceFolder}/sdkconfig"
}
```
Instead, automatically create a file in the root of the project named `sdkconfig` + `_` + configuration name.

I have many more ideas. If You are interested, I can work out all the points in more detail.

SinglWolf, 2025-02-19:
How do like idea - to rename `esp_idf_project_configuration.json` in `espressif.json`? I think this name is unique and concise.
It would be great if the format was `ini` from the very beginning. Would it be much easier to implement the `Project import` for `esp` from a`PlatformIO` extension?

brianignacio5, 2025-06-04:
Hi @SinglWolf 

I had analyze your suggestions but they are very specific to your use case.

1) `partitionFilePath` is defined in the sdkconfig file so it doesn't make sense to duplicate it outside in the `esp_idf_project_configuration.json`.
2) imageFolderPath ? How do you consume this folder path ? As far as I know, implementations of filesystems are part of the partitions table which is defined in custom partition and custom partition csv is defined in sdkconfig.
3) imageFS ? Also how would you use it ? Isn't defined in the partition table too ?
4) monitorNoReset we can add, sounds good.
5) Some users might want to use same sdkconfig settings and just change other settings, so a generated `sdkconfig_configurationName`. Other users might want to set non relative to current project sdkconfig paths, so on.

We can indeed rename `esp_idf_project_configuration.json` to something else. We are discussing this and other fields. If you have any more ideas please let us know.

SinglWolf, 2025-06-04:
Hi @brianignacio5!
It's been a long time and everything has slipped my mind, alas, I'm not young anymore.
My project is currently on hold due to family issues. As soon as the opportunity arises, I will immediately write another series of ideas.

closed 2025-08-29 by brianignacio5