espressif/vscode-esp-idf-extension #1422: [Feature Request]: Nostalgia for the classic menuconfig

SinglWolf, 2025-02-03:
Intellectually, I understand that the graphical display of the sdkconfig settings looks beautiful and cool, but in my heart I still miss the classic menuconfig.
Is it possible to make an option somewhere in the settings with a check mark: "Display Classic menuconfig (for very Old's)"
I think `Old's` will understand me.  😄

brianignacio5, 2025-02-05:
Hehe I understand @SinglWolf ! You could just run the `ESP-IDF: Open ESP-IDF Terminal` and type `idf.py menuconfig` to run the old menuconfig.

I could also add a command to do the steps mentioned before, should be simple enough !

SinglWolf, 2025-02-05:
> You could just run the `ESP-IDF: Open ESP-IDF Terminal` and type `idf.py menuconfig` to run the old menuconfig.
> 
> I could also add a command to do the steps mentioned before, should be simple enough !

Hm. I'm not that lazy. 😁
The thing is, I have a multi project that currently has four configurations. If I'm smart enough 🧠, I'll add the same amount more. For this reason, I would like to work with menuconfig in the editor area rather than in the terminal.
I hope you get my idea💡.

closed 2025-02-27 by github-actions[bot]