espressif/vscode-esp-idf-extension #1423: Problem [BUG] in VSCODE permiting FLASHING and, CLEAN-BUILD-FLASHING in RELEASE MODE (IDFGH-14565)

filzek, 2025-02-04:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Problem [BUG] in VSCODE permiting FLASHING and, CLEAN-BUILD-FLASHING in RELEASE MODE

ESP-IDF V5.4

brianignacio5, 2025-02-05:
Could you update the issue with the bug information you are experiencing @filzek ?

filzek, 2025-02-05:
It's supposed to do break the process if the settings are to flash and are using RELEASE MODE.

brianignacio5, 2025-02-05:
Ok. But there is a issue template in this repository for a reason. I need to understand:

1) Extension setting and environment configuration
2) Extension log and Doctor command output
3) What is the error? What are the steps to reproduce ?

So far my understanding is, basically if project is in release mode it should not allow flashing and clean build flash ? So release mode can't be to flash in ALL circumstances ? I need a better description of the issue.

filzek, 2025-02-05:
> Ok. But there is a issue template in this repository for a reason. I need to understand:
> 
> 1) Extension setting and environment configuration
Just set to secure boot and release mode.

> 2) Extension log and Doctor command output
It just clean, build and flash.

> 3) What is the error? What are the steps to reproduce ?
Just enable secure boot and release mode.

> 
> So far my understanding is, basically if project is in release mode it should not allow flashing and clean build flash ? So release mode can't be to flash in ALL circumstances ? I need a better description of the issue.

The esp-idf by default in prompt mode blocks the flash in release mode, but in VS Code it flashes the firmware, so, the flash cannot boot it and become irresponsive.

When try to boot a RELEASE with secure boot enable the system should no flash it and should stop the action warning RELEASE MODE ENABLE

brianignacio5, 2025-03-11:
Are you sure you are not confusing flash with flash encryption operations as described in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig-reference.html#config-secure-flash-encryption-mode