espressif/vscode-esp-idf-extension #1430: ESP IDF extension ignore the sdkconfigFilePath in the esp_idf_project_configuration.json file

RCapolungo, 2025-02-17:
### OS

Linux

### Operating System version

Ubuntu 24.04

### Visual Studio Code version

1.97.2

### ESP-IDF version

5.4

### Python version

3.12

### Doctor command output

[report.txt](https://github.com/user-attachments/files/18828227/report.txt)

### Extension

[esp_idf_vsc_ext.log](https://github.com/user-attachments/files/18828233/esp_idf_vsc_ext.log)

### Description

Hello,
I'm trying to setup my project using the esp_idf_project_configuration.json, as i need to have multiple projects inside my working directory.
I'm setting the sdkconfigFilePath to tell to the ESP-IDF extension where to find the sdkconfig file for each configuration and everything is working fine for building. The issue comes when i try to flash. 
This is the command that is displayed during building: 

[image]

If you look the sdkconfig path that is using is different from the one that i set: 

[image]
(I also tried using absolute path but with no luck as explained in this other ticket: https://github.com/espressif/vscode-esp-idf-extension/issues/1408).

I'm using the Extension provided in this ticket: https://github.com/espressif/vscode-esp-idf-extension/issues/1425

### Debug Message

When building: 
[image]
When flashing: 
[image]

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2025-02-18:
Based on your doctor command output, it seems that you already defined compiler arguments that include SDKCONFIG file path. If you define it in compiler arguments, the value from sdkconfig file path would not be used.

Compile Arguments: -DPYTHON_DEPS_CHECKED=1,-DESP_PLATFORM=1,-DBOARD=devkit,-DENV=prod,-B,<HOMEPATH>/Projects/grms3-esp-main/build/devkit,-S,<HOMEPATH>/Projects/grms3-esp-main,-DSDKCONFIG="${CMAKE_CURRENT_LIST_DIR}/tests/features_test/configs/esp-devkit/sdkconfig",-DSDKCONFIG_DEFAULTS='<HOMEPATH>/Projects/grms3-esp-main/app/configs/esp-devkit/sdkconfig.defaults'

RCapolungo, 2025-02-18:
Thank you for the fast response @brianignacio5.
This is my [esp_idf_project_configuration.json](https://github.com/user-attachments/files/18841348/esp_idf_project_configuration.json) file.
I didn't set any sdkconfig path in the compile arguments, i think it was the extension that set it.
I don't know if it's the cause but i'm also configuring some other parameters in the [CMakeLists.txt](https://github.com/user-attachments/files/18841488/CMakeLists.txt)

brianignacio5, 2025-02-18:
Hi @RCapolungo !

That's the issue ! 

If you use in CMakeLists.txt
```
set(SDKCONFIG_DEFAULTS "${CMAKE_CURRENT_LIST_DIR}/tests/features_test/configs/esp-flat/sdkconfig.defaults")
        set(SDKCONFIG "${CMAKE_CURRENT_LIST_DIR}/tests/features_test/configs/esp-flat/sdkconfig")
```
Whatever you define before gets ultimately replaced by this when you run the build system.

I suggest to test with another simple example project with multiple project configuration profile (esp_idf_project_configuration.json) and keep CMakeLists.txt without modifying variables like SDKCONFIG or SDKCONFIG_DEFAULTS to see if sdkconfig value from Project configuration editor works.

RCapolungo, 2025-02-18:
Thank you @brianignacio5,
This solved the flash issue, however i'm still having issue on building
This is the report of doctor command: [report.txt](https://github.com/user-attachments/files/18842948/report.txt)
And this is how i changed the [CMakeLists.txt](https://github.com/user-attachments/files/18842965/CMakeLists.txt)

I tried using cli build command and it's working fine.
```idf.py build -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DBOARD=devkit -DENV=prod -DSDKCONFIG=/home/rob/Projects/grms3-esp-main/app/configs/esp-devkit/sdkconfig```

RCapolungo, 2025-02-18:
@brianignacio5 solved by setting "buildDirectoryPath" in the esp_idf_project_configuration.json file to "${workspaceFolder}/build".
Trying to change this path will result in the above error.

One note: Using this configuration now doesn't give me any error, the only strange behavior is that is generating the sdkconfig file in the ${workspaceFolder} directory instead of where is set the sdkconfig/sdkconfig.defaults path

brianignacio5, 2025-02-19:
Could you provide with the latest example `esp_idf_project_configuration.json` for me to reproduce ? I'll try to reproduce it myself to understand the issue here.

RCapolungo, 2025-02-19:
Hi @brianignacio5, sure.
[esp_idf_project_configuration.json](https://github.com/user-attachments/files/18863731/esp_idf_project_configuration.json) however, this morning when i boot up my pc the build was not working anymore unfortunately.

RCapolungo, 2025-02-19:
So i found this error when selecting the IDF_TARGET:
[image]

It seems there's an issue in the path generation

RCapolungo, 2025-02-19:
Using latest vsix installer you shared doesn't seem to solve the issue I'm still having the same error when selecting the IDF_TARGET

RCapolungo, 2025-02-19:
I solved the issue changing the sdkconfig default path to this: ${idf.espIdfPath}/../app/configs/esp-devkit/sdkconfig.defaults
This is my new [esp_idf_project_configuration.json](https://github.com/user-attachments/files/18865284/esp_idf_project_configuration.json)

owen63, 2025-02-21:
This may be related - I'm not sure where to ask this. I want the ESP-IDF Build Your Project command to build multiple variants of a project, either every time I build, or failing that, by changing a config via menuconfig. I want separate build folders and conditionals I can use within CMakeLists.txt files and during compilation. Try as I might, I can't. I also want the Encrypt and Flash command to use the selected build. I understand that under the hood it is invoking `idf.py build` and I effectively want the --project-dir arg to be set by config.

brianignacio5, 2025-02-21:
Hi @owen63 It is not very good (to be updated soon!) but here is the documentation of this project configuration editor:
https://docs.espressif.com/projects/vscode-esp-idf-extension/en/latest/additionalfeatures/project-configuration.html

You could create multiple profiles, each with its own buildPath. Please take a look.

@RCapolungo There was many things you tried that solved your issue. Could we summarize the latest things you did and determine what things remain as bug and need to be fix or improved ?

Thank you very much for testing and reporting this. Hopefully we can end up with a better and easier to use feature.

RCapolungo, 2025-02-21:
Hi @brianignacio5.
I'd like to highlight that my current patch is not completely working, it's true that i'm able to build but i have different issues:

1. The build is working only if the sdkconfig file exists (My workaround is to use the preBuild task to set the target).
2. The build directory cannot be moved and therefore it's fixed to ${workspaceFolder}/build in any other case it will fail the build
3. The preBuild task is performed also in the preFlash task (I verified this by calling the isp.idf set-targe ${myTarget} before building and when flashing it's always called)
4. The sdkconfig default path it's not working if you use the ${workspaceFolder} tag as shown in the previous comments.
5. If in the outer CMakeLists.txt file we do something based on the arguments in the cmake (-DBOARD=MyBoard) command like adding some defines the "set target" button is not calling the build arguments in the idf.py fullclean command

What i'm referring to add some defines based on cmake argumets
[image]

When clicking on set-target button and not calling all the arguments
[image]

What i did from the opening of this tickets is:

1. Removing the set SDKCONFIG and SDKCONFIG_DEFAULTS in the outer CMakeLists.txt file.
2. Forced the build directory in the esp_idf_project_configuration.json to ${workspaceFolder}/build.
3. Forced the sdkconfig_defaults in the esp_idf_project_configuration.json to another relative path as ${workspaceFolder} is not working for this option.

owen63, 2025-02-23:
@brianignacio5 Thanks for your prompt and useful reply. That was just what I was missing!

owen63, 2025-02-24:
@brianignacio5 I expected to find my custom configs in the generated sdkconfig.h file but they are not there. I have attached a sample project to demonstrate. It has 2 versions, each with separate build folder and sdkconfig path. Each has custom sdkconfig default files - one common and one version-specific. But none of the config values specified in these default files appear in the generated sdkconfig or sdkconfig.h file as I expected. Am keen to know why.

[config-test.zip](https://github.com/user-attachments/files/18950969/config-test.zip)

brianignacio5, 2025-03-07:
Hi @RCapolungo  and @owen63 sorry for late reply.

Based on your example project @owen63 , My colleague replied to you the reason why is not visible in sdkconfig. You can review the [ESP-IDF KConfig documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/kconfig/configuration_structure.html#kconfig-files) for more detailed description.

 @RCapolungo  I've mentioned before that using `${workspaceFolder}` is not yet supported. This will be fixed with #1417 
I hope the other settings are clear. If you have any additional blocker, could you perhaps provide me with an example project that we can discuss. Using relative sdkconfig defaults files is the common approach in ESP-IDF Build system as shown in [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#custom-sdkconfig-defaults)