espressif/vscode-esp-idf-extension #1431: Unity testing build does not use custom main Kconfig parameters.

jdbaptista, 2025-02-18:
### OS

Windows

### Operating System version

Windows 10

### Visual Studio Code version

1.97.2

### ESP-IDF version

1.9.0

### Python version

3.11.2

### Doctor command output

[report.txt](https://github.com/user-attachments/files/18854065/report.txt)

### Extension

[esp_idf_vsc_ext.log](https://github.com/user-attachments/files/18854069/esp_idf_vsc_ext.log)

### Description

I have a header file that defines an enum based on the value of a configuration option defined in main/Kconfig.projbuild. 
The configuration option is:
```
config HARDWARE_VERSION
        int "PCB Hardware Version"
        default 1
        help
            The version of the PCB that this build is running on, which
            lets the firmware know the capabilities of the board.
```
The enum is defined in a component header file as:
```
#if CONFIG_HARDWARE_VERSION == 1
enum MatrixLocation {
    MAT1_PAGE0,
    MAT1_PAGE1,
    MAT2_PAGE0,
    MAT2_PAGE1,
    MAT3_PAGE0,
    MAT3_PAGE1,
    MAT_NONE,
};
#endif

#if CONFIG_HARDWARE_VERSION == 2
enum MatrixLocation {
    MAT1_PAGE0,
    MAT1_PAGE1,
    MAT2_PAGE0,
    MAT2_PAGE1,
    MAT3_PAGE0,
    MAT3_PAGE1,
    MAT4_PAGE0,
    MAT4_PAGE1,
    MAT_NONE,
};
#endif
```
When I run `>ESP-IDF: Unit Test: Build and flash unit test app for testing` the build fails, even though building my main application is successful. I noticed that the `unity-app/build/config/sdkconfig.h` file does not contain a definition for CONFIG_HARDWARE_VERSION even though the main application build file `build/config/sdkconfig.h` contains it.

[... 294 lines cut ...]
cc1.exe: some warnings being treated as errors
[963/981] Building C object esp-idf/api_connect/CMakeFiles/__idf_api_connect.dir/src/api_connect.c.obj
C:/Users/bapti/Documents/traffic-pcb-sb/software/esp-firmware/components/api_connect/src/api_connect.c: In function 'tomtomGetServerSpeeds':
C:/Users/bapti/Documents/traffic-pcb-sb/software/esp-firmware/components/api_connect/src/api_connect.c:61:13: warning: unused variable 'responsePosition' [-Wunused-variable]
   61 |     int64_t responsePosition; // keeps track of the position currently
      |             ^~~~~~~~~~~~~~~~
[967/981] Building C object esp-idf/led_matrix_test/CMakeFiles/__led_matrix_test.dir/test_dots_commands.c.obj
C:/Users/bapti/Documents/traffic-pcb-sb/software/esp-firmware/components/led_matrix/test/test_dots_commands.c:12: warning: "TAG" redefined
   12 | #define TAG "test"
      |
In file included from C:/Users/bapti/Documents/traffic-pcb-sb/software/esp-firmware/components/led_matrix/test/test_dots_commands.c:2:
C:/Users/bapti/Documents/traffic-pcb-sb/software/esp-firmware/components/led_matrix/include/dots_commands.h:19: note: this is the location of the previous definition
   19 | #define TAG "dots_commands"
      |
ninja: build stopped: subcommand failed.

 *  The terminal process "C:\Users\bapti\.espressif\tools\ninja\1.11.1\ninja.exe" terminated with exit code: 1. ```

### Other Steps to Reproduce

The repo and commit this problem is occurring in is: https://github.com/zinkwazi/traffic-pcb-sb/commit/954b32c2eca2f1fb7f3c70126c12b90f22aea93a . Building the unity testing app for this commit should fail.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

jdbaptista, 2025-02-18:
This is definitely a clear issue (unless intentionally designed this way), however a workaround exists:

Simply move your main Kconfig.projbuild to some component such as 'common', which will then be included in the unity-app build.

brianignacio5, 2025-03-20:
@jdbaptista The thing is that the main component is not added to the unit-test-app.

Please take a look at the [unit testing documentation](https://docs.espressif.com/projects/vscode-esp-idf-extension/en/latest/additionalfeatures/unit-testing.html)

The vscode is basically adding your current project components as extra components in a unit-test-app esp-idf project. The main component can't be added because all esp-idf project have a main component which is the entry point.

I suggest you to modify your code structure a bit. As you can see, the `testable` component of the [ESP-IDF unit test example](https://github.com/espressif/esp-idf/tree/master/examples/system/unit_test) shows how should you create a component and write tests for that component. Your main component just consumes this component (both in your current project and in the unit-test-app component)

closed 2025-03-20 by jdbaptista