espressif/vscode-esp-idf-extension #1432: macOS Could not create a new process and open a pseudo-tty

Tinyu-Zhao, 2025-02-19:
### OS

MacOS

### Operating System version

macOS 14.7.3

### ESP-IDF version

5.5(master branch)

### Python version

3.13.0

### Doctor command output

[report.txt](https://github.com/user-attachments/files/18857542/report.txt)

### Extension

[esp_idf_vsc_ext.log](https://github.com/user-attachments/files/18857549/esp_idf_vsc_ext.log)

### Description

I have encountered this problem before, on my old macbook pro 16 inch 2019, but replaced the new Mac mini has not encountered this problem for a long time, I have recently been using idf to do a project, need to frequently modify the code, build & flash & monitor, the specific performance is, when you frequently use esp- idf plugin in the build & flash & monitor button, your terminal will probably prompt you to
'[Could not create a new process and open a pseudo-tty.]', you can't use the terminal any more, you can only reboot or force a change
`sudo sysctl -w kern.tty.ptmx_max=768 `
Although I know it's a problem with tty resources, the behavior of the build & flash & monitor buttons in the esp-idf plugin is to always destroy the current terminal, create, destroy, create again, (e.g. if I click on the build & flash & monitor buttons, he will destroy the terminal of the previous build task, and create a new one to execute the build). build command, then destroy the flash task terminal, create a new one to execute the flash command, and then monitor) I think this is the root cause of tty's resource shortage, instead of always calling `idf.py build flash -p /dev/cu.usbmodem11101' directly from the same terminal window. monitor `

https://github.com/user-attachments/assets/27693b53-911e-48d6-9189-ca62e85fa6e9

As in the video, I use the build & flash & monitor commands to destroy->create->destroy->create... terminal, whereas idf.py build flash -p /dev/cu.usbmodem11101 monitor just needs a terminal window, I really like this plugin, but using it too often makes it impossible for me to create a new terminal, which is a real pain!

### Debug Message

No

### Other Steps to Reproduce

Use the build & flash & monitor features of the idf plugin a hundred times in a day!

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Tinyu-Zhao, 2025-02-21:
@brianignacio5 Do you have any thoughts on this, I think 'idf.py build flash -p /dev/cu.usbmodem11101 monitor' can be used instead of the current command.

brianignacio5, 2025-02-21:
I @Tinyu-Zhao  sorry for late reply. 

I was unaware of this issue but I think the build should reuse the same terminal as well as the flash tasks. They are all indeed separated.

I understand that using `idf.py build flash -p /dev/cu.usbmodem11101 monitor` but that would limit the other arguments we used to configure build, flash and monitor. For example you can set custom compile args, custom flash args and define things like if monitor should or not reset which are baked into extension code for build, flash and monitor tasks.

May I suggest using `"idf.customTask": "idf.py build flash -p /dev/cu.usbmodem11101 monitor"` and run this custom task with `ESP-IDF: Execute custom task` command or the status bar icon:

[image]

Tinyu-Zhao, 2025-02-21:
I think it's good as long as it solves the problem, but I'm not sure how to put my task in a Custom Task, and it's useless for me to click on it at the moment.

brianignacio5, 2025-02-21:
Add `"idf.customTask": "idf.py build flash -p /dev/cu.usbmodem11101 monitor"` in your `.vscode/settings.json` and try again.

Tinyu-Zhao, 2025-02-21:
> idf.py build flash -p /dev/cu.usbmodem11101 monitor

Thank you!

brianignacio5, 2025-02-26:
I'm closing this issue hoping this solution works for you.

Comment here if you would like further assistance.

closed 2025-02-26 by brianignacio5