espressif/vscode-esp-idf-extension #1434: [Feature Request]: OpenOCD Profiling

bryghtlabs-richard, 2025-02-19:
**Is your feature request related to a problem? Please describe.**

I'm always frustrated when an embedded device feel slow. Especially if it's my embedded device. Doubly so if I don't know where my device is spending my cycles.

**Describe the solution you'd like**

I'd like the ability to record a profile through OpenOCD and analyze it within VSCode

1. We'd need UI to set the target CPU core, then trigger `openocd profile ${profileTime} ${filename}`. The UI should have the option to specify `profileTime`, `filename`, and select a CPU core(if applicable). If `filename` is not specified, the plugin should choose a temporary file.
2. Once the profile data is captured, the UI should load the profile data, look it up in the ELF file according to function names, then display a list of functions that were captured in the profile, along with a percentage of samples per function, sorted by which functions were sampled most often.

4. When a function is chosen for review, the UI should either open a new text window for the function, or take us to the function in the usual text area. The function's lines should be annotated according to the percentage or number of samples spent on each line.

**Describe alternatives you've considered**

Currently, I use a custom fork of `gprof`, shell scripts, and a text editor. It's the same solution I've been using for ~20 years, but it's frustrating to need to disconnect VSCode and switch to my old tools just to gather profiling data to switch back to VSCode to edit it.

**Additional context**

***Example of Trace32's a per-function view***
[image]

***Example of ARM Realview's function's per-line view***
[image]

gerekon, 2025-08-28:
Thanks! Works now.

gerekon, 2025-08-29:
Ok.
The bad news are that our chips update DPC on halt only.
But good news are that for RISCV chips we have `ASSIST_DEBUG_CORE_0_RCD_PDEBUGPC_REG`  which can be configured to log PC and keep the last value after the reset. So looks like it should not be a big problem to implement feature like for ESP32-S3.