espressif/vscode-esp-idf-extension #1435: v1.9.1 can't detect "IDF_TOOLS_PATH" environment variable

ChiaYen-Kan, 2025-02-20:
### OS

Windows

### Operating System version

windows 10

### Visual Studio Code version

1.97.2

### ESP-IDF version

5.4

### Python version

3.10

### Doctor command output

[report.txt](https://github.com/user-attachments/files/18879678/report.txt)

### Extension

[esp_idf_vsc_ext.log](https://github.com/user-attachments/files/18879565/esp_idf_vsc_ext.log)

### Description

idf is esp-idf-tools-setup-offline-5.4
after update v1.9.1, extension can't detect "IDF_TOOLS_PATH" environment variable in "USE EXISTING SETUP"

environment variable
[image]

v1.9.0
[image]

v1.9.1
[image]

### Debug Message

N/A

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

ChiaYen-Kan, 2025-02-24:
any one know how to make v1.9.1 can detect IDF_TOOLS_PATH environment variable for using pre-install IDF sdk ?

Fabricio-ESP, 2025-02-24:
Hi @ChiaYen-Kan , I tried to reproduce the issue you are facing but did not have any success. I installed IDF 5.4 using the offline installer and later installed VSCode with ESP-IDF extension version 1.8.1, upgrade to 1.9.0 and then to 1.9.1. On the 3 versions the IDF installation was successfully available in VSCode. Would you be able to provide details on which steps you took before you  face the issue you reported?

Could it be that the extension failed to find the correct configuration when restarting after the upgrade? if you close and re-open VSCode and then run the command "ESP-IDF: Configure ESP-IDF Extension" and select "Use Existing Setup" Is IDF 5.4 listed as one installed version?

ChiaYen-Kan, 2025-02-25:
v1.9.0 can auto find "IDF_TOOLS_PATH" environment variable 
[image]

v1.9.1 can't auto find "IDF_TOOLS_PATH" environment variable even restart vscode
[image]

brianignacio5, 2025-02-28:
I think I might have found the issue and fix it in #1451 

You can test these changes by installing this [esp-idf-extension.vsix.zip](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/2669612027.zip)
- click menu **View** -> **Command Palette...**, 
- type **Install from VSIX** and then 
- select downloaded **esp-idf-extension.vsix** file to install the extension.

ChiaYen-Kan, 2025-03-03:
> I think I might have found the issue and fix it in [#1451](https://github.com/espressif/vscode-esp-idf-extension/pull/1451)
>
> You can test these changes by installing this [...]

@brianignacio5 this test version is also can't find "IDF_TOOLS_PATH" environment variable

brianignacio5, 2025-03-03:
I've tested on Windows 11 with offline installer 5.4 and it was working. I think the issue is that the extension has setting `idf.toolsPathWin` in `.vscode/settings.json` pointing to a different location from global system env variable IDF_TOOLS_PATH.

Could you share the latest doctor command output using last vsix installer ?

ChiaYen-Kan, 2025-03-03:
i remove this setting
```
"idf.hasWalkthroughBeenShown": true,
"idf.pythonInstallPath": "F:\\esp32-c6\\Espressif\\tools\\idf-python\\3.11.2\\python.exe",
"idf.gitPathWin": "F:/esp32-c6/Espressif/tools/idf-git/2.44.0/cmd/git.exe",
"idf.espIdfPathWin": "F:/esp32-c6/Espressif/frameworks/esp-idf-v5.4/",
"idf.toolsPathWin": "F:\\esp32-c6\\Espressif",
```

then test v1.9.0, v1.9.1, @brianignacio5's test version, only v1.9.0 can detect IDF_TOOLS_PATH

ChiaYen-Kan, 2025-03-03:
doctor command of @brianignacio5's test version
[report.txt](https://github.com/user-attachments/files/19050604/report.txt)

brianignacio5, 2025-03-03:
I've noticed there is an error in your report that doesn't happen in my setup.

```json
{"message":"WARNING: The following issue occurred while accessing the ESP-IDF version file in the Python environment: [Errno 2] No such file or directory: 'F:\\\\esp32-c6\\\\Espressif\\\\python_env\\\\idf5.4_py3.11_env\\\\idf_version.txt'. (Diagnostic information. It can be ignored.)\r\nERROR: Both of the directories (..\\idf-exe\\.. and ..\\tools) has to be in the PATH:\r\n\r\nF:\\esp32-c6\\Espressif\\tools\\xtensa-esp-elf-gdb\\14.2_20240403\\xtensa-esp-elf-gdb\\bin;F:\\esp32-c6\\Espressif\\tools\\riscv32-esp-elf-gdb\\14.2_20240403\\riscv32-esp-elf-gdb\\bin;F:\\esp32-c6\\Espressif\\tools\\xtensa-esp-elf\\esp-14.2.0_20241119\\xtensa-esp-elf\\bin;F:\\esp32-c6\\Espressif\\tools\\esp-clang\\esp-18.1.2_20240912\\esp-clang\\bin;F:\\esp32-c6\\Espressif\\tools\\riscv32-esp-elf\\esp-14.2.0_20241119\\riscv32-esp-elf\\bin;F:\\esp32-c6\\Espressif\\tools\\esp32ulp-elf\\2.38_20240113\\esp32ulp-elf\\bin;F:\\esp32-c6\\Espressif\\tools\\cmake\\3.30.2\\bin;F:\\esp32-c6\\Espressif\\tools\\openocd-esp32\\v0.12.0-esp32-20241016\\openocd-esp32\\bin;F:\\esp32-c6\\Espressif\\tools\\ninja\\1.12.1\\;F:\\esp32-c6\\Espressif\\tools\\idf-exe\\1.0.3\\;F:\\esp32-c6\\Espressif\\tools\\ccache\\4.10.2\\ccache-4.10.2-windows-x86_64;F:\\esp32-c6\\Espressif\\tools\\dfu-util\\0.11\\dfu-util-0.11-win64;F:\\esp32-c6\\Espressif\\python_env\\idf5.4_py3.11_env\\Scripts;F:/esp32-c6/Espressif/frameworks/esp-idf-v5.4/tools;%PATH%C:\\Program Files\\Basler\\pylon 4\\pylon\\bin\\x64;C:\\Program Files\\Basler\\pylon 4\\pylon\\bin\\Win32;C:\\Program Files\\Basler\\pylon 4\\genicam\\Bin\\Win64_x64;C:\\Program Files\\Basler\\pylon 4\\pylonc\\bin\\Win32;C:\\Program Files\\Basler\\pylon 4\\pylonc\\bin\\x64;C:\\Program Files\\Basler\\pylon 4\\genicam\\Bin\\Win32_i86;C:\\Program Files\\Python310\\Scripts\\;C:\\Program Files\\Python310\\;C:\\Windows\\system32;C:\\Windows;C:\\Windows\\System32\\Wbem;C:\\Windows\\System32\\WindowsPowerShell\\v1.0\\;C:\\Windows\\System32\\OpenSSH\\;C:\\Program Files\\Microsoft VS Code\\bin;C:\\Program Files (x86)\\STMicroelectronics\\STM32 ST-LINK Utility\\ST-LINK Utility;C:\\Program Files\\nodejs\\;C:\\Program Files (x86)\\IVI Foundation\\VISA\\WinNT\\Bin\\;C:\\Program Files\\IVI Foundation\\VISA\\Win64\\Bin\\;C:\\Program Files\\Git\\cmd;C:\\Program Files\\TortoiseGit\\bin;C:\\Program Files (x86)\\NVIDIA Corporation\\PhysX\\Common;C:\\Program Files\\NVIDIA Corporation\\NVIDIA NvDLISR;C:\\Program Files\\dotnet\\;C:\\Users\\kcy\\AppData\\Local\\Microsoft\\WindowsApps;F:\\Tool\\jdk-21.0.3+9\\bin;C:\\Users\\kcy\\AppData\\Roaming\\npm;D:\\aria2-1.36.0-clang-x86_64-gnutls;F:\\OpenSrcBuild\\nasm-2.16.03;D:\\work3\\gradle-8.12;C:\\Users\\kcy\\.dotnet\\tools\r\n\r\n","stack":"Error: WARNING: The following issue occurred while accessing the ESP-IDF version file in the Python environment: [Errno 2] No such file or directory: 'F:\\\\esp32-c6\\\\Espressif\\\\python_env\\\\idf5.4_py3.11_env\\\\idf_version.txt'. (Diagnostic information. It can be ignored.)\r\nERROR: Both of the directories (..\\idf-exe\\.. and ..\\tools) has to be in the PATH:\r\n\r\nF:\\esp32-c6\\Espressif\\tools\\xtensa-esp-elf-gdb\\14.2_20240403\\xtensa-esp-elf-gdb\\bin;F:\\esp32-c6\\Espressif\\tools\\riscv32-esp-elf-gdb\\14.2_20240403\\riscv32-esp-elf-gdb\\bin;F:\\esp32-c6\\Espressif\\tools\\xtensa-esp-elf\\esp-14.2.0_20241119\\xtensa-esp-elf\\bin;F:\\esp32-c6\\Espressif\\tools\\esp-clang\\esp-18.1.2_20240912\\esp-clang\\bin;F:\\esp32-c6\\Espressif\\tools\\riscv32-esp-elf\\esp-14.2.0_20241119\\riscv32-esp-elf\\bin;F:\\esp32-c6\\Espressif\\tools\\esp32ulp-elf\\2.38_20240113\\esp32ulp-elf\\bin;F:\\esp32-c6\\Espressif\\tools\\cmake\\3.30.2\\bin;F:\\esp32-c6\\Espressif\\tools\\openocd-esp32\\v0.12.0-esp32-20241016\\openocd-esp32\\bin;F:\\esp32-c6\\Espressif\\tools\\ninja\\1.12.1\\;F:\\esp32-c6\\Espressif\\tools\\idf-exe\\1.0.3\\;F:\\esp32-c6\\Espressif\\tools\\ccache\\4.10.2\\ccache-4.10.2-windows-x86_64;F:\\esp32-c6\\Espressif\\tools\\dfu-util\\0.11\\dfu-util-0.11-win64;F:\\esp32-c6\\Espressif\\python_env\\idf5.4_py3.11_env\\Scripts;F:/esp32-c6/Espressif/frameworks/esp-idf-v5.4/tools;%PATH%C:\\Program Files\\Basler\\pylon 4\\pylon\\bin\\x64;C:\\Program Files\\Basler\\pylon 4\\pylon\\bin\\Win32;C:\\Program Files\\Basler\\pylon 4\\genicam\\Bin\\Win64_x64;C:\\Program Files\\Basler\\pylon 4\\pylonc\\bin\\Win32;C:\\Program Files\\Basler\\pylon 4\\pylonc\\bin\\x64;C:\\Program Files\\Basler\\pylon 4\\genicam\\Bin\\Win32_i86;C:\\Program Files\\Python310\\Scripts\\;C:\\Program Files\\Python310\\;C:\\Windows\\system32;C:\\Windows;C:\\Windows\\System32\\Wbem;C:\\Windows\\System32\\WindowsPowerShell\\v1.0\\;C:\\Windows\\System32\\OpenSSH\\;C:\\Program Files\\Microsoft VS Code\\bin;C:\\Program Files (x86)\\STMicroelectronics\\STM32 ST-LINK Utility\\ST-LINK Utility;C:\\Program Files\\nodejs\\;C:\\Program Files (x86)\\IVI Foundation\\VISA\\WinNT\\Bin\\;C:\\Program Files\\IVI Foundation\\VISA\\Win64\\Bin\\;C:\\Program Files\\Git\\cmd;C:\\Program Files\\TortoiseGit\\bin;C:\\Program Files (x86)\\NVIDIA Corporation\\PhysX\\Common;C:\\Program Files\\NVIDIA Corporation\\NVIDIA NvDLISR;C:\\Program Files\\dotnet\\;C:\\Users\\kcy\\AppData\\Local\\Microsoft\\WindowsApps;F:\\Tool\\jdk-21.0.3+9\\bin;C:\\Users\\kcy\\AppData\\Roaming\\npm;D:\\aria2-1.36.0-clang-x86_64-gnutls;F:\\OpenSrcBuild\\nasm-2.16.03;D:\\work3\\gradle-8.12;C:\\Users\\kcy\\.dotnet\\tools\r\n\r\n\n\tat \\dist\\extension.js:2:1436281\n\tat ChildProcess.exithandler (node:child_process:415:7)\n\tat ChildProcess.emit (node:events:518:28)\n\tat maybeClose (node:internal/child_process:1104:16)\n\tat ChildProcess._handle.onexit (node:internal/child_process:304:5)","level":"error","timestamp":"2025-03-03T08:51:44.260Z"}
```
Could you share me which offline installer you use ?

I've used offline installer v5.4 from  https://dl.espressif.com/dl/esp-idf/ on Windows 11

ChiaYen-Kan, 2025-03-03:
my idf is download from https://dl.espressif.com/dl/esp-idf/
only install c6 toolkit without other chip

[image]

brianignacio5, 2025-03-03:
Maybe there are settings defined in `%APPDATA%\Code\User\settings.json` which you can check with the `Preferences: Open User Settings (JSON)`. 

I've added a new commit to also always include system IDF_TOOLS_PATH. Here is the new [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/2680890198.zip) Please give it a try. If it doesn't work, please attach a new doctor command output.

ChiaYen-Kan, 2025-03-04:
> Maybe there are settings defined in `%APPDATA%\Code\User\settings.json` which you can check with the `Preferences: Open User Settings (JSON)`.
>
> I've added a new commit to also always include system [...]

this new vision is can't IDF_TOOLS_PATH

doctor command
[report.txt](https://github.com/user-attachments/files/19062180/report.txt)

brianignacio5, 2025-03-04:
Now is probably fixed with this latest [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/2685986833.zip). Please give it a try, I hope this one does fix the issue.

Turns out that the extension global state from before is still being used but was not resolved properly. Since your global state was from 1.9.0 even if IDF_TOOLS_PATH was found, the setup from global state was still used.

ChiaYen-Kan, 2025-03-04:
> Now is probably fixed with this latest [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/2685986833.zip). Please give it a try, I hope this one does fix the [...]

this version is work

[image]

closed 2025-03-06 by brianignacio5