espressif/vscode-esp-idf-extension #1437: 使用VScode ESP-IDF插件烧录程序出现问题

XiaPeng1012, 2025-02-20:
### OS

Windows

### Operating System version

windows 11

### Visual Studio Code version

1.97.2

### ESP-IDF version

5.3.2

### Python version

3.11.2

### Doctor command output

d:\Software\ESP\tools\tools\openocd-esp32\v0.12.0-esp32-20241016\openocd-esp32\bin\openocd.exe

Open On-Chip Debugger v0.12.0-esp32-20241016 (2024-10-16-14:17)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html

d:\Software\ESP\tools\tools\openocd-esp32\v0.12.0-esp32-20241016\openocd-esp32\bin\openocd.exe

[OpenOCD]
Open On-Chip Debugger v0.12.0-esp32-20241016 (2024-10-16-14:17)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html

debug_level: 2

Info : only one transport option; 
autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000

Info : Listening on port 6666 for tcl conn
ections
Info : Listening on port 4444 for telnet connections

Info : esp_usb_jtag: serial (A0:85:E3:
FD:BE:74)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz

Info : JTAG tap: esp32s3.ta
p0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.tap1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)

Info : [esp32s3.cpu0] Examinatio
n succeed

Info : [esp32s3.cpu1] Examination succ
eed
Info : [esp32s3.cpu0] starting gdb server on 3333
Info : Listening on port 3333 for gdb connections

Info : [esp32s3.cpu0] Target halted, PC=0x40376207, debug_reason=00000000

Info : [esp32s3.cpu0] Reset ca
use (3) - (Software core reset)
Info : Set GDB target to 'esp32s3.cpu0'

Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000

Info : [esp32s3.cpu1] Reset cau
se (3) - (Software core reset)

Info : accepting 'tcl' connection on tcp/6666
Info : dropped 'tcl' connection

Info : accepting 'tcl' connection on tcp/6666
ready

Flashing "e:/Project/2025/Wireless_Tool/wireless_tool/build/bootloader/bootloader.bin" at 0x0

** program_esp input args <0x0 verify> **

Info : JTAG tap: esp32s3.tap0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.tap1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)

Info : [esp32s3.cpu0] requesting target halt and execut
ing a soft reset

Info : [esp32s3.cpu0] Debug controller was reset.

Info : [esp32s3.cpu0] Core was reset.

Info : [esp32s3.cpu0] Target halted, PC=0x500000EF
, debug_reason=00000000

Info : [esp32s3.cpu0] Reset cause (3)
 - (Software core reset)

Info : [esp32s3.cpu0] Core
 was reset.

Info : [esp32s3.cpu0] Target halted, PC
=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Debug controller was res
et.
Info : [esp32s3.cpu1] Core was reset.

Info : [esp32s3.cpu1] Target halted,
 PC=0x40000400, debug_reason=00000000

Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)

Info : [esp32s3.cpu0] Reset cause (3
) - (Software core reset)

** Programming Started **

Info : [esp32s3.cpu0] Target halted, PC=0x4038CB86, deb
ug_reason=00000001

Info : Flash mapping 0: 0x10020 -> 0x3
c0e0020, 212 KB
Info : Flash mapping 1: 0x50020 -> 0x42000020, 867 KB
Info : Auto-detected flash bank 'esp32s3.cpu0.flash' size 16384 KB
Info : Using flash bank 'esp32s3.cpu0.flash' size 16384 KB

❌ Error: timed out while waiting for target halted

Info : [esp32s3.cpu0] T
arget halted, PC=0x40376207, debug_reason=00000000

❌ Error: [esp32s3.cpu0] not halted 0, pc 0x40376207, ps 0x60623
Error: Failed to wait algorithm (-302)!
Error: Algorithm run failed (-302)!

❌ Error: Failed to run flasher stub (-302)!

** Flashing Failed **
-1

[/OpenOCD]
[Flash]
Failed to flash the device (JTag), please try again [got response: '-1', expecting: '0']

### Description

第一次烧录程序正常，在进行第二次程序烧录后烧录异常，使用的VSCode烧录方式为JTAG烧录。
[image]

经过测试发现，烧录程序小于某一个大小的时候多次烧录不会出现问题，当程序大小大于某一个范围的时候，多次烧录程序就会异常，在程序烧录异常后，手动擦除程序flash数据后，使用jtag方式烧录程序不在发生异常。在JTAG烧录程序发生异常的时候，使用UART烧录方式能够进行正常程序烧录。

### Debug Message

NULL

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

erhankur, 2025-02-20:
@XiaPeng1012 Could you please share the verbose OpenOCD logs. You can append below commands to the and of config script.

```
-c "esp stub_log on" -d3 -l log.txt
```

gerekon, 2025-02-20:
@XiaPeng1012 Can I ask you to try flashing using command line?

`openocd -f board/esp32s3-builtin.cfg -d3 -l ocd.log -c 'init; program_esp_bins e:/Project/2025/Wireless_Tool/wireless_tool/build flasher_args.json'`.

And then attach produced `ocd.log` here.

XiaPeng1012, 2025-02-21:
> [@XiaPeng1012](https://github.com/XiaPeng1012) Can I ask you to try flashing using command line?
> 
> `openocd -f board/esp32s3-builtin.cfg -d3 -l ocd.log -c 'init; program_esp_bins e:/Project/2025/Wireless_Tool/wireless_tool/build flasher_args.json'`.
> 
> And then attach produced `ocd.log` here.

[ocd.log](https://github.com/user-attachments/files/18897228/ocd.log)好的，我应按照您的说明方式进行操作，这是生成的ocd.log

XiaPeng1012, 2025-02-21:
> [@XiaPeng1012](https://github.com/XiaPeng1012)我可以请您尝试使用命令行进行刷新吗？
> 
> `openocd -f board/esp32s3-builtin.cfg -d3 -l ocd.log -c 'init; program_esp_bins e:/Project/2025/Wireless_Tool/wireless_tool/build flasher_args.json'`。
> 
> 然后将制作的附加`ocd.log`到这里。

@gerekon 您好，我已经按照您的说明进行操作，我会将生成的ocd.log作为附件添加。

[ocd.log](https://github.com/user-attachments/files/18897259/ocd.log)

XiaPeng1012, 2025-02-21:
> [@XiaPeng1012](https://github.com/XiaPeng1012) Could you please share the verbose OpenOCD logs. You can append below commands to the and of config script.
> 
> ```
> -c "esp stub_log on" -d3 -l log.txt
> ```

@erhankur 您好，我已经按照gerkon的方式生成了OpenOCD的 log,现在将生成的日志以附件的形式发送给您。

[ocd.log](https://github.com/user-attachments/files/18897309/ocd.log)

erhankur, 2025-02-21:
@XiaPeng1012 I don't see the stub logs here. Please add `-c "esp stub_log on" -d3 -l log.txt`

XiaPeng1012, 2025-02-21:
> [@XiaPeng1012](https://github.com/XiaPeng1012) I don't see the stub logs here. Please add `-c "esp stub_log on" -d3 -l log.txt`

@erhankur 抱歉，我可能不知道如何添加```-c "esp stub_log on" -d3 -l log.txt```，你能告诉我更为详细的添加方式么

XiaPeng1012, 2025-02-22:
> You can open your project’s `.vscode/settings.json` and add or modify:
> 
> ```
> {
>   "idf.openOcdLaunchArgs": [
>       "-c",
>       "init",
>       "-c",
>       "reset halt",
>       "-c",
>       "esp stub_log on",
>       "-d3",
>       "-l log.txt"
>   ]
> }
> ```

@erhankur 按照您的操作步骤，已经生成了log文件，我将在链接中附带log.txt，请您查收，谢谢！

[log.txt](https://github.com/user-attachments/files/18918267/log.txt)

erhankur, 2025-02-25:
@XiaPeng1012 Could you please try adding `no_clock_boost` option to the programming command.

```
openocd -f board/esp32s3-builtin.cfg -d3 -l ocd.log -c 'init; program_esp_bins e:/Project/2025/Wireless_Tool/wireless_tool/build flasher_args.json no_clock_boost'
```

@brianignacio5 @radurentea Is it possible to add this option into the json file?

XiaPeng1012, 2025-02-26:
> [@XiaPeng1012](https://github.com/XiaPeng1012) Could you please try adding `no_clock_boost` option to the programming command.
>
> ```
> openocd -f board/esp32s3-builtin.cfg -d3 -l ocd.log -c 'init; [...]

很奇怪，现在即使没有这条命令也无法复现该问题，使用Jtag也能烧录成功。

brianignacio5, 2025-02-26:
> [@XiaPeng1012](https://github.com/XiaPeng1012) Could you please try adding `no_clock_boost` option to the programming command.
>
> ```
> openocd -f board/esp32s3-builtin.cfg -d3 -l ocd.log -c 'init; [...]

Sure we can, I'm just wondering, how can we determine when to add it ?

Right now extension sends a fixed TCL command `program_esp_bins <buildPath> flasher_args.json verify reset` to perform a JTAG flash. We could move last 2 arguments to an extension setting like `"idf.jtagFlashCommandExtraArgs": ["verify", "reset"]` and allow user to customize it. Does this solution makes sense @erhankur  ?

erhankur, 2025-02-26:
@brianignacio5 "idf.jtagFlashCommandExtraArgs" is a good idea. We can keep `verify reset` as the default, and the others can be added when needed. For example, the `compress` option might be useful to reduce programming time. 

The full list is here;

https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/index.html#upload-application-for-debugging

PR #1450 (2025-03-25): [VSC-1596] add idf.jtagFlashCommandExtraArgs to modify jtag flash behaviour
## Description

Add `idf.jtagFlashCommandExtraArgs` configuration setting to allow customization of OpenOCD JTAG flash command.

Please review https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/index.html#upload-application-for-debugging for possible arguments to use.

Fixes #1437

## Type of change

- New feature (non-breaking change which adds functionality)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Open a ESP-I [... cut]
closed 2025-03-25 by brianignacio5