espressif/vscode-esp-idf-extension #1442: How to make the plugin use the installed SDK？

goukisun, 2025-02-24:
I have offline installations of 5.1.1, 5.2.1, 5.3.1. The latter installation overwrites system environment variables. You can use a different version of the SDK from the command line by modifying idf_cmd_init.bat Initialize-Idf.ps1 to force IDF_TOOLS_PATH to be specified.
esp idf extension can also support different SDKs by setting the corresponding variables in vscode settings. everything is fine.
But after upgrading esp idf extension, especially after 1.9, it can't find the idf framework, and the gdb debugging also has problems.
I want to know if I am using it in the right way. What are the necessary settings for esp idf extension to recognize the installed SDK version correctly?

Translated with DeepL.com (free version)

brianignacio5, 2025-02-25:
Follow the documentation https://docs.espressif.com/projects/vscode-esp-idf-extension/en/latest/installation.html

brianignacio5, 2025-02-28:
I think I might have found the issue and fix it in #1451 

You can test these changes by installing this [esp-idf-extension.vsix.zip](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/2669612027.zip)
- click menu **View** -> **Command Palette...**, 
- type **Install from VSIX** and then 
- select downloaded **esp-idf-extension.vsix** file to install the extension.

brianignacio5, 2025-03-06:
Any updates @goukisun  ? Please try the [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/2700816629.zip) to see if issue is fixed.

closed 2025-03-06 by brianignacio5