espressif/vscode-esp-idf-extension #1443: Why can't the vscode debug console enter commands?

goukisun, 2025-02-25:
I have installed the plugin and SDK, and I can enter debug mode by creating an example. But the debug console can not enter commands, reported error ”-var-create: unable to create variable object“

brianignacio5, 2025-02-25:
Please follow the issue template.

Also please provide the steps you are doing to create for the error you are having.

If you are using the extension CDT debug adapter please take a look at https://docs.espressif.com/projects/vscode-esp-idf-extension/en/latest/debugproject.html#send-commands-to-gdb 

You need to pass `>` before gdb command.

goukisun, 2025-02-25:
You need to pass > before the gdb command.

Please follow the issue template.
>
Also please provide the steps you are taking to create the error you are experiencing.
>
If you are using the extension CDT debug adapter, please take a look at https://docs.espressif.com/projects/vscode-esp-idf-extension/en/latest/debugproject.html#send-commands-to-gdb
>
You need to pass `>` before the gdb command.

I am too stupid ！！！ need '>'

closed 2025-02-26 by brianignacio5