espressif/vscode-esp-idf-extension #1447: [Feature Request]: Add support for additional options to be passed to QEMU

bdpdx, 2025-02-27:
The arguments the ESP-IDF extension passes to QEMU are hardcoded in getLaunchArguments() in qemuManager.ts.

It would be nice to be able to pass additional arguments, for example the ability to enable additional PSRAM (-m 4M).

Currently one must modify ~/.vscode/extensions/espressif.esp-idf-extension-1.9.1/dist/extension.js which is a bit unwieldy.

I created a pull request with this change, see https://github.com/espressif/vscode-esp-idf-extension/pull/1446

brianignacio5, 2025-03-10:
Hi @bdpdx I've created a PR that basically execute `idf.py qemu` that allows you to run QEMU to debug and monitor with `"idf.qemuDebugMonitor": true` configuration setting in .vscode/settings.json and `"idf.customExtraArgs": ["--qemu-extra-args"]` to pass custom arguments to QEMU in the format of `idf.py qemu ${idf.customExtraArgs} ${idf.qemuDebugMonitor ? "-gdb": ""} monitor`

Please take a look when you can.

You can test these changes by installing this VSIX by click menu View -> Command Palette..., type Install from VSIX and then select downloaded [esp-idf-extension.vsix](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/2721176235.zip) file to install the extension.

JohannesKaranitsch, 2025-03-18:
Hi @brianignacio5 I've tried to pass the custom argument "-nic user,model=open_eth,id=lo0,hostfwd=tcp:127.0.0.1:8080-:80" to QEMU, but didn't get it to work. See attached settings.json.
Tested also with differnt "idf.customExtraArgs" but nothing worked.
Didn't find the correct syntax or sample in the [Espressif documentation preview](https://preview-docs.espressif.com/projects/vscode-esp-idf-extension/en/1462-merge/index.html)
What is the correct syntax to make (multiple) port forwarding(s) as --qemu-extra-args?

[settings.json](https://github.com/user-attachments/files/19321517/settings.json)

brianignacio5, 2025-03-19:
Apologies @JohannesKaranitsch I gave you the incorrect setting in the previous comment. 

Try `"idf.qemuExtraArgs": ["--qemu-extra-args -nic user,model=open_eth,id=lo0,hostfwd=tcp:127.0.0.1:8080-:80"]`

JohannesKaranitsch, 2025-03-19:
Hi @brianignacio5,
I changed the settings.json to the correct settings. Now I can see the extra args on the first line when I start debugging:
C:\esp\.espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe C:\esp\v5.3.2\esp-idf\tools\idf.py -B c:\esp\projects\WebConfig\build qemu --gdb --qemu-extra-args -nic user,model=open_eth,id=lo0,hostfwd=tcp:127.0.0.1:8080-:80 monitor

But when qemu ist started the extra arg is missing and a timeout is occurring:
Generating efuse image: C:\esp\projects\WebConfig\build\qemu_efuse.bin
**Running qemu (bg): qemu-system-xtensa -M esp32 -m 4M -drive file=C:\esp\projects\WebConfig\build\flash_image.bin,if=mtd,format=raw -drive file=C:\esp\projects\WebConfig\build\qemu_efuse.bin,if=none,format=raw,id=efuse -global driver=nvram.esp32.efuse,property=drive,value=efuse -global driver=timer.esp32.timg,property=wdt_disable,value=true -gdb tcp::3333 -S -nic -nographic -serial tcp::5555,server**
Timed out waiting for port 5555 to be open

brianignacio5, 2025-03-19:
Probably you need to add extra quotes to `"idf.qemuExtraArgs": ["--qemu-extra-args '-nic user,model=open_eth,id=lo0,hostfwd=tcp:127.0.0.1:8080-:80'"]`

It might not be parsing it properly.

You can test by open a idf terminal with `ESP-IDF: Open ESP-IDF Terminal` and type a modified command (without monitor) to see if it parsing it correctly `C:\esp.espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe C:\esp\v5.3.2\esp-idf\tools\idf.py -B c:\esp\projects\WebConfig\build qemu --gdb --qemu-extra-args -nic user,model=open_eth,id=lo0,hostfwd=tcp:127.0.0.1:8080-:80`

JohannesKaranitsch, 2025-03-19:
I tried now:
idf.py -B c:\esp\projects\WebConfig\build qemu --gdb --qemu-extra-args -nic user,model=open_eth,id=lo0,hostfwd=tcp:127.0.0.1:8080-:80
and with quotes:
idf.py -B c:\esp\projects\WebConfig\build qemu --gdb --qemu-extra-args '-nic user,model=open_eth,id=lo0,hostfwd=tcp:127.0.0.1:8080-:80'

Same result:
Running qemu (fg): qemu-system-xtensa -M esp32 -m 4M -drive file=C:\esp\projects\WebConfig\build\flash_image.bin,if=mtd,format=raw -drive file=C:\esp\projects\WebConfig\build\qemu_efuse.bin,if=none,format=raw,id=efuse -global driver=nvram.esp32.efuse,property=drive,value=efuse -global driver=timer.esp32.timg,property=wdt_disable,value=true -gdb tcp::3333 -S -nic -nographic -serial mon:stdio
qemu-system-xtensa: -nic -nographic: Parameter 'type' does not accept value '-nographic'
Executing action: user,model=open_eth,id=lo0,hostfwd=tcp:127.0.0.1:8080-:80
ninja: error: unknown target 'user,model=open_eth,id=lo0,hostfwd=tcp:127.0.0.1:8080-:80'
command "user,model=open_eth,id=lo0,hostfwd=tcp:127.0.0.1:8080-:80" is not known to idf.py and is not a Ninja target

brianignacio5, 2025-03-19:
"idf.qemuExtraArgs": ["--qemu-extra-args", "'-nic user,model=open_eth,id=lo0,hostfwd=tcp:127.0.0.1:8080-:80'"] works for me @JohannesKaranitsch

JohannesKaranitsch, 2025-03-19:
@brianignacio5 works also for me.

PR #1462 (2025-03-27): [VSC-1604 | VSC-1606 | VSC-1624] use idf py qemu Allow qemu debug monitor
## Description

This PR modifies the QEMU Debug and monitor implementation to use `idf.py qemu` instead. This allows to offload the extension to keep track of arguments for QEMU.

Add `idf.qemuDebugMonitor`  to allow start monitor session after debug session and `idf.qemuExtraArgs` to define additional `idf.py qemu` arguments.

Fixes #1447 
Fixes #1449
Fixes #1473

## Type of change

- New feature (non-breaking change which adds functionality)

## Steps to test this pull request

Provide a list of steps to test changes i [... cut]
closed 2025-03-27 by brianignacio5