espressif/vscode-esp-idf-extension #1448: Building for device "linux" calls the esp_idf_size which fails.

i400s, 2025-02-27:
### OS

Linux

### Operating System version

linux debian trixie

### Visual Studio Code version

1.97.2

### ESP-IDF version

v5.5-dev-2119-g0461e2ff88

### Python version

3.13.2

### Doctor command output

n/a

### Description

When the compile for linux finishes the program tries to show the usual memory usage summary build information which obviously it can't do.

eg:

error:  cannot read memory types file: [Errno 2] No such file or directory: '/home/wilsonj/.espressif/python_env/idf5.5_py3.13_env/lib/python3.13/site-packages/esp_idf_size/ng/../chip_info/linux.yaml'

 *  The terminal process "/home/wilsonj/.espressif/python_env/idf5.5_py3.13_env/bin/python '/home/wilsonj/Documents/foldered/espressif/esp-idf/tools/idf_size.py', '/home/wilsonj/Documents/foldered/espressif/vscode/esp-idf/i400s_temp__/gpio-relay/test-apps/test-gpio-relay-linux/build/test-gpio-relay-linux.map'" terminated with exit code: 1. 

### Debug Message

As above

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2025-02-28:
You can set `"idf.enableSizeTaskAfterBuildTask": false` in your project's `.vscode/settings.json` to disable the IDF Size task after build task.

i400s, 2025-02-28:
Thanks for the hint. Will remember to check and _**double**_ check the docs next time. I didn't notice it when I looked, so my mistake. Will close.

closed 2025-02-28 by i400s