espressif/vscode-esp-idf-extension #1449: [Feature Request]: QEMU support for debugging and viewing the monitor at the same time

bdpdx, 2025-02-28:
QEMU debugging in VSCode is very useful. Currently however you can only open the monitor or a debugging window but not both. idf.py qemu supports starting qemu monitor with the --gdb flag so it will wait for GDB. In another terminal you can then start idf.py qemu gdb.

Debugging in VSCode is much than at the command line so it would ideal if there was a ESP-IDF: Launch Monitor and Debug Firmware option. Ideally there would also be a flash, open monitor, and debug.

Some things to do would be to support the idf.py passing of flash and efuse file via arguments, and the restarting the command should clean up any existing windows and reload everything.

This would make debugging ESP32 incredibly easier.

brianignacio5, 2025-03-10:
Hi @bdpdx I've created a PR that basically execute `idf.py qemu` that allows you to run QEMU to debug and monitor with `"idf.qemuDebugMonitor": true` configuration setting in .vscode/settings.json and `"idf.customExtraArgs": ["--qemu-extra-args"]` to pass custom arguments to QEMU in the format of `idf.py qemu ${idf.customExtraArgs} ${idf.qemuDebugMonitor ? "-gdb": ""} monitor`

Please take a look when you can.

You can test these changes by installing this VSIX by click menu View -> Command Palette..., type Install from VSIX and then select downloaded [esp-idf-extension.vsix](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/2721176235.zip) file to install the extension.

bdpdx, 2025-04-03:
See my comments here: https://github.com/espressif/vscode-esp-idf-extension/pull/1462#issuecomment-2774446433

PR #1462 (2025-03-27): [VSC-1604 | VSC-1606 | VSC-1624] use idf py qemu Allow qemu debug monitor
## Description

This PR modifies the QEMU Debug and monitor implementation to use `idf.py qemu` instead. This allows to offload the extension to keep track of arguments for QEMU.

Add `idf.qemuDebugMonitor`  to allow start monitor session after debug session and `idf.qemuExtraArgs` to define additional `idf.py qemu` arguments.

Fixes #1447 
Fixes #1449
Fixes #1473

## Type of change

- New feature (non-breaking change which adds functionality)

## Steps to test this pull request

Provide a list of steps to test changes i [... cut]