espressif/vscode-esp-idf-extension #1455: Cannot create property 'IDF_TARGET' on string

pkolt, 2025-03-07:
### OS

MacOS

### Operating System version

macOS Sequoia 15.3.1

### Visual Studio Code version

1.98.0

### ESP-IDF version

5.4.0

### Python version

3.13.2

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS darwin arm64 24.3.0 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 <HOMEPATH>/.bun/bin:/opt/homebrew/opt/openjdk/bin:<HOMEPATH>/.nvm/versions/node/v20.14.0/bin:/opt/homebrew/bin:/usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin:<HOMEPATH>/.go/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.98.0 
Visual Studio Code language en 
Visual Studio Code shell /bin/zsh 
ESP-IDF Extension version 1.9.1 
Workspace folder <HOMEPATH>/my/projects/microelectronics/esp32/zigbee3 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/esp/v5.4/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin:<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin:<HOMEPATH>/.espressif/tools/cmake/3.30.2/CMake.app/Contents/bin:<HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/bin:<HOMEPATH>/.espressif/tools/ninja/1.12.1:<HOMEPATH>/.espressif/tools/esp-rom-elfs/20241011
ESP-IDF extra vars
    OPENOCD_SCRIPTS: <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/share/openocd/scripts
    ESP_ROM_ELF_DIR: <HOMEPATH>/.espressif/tools/esp-rom-elfs/20241011/
    IDF_PYTHON_ENV_PATH: <HOMEPATH>/.espressif/python_env/idf5.4_py3.13_env
    ESP_IDF_VERSION: 5.4
    PATH: <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin:<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin:<HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/bin:<HOMEPATH>/.espressif/python_env/idf5.4_py3.13_env/bin:<HOMEPATH>/esp/v5.4/esp-idf/tools:$PATH
    IDF_DEACTIVATE_FILE_PATH: /var/folders/61/23_5ymcs5y79c_zkfg6fsch00000gn/T/tmpaak98f2midf_89658
User extra vars (idf.customExtraVars)
    0: {
    1: "
    2: O
    3: P
    4: E
    5: N
    6: O
    7: C
    8: D
    9: _
    10: S
    11: C
    12: R
    13: I
    14: P
    15: T
    16: S
    17: "
    18: :
    19: "
[... 298 lines cut ...]
{"message":"The \"file\" argument must be of type string. Received undefined","stack":"TypeError: The \"file\" argument must be of type string. Received undefined\n    at normalizeSpawnArguments (node:child_process:549:3)\n    at spawn (node:child_process:789:13)\n    at Object.execFile (node:child_process:352:17)\n    at Object.t.<computed> [as execFile] (node:electron/js2c/node_init:2:2955)\n    at /dist/extension.js:2:1435853\n    at new Promise (<anonymous>)\n    at Xe (/dist/extension.js:2:1435830)\n    at Se (/dist/extension.js:2:1422686)\n    at Co (/dist/extension.js:2:1558146)\n    at Po (/dist/extension.js:2:1558519)\n    at /dist/extension.js:2:1881906\n    at /dist/extension.js:2:1881359","level":"error","timestamp":"2025-03-07T12:45:01.004Z"}
{"user":true,"message":"Could not get ESP-IDF: New project arguments","stack":"Error: Could not get ESP-IDF: New project arguments\n    at /dist/extension.js:2:1883170","level":"error","timestamp":"2025-03-07T12:45:33.261Z"}

### Extension

{"message":"Cannot access filePath: /Users/pkolt/my/projects/pkolt.ru/CMakeLists.txt","stack":"Error: ENOENT: no such file or directory, access '/Users/pkolt/my/projects/pkolt.ru/CMakeLists.txt'\n    at Object.accessSync (node:fs:246:11)\n    at Object.t.accessSync (node:electron/js2c/node_init:2:9332)\n    at Fe (/Users/pkolt/.vscode/extensions/espressif.esp-idf-extension-1.9.1/dist/extension.js:2:1432827)\n    at /Users/pkolt/.vscode/extensions/espressif.esp-idf-extension-1.9.1/dist/extension.js:2:1434637\n    at We (/Users/pkolt/.vscode/extensions/espressif.esp-idf-extension-1.9.1/dist/extension.js:2:1434823)\n    at Ke (/Users/pkolt/.vscode/extensions/espressif.esp-idf-extension-1.9.1/dist/extension.js:2:1435233)\n    at me (/Users/pkolt/.vscode/extensions/espressif.esp-idf-extension-1.9.1/dist/extension.js:2:1419497)\n    at wt.configureWithDefaultValues (/Users/pkolt/.vscode/extensions/espressif.esp-idf-extension-1.9.1/dist/extension.js:2:1448685)\n    at new wt (/Users/pkolt/.vscode/extensions/espressif.esp-idf-extension-1.9.1/dist/extension.js:2:1445209)\n    at Function.init (/Users/pkolt/.vscode/extensions/espressif.esp-idf-extension-1.9.1/dist/extension.js:2:1445158)\n    at Dp (/Users/pkolt/.vscode/extensions/espressif.esp-idf-extension-1.9.1/dist/extension.js:2:1835796)\n    at Function.kb (file:///Applications/Visual%20Studio%20Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:116:15214)\n    at Function.jb (file:///Applications/Visual%20Studio%20Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:116:14921)\n    at file:///Applications/Visual%20Studio%20Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:116:13079\n    at aw.n (file:///Applications/Visual%20Studio%20Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:112:4447)\n    at aw.m (file:///Applications/Visual%20Studio%20Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:112:4410)\n    at aw.l (file:///Applications/Visual%20Studio%20Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:112:3866)","level":"error","timestamp":"2025-02-24T13:22:15.006Z"}

### Description

I can't select another device.

### Debug Message

Cannot create property 'IDF_TARGET' on string '{"OPENOCD_SCRIPTS":"/Users/pkolt/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts"}'

### Other Steps to Reproduce

1. Open example https://github.com/espressif/esp-idf/tree/master/examples/zigbee/light_sample/HA_on_off_switch
2. Change `esp32` to other device.

[image]

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2025-03-10:
For some reason your `idf.customExtraVars` is defined as string as string where it should be an object.

Open `.vscode/settings.json` and rewrite `idf.customExtraVars` to

```JSON
"idf.customExtraVars": {
 "OPENOCD_SCRIPTS": "/Users/pkolt/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/share/openocd/scripts",
"IDF_TARGET": "esp32h2"
}
```
or use the `Preferences: Open Settings(UI)` and select workspace tab, find `idf.customExtraVars` and add both OPENOCD_SCRIPTS and IDF_TARGET.

This is probably because the settings was handled as string in older extension versions but was later updated to using an object.

brianignacio5, 2025-03-20:
Closing this issue for lack of response.

Comment here if you need more assistance.

closed 2025-03-20 by brianignacio5