espressif/vscode-esp-idf-extension #1460: Problem [BUG] in VSCODE interface while monitor terminal enable and click on Build, Flash and Monitor again it corrupts COM port in Windows as the Monitor is not terminated previously. (IDFGH-14530)

filzek, 2025-03-09:
Problem [BUG] in VSCODE interface while monitor terminal enable and click on Build, Flash and Terminal again.

in Windows:
If the terminal is running and click on the fire icon (Build, Flash and Monitor) the COM port be locked in a background Python task.

Correct Behavior:
Upon clicking in any action that will rewrite the terminal it should CLOSE the monitor so the COM port will be never locked.

brianignacio5, 2025-03-10:
Could you share the troubleshooting information ?
https://docs.espressif.com/projects/vscode-esp-idf-extension/en/latest/troubleshooting.html

brianignacio5, 2025-03-20:
Closing this issue for lack of response. Comment here if you need further assistance.

closed 2025-03-20 by brianignacio5