espressif/vscode-esp-idf-extension #1469: Can't run OpenOCD Server when debugging

Bennyggcc, 2025-03-11:
### OS

Windows

### Operating System version

windows

### Visual Studio Code version

1.91.1

### ESP-IDF version

5.2.3

### Python version

3.10

### Doctor command output

launch.json:
{
  "version": "0.2.0",
  "configurations": [
    {
      "type": "gdbtarget",
      "request": "attach",
      "name": "Eclipse CDT GDB Adapter"
    },
    {
      "type": "espidf",
      "name": "Launch",
      "request": "launch"
    }
  ]
}

setting.json:
{
    "C_Cpp.intelliSenseEngine": "default",
    "idf.espIdfPathWin": "C:\\Users\\35222\\esp\\v5.2.3\\esp-idf",
    "idf.openOcdConfigs": [
        "board/esp32s3-builtin.cfg"
    ],
    "idf.portWin": "COM16",
    "idf.toolsPathWin": "C:\\Users\\35222\\.espressif",
    "idf.flashType": "UART",
    "files.associations": {
        "esp_system.h": "c",
        "esp_bit_defs.h": "c",
        "task.h": "c",
        "esp_flash.h": "c"
    },
    "idf.customExtraVars": {
        "IDF_TARGET": "esp32s3"
    }
}

c_cpp_properties.json:
{
    "configurations": [
        {
            "name": "ESP-IDF",
            "compilerPath": "${config:idf.toolsPathWin}\\tools\\xtensa-esp-elf\\esp-13.2.0_20230928\\xtensa-esp-elf\\bin\\xtensa-esp32s3-elf-gcc.exe",
            "compileCommands": "${config:idf.buildPath}/compile_commands.json",
            "includePath": [
                "${config:idf.espIdfPath}/components/**",
                "${config:idf.espIdfPathWin}/components/**",
                "${workspaceFolder}/**"
            ],
            "browse": {
                "path": [
                    "${config:idf.espIdfPath}/components",
                    "${config:idf.espIdfPathWin}/components",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": true
            }
        }
    ],
    "version": 4
}

### Description

can't run OpenOCD Server when debugging

### Debug Message

[OpenOCD]
Open On-Chip Debugger v0.12.0-esp32-20240821 (2024-08-21-14:45)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html

debug_level: 2

Info : only one transport option; autoselecting 'jtag'

Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000

C:\Users\35222\.espressif\tools\openocd-esp32\v0.12.0-esp32-20240821\openocd-esp32\bin\openocd.exe

Info : Listening on port 666
6 for tcl connections
Info : Listening on port 4444 for telnet connections

Info : esp_usb_jtag: serial (34:85:18:A5:50:84)

Info : esp_usb_jtag: Device f
ound. Base speed 40000KHz, div range 1 to 255

Info : clock speed 40000 kHz

❌ Error: libusb_bulk_write error: LIBUSB_ERROR_TIMEOUT

Info : esp_usb_jtag: serial (34:8
5:18:A5:50:84)

Info : esp_usb_jtag: Device found
. Base speed 40000KHz, div range 1 to 255

❌ Error: libusb_bulk_write error: LIBUSB_ERROR_TIMEOUT

Info : esp_usb_jtag: serial (34:85:18:A5:50:84)

Info : esp_usb_jtag: Dev
ice found. Base speed 40000KHz, div range 1 to 255

❌ Error: libusb_bulk_write error: LIBUSB_ERROR_TIMEOUT

Info : esp_usb_jtag: serial (34:85:18:A5:50:84)

Info : esp_usb_jtag: Device found.
 Base speed 40000KHz, div range 1 to 255

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2025-03-12:
Please review the [OpenOCD troubleshooting FAQ](https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ).

Please add `" idf.openOcdLaunchArgs": ["-c", "esp stub_log on", "-d3:, "-l", "log.txt"]` in your `.vscode/settings.json` to see more information in OpenOCD output.

Any thoughts @erhankur ?

erhankur, 2025-03-12:
Logs will be useful to understand what is going on. As an addition; @Bennyggcc 
- Which devkit is used? Custom or Espressif made?
- Which application is installed? Did you try with a basic hello-world example?

Bennyggcc, 2025-03-13:
> * Which devkit is used? Custom or Espressif made?
> * Which application is installed? Did you try with a basic hello-world example?

Hello, I have added the log file. I used Espressif made devkit: ESP32-S3-DevKitC-1, and I tested on a basic hello-world example.

But this time I use debugger, I got error as follow:

[image]

[image]

[log.txt](https://github.com/user-attachments/files/19224115/log.txt)

erhankur, 2025-03-15:
It seems there is a USB timeout that might be related to the driver. I recommend reinstalling it from https://dl.espressif.com/dl/esp-idf/

Bennyggcc, 2025-03-27:
> It seems there is a USB timeout that might be related to the driver. I recommend reinstalling it from https://dl.espressif.com/dl/esp-idf/

Sorry for the late reply and thanks for your guidance!

closed 2025-03-27 by Bennyggcc