espressif/vscode-esp-idf-extension #1470: [Bug] Unable to open the serial port

shiinakaze, 2025-03-12:
### OS

Windows

### Operating System version

Windows 11

### Visual Studio Code version

1.98.1

### ESP-IDF version

5.4.0

### Python version

3.11.2

### Doctor command output

none

### Description

Shutting down vscode while the monitor is running will occupy the serial port.

[image]

### Debug Message

none

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

jdbaptista, 2025-03-13:
I noticed this as well. The same issue occurs if pressing the 'Kill Terminal' button instead of properly closing the monitor with Ctrl+[ . This seems to be happening in both extension version 1.9.1 and version 1.9.0 with ESP-IDF version 5.3.1. The workaround is to end the monitor before closing the vscode terminal, however I don't know how to recover from the bug if it does occur other than to restart my computer.

shiinakaze, 2025-03-14:
> I noticed this as well. The same issue occurs if pressing the 'Kill Terminal' button instead of properly closing the monitor with Ctrl+[ . This seems to be happening in both extension version 1.9.1 [...]

As you can see from the picture I posted, I have a solution to temporarily solve this problem, as follows:
1. Query the computer `Computer\HKEY_LOCAL_MACHINE\HARDWARE\DEVICEMAP\SERIALCOMM` through Windows Registry editor, and then you can see the name `\Device\Serialx` (x is a number), The data is `COMy` (y is a number).
2. Open the resource Monitor, select CPU > associated handle, and then search `\Device\Serialx`, you can find the corresponding process, and kill it.

daabutts, 2025-03-15:
I have this same problem.  
VSCODE ESP-IDF (v5.2.2) "Monitor Device" button not working as expected (IDFGH-13199) (VSC-1424) #1242 
[https://github.com/espressif/vscode-esp-idf-extension/issues/1242](url)
indicates that this is a problem with Espressif using single quotes instead of double quotes.  If you look at the command line both above and below, just after --make there are three single quotes.  I tried to reformat that command line with better quotes, but I couldn't figure out the proper configuration.  I think this is a relatively new problem.  I suspect a recent update...
`

> PS C:\DevinFiles\Projects\dataRecorder\ESP32-P4-DevKit\examples\reblink>  & 'C:\Users\Devin\.espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe' 'C:\Users\Devin\esp\v5.3.2\esp-idf\tools\idf_monitor.py' -p COM3 -b 115200 --toolchain-prefix riscv32-esp-elf- --make '''C:\Users\Devin\.espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe'' ''C:\Users\Devin\esp\v5.3.2\esp-idf\tools\idf.py''' --target esp32p4 'c:\DevinFiles\Projects\dataRecorder\ESP32-P4-DevKit\examples\reblink\build\reblink.elf'

`
I had 5.4 installed, and went back to 5.3.2 in the hopes that it would work, but it didn't.  There is a reference to a workaround in 1242, but I didn't understand it.  If someone can help explain a good workaround, it would be appreciated.

brianignacio5, 2025-03-19:
I have tested this issue in Windows and MacOS and I was not able to reproduce it.

I made sure to kill the terminal and just close vscode and check if there is a process tagged to `\Device\Serialx` as described by @shiinakaze .

If the issue were quotes I would think that everybody would be affected by this issue. Can you try opening a idf terminal with `ESP-IDF: Open ESP-IDF Terminal` command and run `idf.py monitor -p COM#`. Could you then try both kill the terminal and close all vscode windows and see if the serial port is still being used or not ?

shiinakaze, 2025-03-19:
> I have tested this issue in Windows and MacOS and I was not able to reproduce it.
>
> I made sure to kill the terminal and just close vscode and check if there is a process tagged to `\Device\Serialx` [...]

Thank you for your reply, this problem will occur if you turn on the serial monitor, directly close vscode, and then open vscode again and turn on the serial monitor.

brianignacio5, 2025-03-19:
As I've said, I tested both uses cases: (1) Kill the terminal without Ctrl+] and (2) Close vscode without kill the IDF Monitor.

shiinakaze, 2025-03-19:
> As I've said, I tested both uses cases: (1) Kill the terminal without Ctrl+] and (2) Close vscode without kill the IDF Monitor.

https://github.com/user-attachments/assets/e6ae21ac-91b7-4e8b-a912-1623d2e68e53

brianignacio5, 2025-03-19:
I"m not saying that the issue is not present in your system. I'm just trying to understand why so could you please

> Can you try opening a idf terminal with ESP-IDF: Open ESP-IDF Terminal command and run idf.py monitor -p COM#. Could you then try both kill the terminal and close all vscode windows and see if the serial port is still being used or not ?

shiinakaze, 2025-03-19:
> I"m not saying that the issue is not present in your system. I'm just trying to understand why so could you please
> 
> > Can you try opening a idf terminal with **ESP-IDF: Open ESP-IDF Terminal** command and run idf.py monitor -p COM#. Could you then try both kill the terminal and close all vscode windows and see if the serial port is still being used or not ?
> 
> Running this terminal command directly would tell me if this is a idf.py monitor itself issue or a vscode esp-idf extension issue.

Yes, after killing all terminals and disabling vscode, he's still occupied, as I can see from my video.

brianignacio5, 2025-03-19:
Yes, I understand. What I want you to try is to run `idf.py monitor` in vscode terminal (ESP-IDF: Open ESP-IDF Terminal command) **instead of ** using the vscode monitor command icon.

brianignacio5, 2025-03-19:
Did you start `idf.py monitor`, got a successful connection, close vscode, open vscode again, run `idf.py monitor` again, and see the permission error ?

if you that is the case, that running IDF Monitor directly and still is occupied then this issue might not be related to the vscode extension but of idf_monitor itself not releasing the serial port.

shiinakaze, 2025-03-19:
> Did you start `idf.py monitor`, got a successful connection, close vscode, open vscode again, run `idf.py monitor` again, and see the permission error ?
>
> if you that is the case, that running IDF [...]

Yes, that's correct. So, to whom should the issue be submitted?

brianignacio5, 2025-03-20:
Please take a look at the https://github.com/espressif/esp-idf-monitor repository.

shiinakaze, 2025-03-20:
> Please take a look at the https://github.com/espressif/esp-idf-monitor repository.

ok, thanks.

brianignacio5, 2025-03-20:
Closing this issue. Feel free to comment here if you need more assistance.

closed 2025-03-20 by brianignacio5