espressif/vscode-esp-idf-extension #1477: ESP32-S3 JTAG Debug problem : Unable to open ftdi device (IDFGH-11316)

Sotung96, 2025-03-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi all, I try to use JTAG Debug on ESP32-S3 but it fails. I got some Error on it. I try to google it and follow to fix it but it's still not working. I have use `USB Device Tools` or `Zadig`  to install `libusb` to the com port. I checked the PID & VID is correct in Device Manager. I type `openocd -f board/esp32s3-builtin.cfg` in VS Code Terminal and it is success to open. Do anyone can help me. Thanks. 
```c
OS
Windows

Operating System version
Windows 11

Visual Studio Code version
1.83.1

ESP-IDF version
5.1.1

Python version
3.11.2

==================================================================================
Settings.json :

{
    "idf.adapterTargetName": "esp32s3",
    "idf.portWin": "COM14",
    "idf.flashType": "UART",
    "files.associations": {
        "ethernet_init.h": "c"
    },
    "idf.openOcdConfigs": [
        "board/esp32s3-builtin.cfg"
    ],
}

==================================================================================
Launch.json :

{
    "version": "0.2.0",
    "configurations": [
      {
        "type": "espidf",
        "name": "Launch-name",
        "request": "launch"
      }
    ]
  }

==================================================================================
Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (EC:DA:3B:94:E0:D4)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
[... 888 lines cut ...]
  File "threading.py", line 1038, in _bootstrap_inner
  File "threading.py", line 975, in run
  File "threading.py", line 975, in run
  File "subprocess.py", line 1552, in _readerthread
  File "subprocess.py", line 1552, in _readerthread

[Debug Adapter]
OSError: [Errno 22] Invalid argument
OSError: [Errno 22] Invalid argument

[Debug Adapter]
2023-10-26 12:29:05,709 - Debug Adapter (main) - DEBUG - Closing the Socket
2023-10-26 12:29:05,709 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Closing the Socket\n", "source": {}}, "seq": 43}

2023-10-26 12:29:05,709 - Debug Adapter (main) - DEBUG - Stopping of the Writer thread
2023-10-26 12:29:05,709 - Debug Adapter (main) - DEBUG - Stopping of the Reader thread

[Debug Adapter]
2023-10-26 12:29:05,709 - Debug Adapter (main) - DEBUG - Adapter is Stopped
DEBUG_ADAPTER_STOPPED
2023-10-26 12:29:05,709 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED
2023-10-26 12:29:05,709 - Debug Adapter(Command Processor) - DEBUG - Processed command: disconnect

[Stopped] : ESP-IDF Debug Adapter
```

brianignacio5, 2025-03-14:
Hi @Sotung96  any updates regarding this issue?

We have since the original date of this GitHub issue improved the vscode extension and solved many issues, including a new debug adapter that performs faster that the previous python implementation.

Please let me know if you need more assistance.

brianignacio5, 2025-03-27:
Closing this issue for lack of response. 

Comment here if you need more assistance.

closed 2025-03-27 by brianignacio5