espressif/vscode-esp-idf-extension #1486: Tools Not Found, But Tools is downloaded

WeDoX, 2025-03-26:
### OS

MacOS

### Operating System version

macOS 13.6.1 (22G313)

### Visual Studio Code version

1.98.2

### ESP-IDF version

5.4

### Python version

3.9.6

### Doctor command output

1

### Extension

ESP的工具包都已经下载在：

/Users/zl/.espressif/python_env/idf5.4_py3.9_env/lib/python3.9/site-packages

终端或者

[image]

编辑器老是无法检测到工具：

Error while checking requirement 'pyserial<4,>=3.3'. Package was not found and is required by the application: ruamel.yaml.clib

我的ESP环境：

IDF_PYTHON_ENV_PATH: /Users/zl/.espressif/python_env/idf5.4_py3.9_env

Python interpreter used: /Users/zl/.espressif/python_env/idf5.4_py3.9_env/bin/python

该如何排查这个问题

[image]

### Description

1

### Debug Message

1

### Other Steps to Reproduce

1

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2025-03-27:
You are sharing a command line environment. Are you using the esp-idf vscode extension ?

If you are just using the CLI, you should open the issue in the https://github.com/espressif/esp-idf repository.

Closing this issue as it seems unrelated to esp-idf vscode extension and the issue report doesn't include vscode information.

closed 2025-03-27 by brianignacio5

WeDoX, 2025-03-27:
Ok, this is my mistake, thank you for your answer. Can you delete this issue?

> You are sharing a command line environment. Are you using the esp-idf vscode extension ?
>
> If you are just using the CLI, you should open the issue in the https://github.com/espressif/esp-idf [...]