espressif/vscode-esp-idf-extension #1487: 2+ second delay when changing COM port or IDF version

rtheil-growlink, 2025-03-26:
### OS

Windows

### Operating System version

WIndows 10 and 11

### Visual Studio Code version

1.98.2

### ESP-IDF version

4.3.7 & 4.4.8

### Python version

3.8.7

### Doctor command output

N/A

### Extension

{"message":"non zero exit code null\r\n\r\nesptool.py v3.3.4-dev\r\nSerial port COM3\r\nConnecting.........","stack":"Error: non zero exit code null\r\n\r\nesptool.py v3.3.4-dev\r\nSerial port COM3\r\nConnecting.........\n\tat ChildProcess.<anonymous> (\\dist\\extension.js:2:1432639)\n\tat ChildProcess.emit (node:events:518:28)\n\tat ChildProcess._handle.onexit (node:internal/child_process:293:12)","level":"error","timestamp":"2025-03-26T14:29:56.218Z"}
{"message":"non zero exit code null\r\n\r\nesptool.py v3.3.4-dev\r\nSerial port COM1\r\nConnecting.........","stack":"Error: non zero exit code null\r\n\r\nesptool.py v3.3.4-dev\r\nSerial port COM1\r\nConnecting.........\n\tat ChildProcess.<anonymous> (\\dist\\extension.js:2:1432639)\n\tat ChildProcess.emit (node:events:518:28)\n\tat ChildProcess._handle.onexit (node:internal/child_process:293:12)","level":"error","timestamp":"2025-03-26T14:29:56.219Z"}
{"message":"non zero exit code null\r\n\r\nesptool.py v3.3.4-dev\r\nSerial port COM4\r\nConnecting.....\r\nDetecting chip type... Unsupported detection protocol, switching and trying again...\r\nConnecting....\r\nDetecting chip type...","stack":"Error: non zero exit code null\r\n\r\nesptool.py v3.3.4-dev\r\nSerial port COM4\r\nConnecting.....\r\nDetecting chip type... Unsupported detection protocol, switching and trying again...\r\nConnecting....\r\nDetecting chip type...\n\tat ChildProcess.<anonymous> (\\dist\\extension.js:2:1432639)\n\tat ChildProcess.emit (node:events:518:28)\n\tat ChildProcess._handle.onexit (node:internal/child_process:293:12)","level":"error","timestamp":"2025-03-26T14:29:56.234Z"}
{"message":"non zero exit code null\r\n\r\nesptool.py v3.3.4-dev\r\nSerial port COM5\r\nConnecting.....\r\nDetecting chip type... Unsupported detection protocol, switching and trying again...\r\nConnecting....\r\nDetecting chip type...","stack":"Error: non zero exit code null\r\n\r\nesptool.py v3.3.4-dev\r\nSerial port COM5\r\nConnecting.....\r\nDetecting chip type... Unsupported detection protocol, switching and trying again...\r\nConnecting....\r\nDetecting chip type...\n\tat ChildProcess.<anonymous> (\\dist\\extension.js:2:1432639)\n\tat ChildProcess.emit (node:events:518:28)\n\tat ChildProcess._handle.onexit (node:internal/child_process:293:12)","level":"error","timestamp":"2025-03-26T14:29:56.234Z"}
{"message":"non zero exit code null\r\n\r\nesptool.py v3.3.4-dev\r\nSerial port COM6\r\nConnecting......\r\nDetecting chip type... Unsupported detection protocol, switching and trying again...\r\nConnecting....\r\nDetecting chip type...","stack":"Error: non zero exit code null\r\n\r\nesptool.py v3.3.4-dev\r\nSerial port COM6\r\nConnecting......\r\nDetecting chip type... Unsupported detection protocol, switching and trying again...\r\nConnecting....\r\nDetecting chip type...\n\tat ChildProcess.<anonymous> (\\dist\\extension.js:2:1432639)\n\tat ChildProcess.emit (node:events:518:28)\n\tat ChildProcess._handle.onexit (node:internal/child_process:293:12)","level":"error","timestamp":"2025-03-26T14:29:56.235Z"}
{"message":"non zero exit code null\r\n\r\nesptool.py v3.3.4-dev\r\nSerial port COM3\r\nConnecting....\r\nDetecting chip type... Unsupported detection protocol, switching and trying again...\r\nConnecting......\r\nDetecting chip type...","stack":"Error: non zero exit code null\r\n\r\nesptool.py v3.3.4-dev\r\nSerial port COM3\r\nConnecting....\r\nDetecting chip type... Unsupported detection protocol, switching and trying again...\r\nConnecting......\r\nDetecting chip type...\n\tat ChildProcess.<anonymous> (\\dist\\extension.js:2:1432639)\n\tat ChildProcess.emit (node:events:518:28)\n\tat ChildProcess._handle.onexit (node:internal/child_process:293:12)","level":"error","timestamp":"2025-03-26T14:30:10.330Z"}
{"message":"non zero exit code null\r\n\r\nesptool.py v3.3.4-dev\r\nSerial port COM1\r\nConnecting.........","stack":"Error: non zero exit code null\r\n\r\nesptool.py v3.3.4-dev\r\nSerial port COM1\r\nConnecting.........\n\tat ChildProcess.<anonymous> (\\dist\\extension.js:2:1432639)\n\tat ChildProcess.emit (node:events:518:28)\n\tat ChildProcess._handle.onexit (node:internal/child_process:293:12)","level":"error","timestamp":"2025-03-26T14:30:10.331Z"}
{"message":"non zero exit code null\r\n\r\nesptool.py v3.3.4-dev\r\nSerial port COM4\r\nConnecting.....\r\nDetecting chip type... Unsupported detection protocol, switching and trying again...\r\nConnecting.....\r\nDetecting chip type...","stack":"Error: non zero exit code null\r\n\r\nesptool.py v3.3.4-dev\r\nSerial port COM4\r\nConnecting.....\r\nDetecting chip type... Unsupported detection protocol, switching and trying again...\r\nConnecting.....\r\nDetecting chip type...\n\tat ChildProcess.<anonymous> (\\dist\\extension.js:2:1432639)\n\tat ChildProcess.emit (node:events:518:28)\n\tat ChildProcess._handle.onexit (node:internal/child_process:293:12)","level":"error","timestamp":"2025-03-26T14:30:10.331Z"}
{"message":"non zero exit code null\r\n\r\nesptool.py v3.3.4-dev\r\nSerial port COM6\r\nConnecting.....\r\nDetecting chip type... Unsupported detection protocol, switching and trying again...\r\nConnecting....\r\nDetecting chip type...","stack":"Error: non zero exit code null\r\n\r\nesptool.py v3.3.4-dev\r\nSerial port COM6\r\nConnecting.....\r\nDetecting chip type... Unsupported detection protocol, switching and trying again...\r\nConnecting....\r\nDetecting chip type...\n\tat ChildProcess.<anonymous> (\\dist\\extension.js:2:1432639)\n\tat ChildProcess.emit (node:events:518:28)\n\tat ChildProcess._handle.onexit (node:internal/child_process:293:12)","level":"error","timestamp":"2025-03-26T14:30:10.331Z"}
{"message":"non zero exit code null\r\n\r\nesptool.py v3.3.4-dev\r\nSerial port COM6\r\nConnecting.......\r\nDetecting chip type... Unsupported detection protocol, switching and trying again...\r\nConnecting.......","stack":"Error: non zero exit code null\r\n\r\nesptool.py v3.3.4-dev\r\nSerial port COM6\r\nConnecting.......\r\nDetecting chip type... Unsupported detection protocol, switching and trying again...\r\nConnecting.......\n\tat ChildProcess.<anonymous> (\\dist\\extension.js:2:1432639)\n\tat ChildProcess.emit (node:events:518:28)\n\tat ChildProcess._handle.onexit (node:internal/child_process:293:12)","level":"error","timestamp":"2025-03-26T14:31:03.258Z"}
{"message":"non zero exit code null\r\n\r\nesptool.py v3.3.4-dev\r\nSerial port COM1\r\nConnecting..........","stack":"Error: non zero exit code null\r\n\r\nesptool.py v3.3.4-dev\r\nSerial port COM1\r\nConnecting..........\n\tat ChildProcess.<anonymous> (\\dist\\extension.js:2:1432639)\n\tat ChildProcess.emit (node:events:518:28)\n\tat ChildProcess._handle.onexit (node:internal/child_process:293:12)","level":"error","timestamp":"2025-03-26T14:31:03.258Z"}
{"message":"Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat \\dist\\extension.js:2:1436281\n\tat ChildProcess.exithandler (node:child_process:415:7)\n\tat ChildProcess.emit (node:events:518:28)\n\tat maybeClose (node:internal/child_process:1104:16)\n\tat Socket.<anonymous> (node:internal/child_process:456:11)\n\tat Socket.emit (node:events:518:28)\n\tat Pipe.<anonymous> (node:net:343:12)","level":"error","timestamp":"2025-03-26T14:31:17.891Z"}

### Description

I've tested this on a fresh install of the OS, VSCode and the plugin.

Clicking the COM port selection button results in a 2 second delay before the com port selection menu comes up. This was not present in 1.7 or 1.8 version of the espressif plugin.

In addition, there is a longer delay (3+ seconds) when clicking the IDF selection button, before the IDF versions list comes up. 

I have 5 COM ports and 2 IDF versions installed, so it's not like either list is huge, and I'm not using COM port filtering.

### Debug Message

N/A

### Other Steps to Reproduce

Same behavior on my laptop with windows 10, same on my machine before a fresh install of Windows 11. Same on new install of Windows 11.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2025-03-27:
Hi @rtheil-growlink  the reason is that a feature to identify the chip and show it in the dropdown has been added.

To disable it, set `"idf.enableSerialPortChipIdRequest": false` and `"idf.useSerialPortVendorProductFilter": false` in your project's .vscode/settings.json. This should remove the delay you are observing.

The feature was implemented in https://github.com/espressif/vscode-esp-idf-extension/pull/1276

rtheil-growlink, 2025-03-27:
Thanks, Brian! Well, I added those two and restarted vscode, but that made no difference. Also seems like that would have no effect on the IDF selection. Not sure why those options are a slightly different color

[image]

brianignacio5, 2025-03-28:
Which version of the extension are you using ? The doctor command is missing from issue report.

rtheil-growlink, 2025-03-28:
Sweet, that fixes it! Thank you.

Still a 2-5 second delay on selecting the IDF version though. Can't figure out why sometimes it's only 2 and others it's almost 5 seconds. Only IDF 4.3.7 and 4.4.8 installed. And actually... I realize that extension v1.8.x does have this IDF selection delay. I can't tell at this point if 1.9 makes this worse or if they are randomly the same.

brianignacio5, 2025-03-31:
Yes the select version does have a delay because it is validating all ESP-IDF tools and python packages are valid.

We are working in integrating with https://github.com/espressif/idf-im-ui our new cross platform ESP-IDF Installer, and we hope to remove validation from IDE altogether. This will take some time to do so, apologies for this.

I'll close this since the original issue is fixed. I'll give a try to see if I can optimize version selection delay before IDF-IM integration.