espressif/vscode-esp-idf-extension #1491: Unable to debug: GDB failure with error dyld[9651]: missing symbol called

drriguz, 2025-03-29:
### OS

MacOS

### Operating System version

12.7.6

### Visual Studio Code version

1.98.2

### ESP-IDF version

5.5

### Python version

3.12.2

### Doctor command output

[report.txt](https://github.com/user-attachments/files/19521452/report.txt)

### Extension

[esp_idf_vsc_ext.log](https://github.com/user-attachments/files/19521459/esp_idf_vsc_ext.log)

### Description

Cannot start debug and got the following error:

```
GNU gdb (esp-gdb) 15.2_20241112
Copyright (C) 2024 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-apple-darwin21.1 --target=xtensa-esp-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
dyld[9651]: missing symbol called

Fatal signal: Abort trap: 6
----- Backtrace -----
Backtrace unavailable
---------------------
A fatal error internal to GDB has been detected, further
debugging is not possible.  GDB will now terminate.

This is a bug, please report it.  For instructions, see:
<https://www.gnu.org/software/gdb/bugs/>.
```

A similar issus fond on stackoverflow:  https://stackoverflow.com/questions/78855046/esp32-gdb-debugging-error-on-m1-mac-dyld-missing-symbol-called

### Debug Message

N/A

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2025-03-31:
Could you share the openOCD output. Click menu `View` -> `Output` -> `OpenOCD`.  ?

brianignacio5, 2025-04-14:
Closing this issue for lack of user response. 

Comment here if you need more assistance.

closed 2025-04-14 by brianignacio5