espressif/vscode-esp-idf-extension #1497: IDF TARGET changes when I change project configuration

citisend-app-libraries, 2025-04-02:
### OS

Windows

### Operating System version

Windows11

### Visual Studio Code version

1.97.2

### ESP-IDF version

4.4.7

### Python version

3.8.7

### Doctor command output

This happens since the last update of vscode extension. If I start with this:

[image]

If I change the project configuration and delete the sdkconfig file (so the new configuration is applied):

[image]

The target reverts to ESP32. This happens the moment I delete the sdkconfig file, not change the configuration

So then I have to change the target every time I change project configuration. Except I never remeber so I have to compile two times.

I have tried setting:
```
CONFIG_IDF_TARGET="esp32s3"
CONFIG_IDF_TARGET_ESP32S3=y
```

On the sdkconfig.defaults file but it doesn't work still

### Extension

V1.9.1

### Description

.

### Debug Message

.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2025-04-03:
How do you select a ESP target ? Do you use the `ESP-IDF: Set Espressif Device Target` ?

The current IDF_TARGET is defined in the sdkconfig file. When you delete it, the default target is `esp32` if you didn't use the command from before. Why? The `ESP-IDF: Set Espressif Device Target` command also set `idf.customExtraVars: { "IDF_TARGET": "selectedTarget"`. And that would be used for later command to set the target to generated the sdkconfig file.

Since you didn't share the doctor command output, I can't see your current settings

citisend-app-libraries, 2025-04-03:
> How do you select a ESP target ? Do you use the `ESP-IDF: Set Espressif Device Target` ?
>
> The current IDF_TARGET is defined in the sdkconfig file. When you delete it, the default target is `esp32` [...]

Hi,,

Yes, I use `ESP-IDF: Set Espressif Device Target`

Moreover, `idf.customExtraVars: { "IDF_TARGET": "selectedTarget"` is set to esp32s3. But when I delete sdkconfig it still defaults to esp32

See here:

[image]

brianignacio5, 2025-04-03:
Thanks for your reply @citisend-app-libraries I will take a look at the code that updates the target in status bar when sdkconfig file is deleted.

After you delete the sdkconfig file, status bar changes to esp32. But if you try to build, does it build with the correct target (esp32s3 from idf.customExtraVars) ?

citisend-app-libraries, 2025-04-03:
> Thanks for your reply [@citisend-app-libraries](https://github.com/citisend-app-libraries) I will take a look at the code that updates the target in status bar when sdkconfig file is deleted.
>
> After [...]

No, it builds with the wrong target. That's why I noticed the issue :)

brianignacio5, 2025-04-03:
I think I found a solution with #1498 

Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/2873862444.zip)

You can test these changes by installing this VSIX by click menu View -> Command Palette..., type Install from VSIX and then select downloaded esp-idf-extension.vsix file to install the extension.

PR #1498 (2025-04-21): [VSC-1640] profiles env should merge customExtraVars with profile env
## Description

When the user is working with multiple profiles, the profile env should merge with `idf.customExtraVars` instead of fully replace it. This is necessary to support IDF_TARGET when sdkconfig file doesn't exists.

Fixes #1497
Fixes #1302

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Create a project with multiple profile configurat [... cut]
closed 2025-04-21 by brianignacio5