espressif/vscode-esp-idf-extension #1500: ESP-IDF Extension for VS Code does not install

Isarhamster, 2025-04-04:
### OS

Windows

### Operating System version

Windows 10

### Visual Studio Code version

1.98.2

### ESP-IDF version

5.4.2

### Python version

3.11.2 (I guess from the error message)

### Doctor command output

I guess, this is not helpful, as the installation itself fails.

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS undefined undefined undefined 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 undefined 
System environment variable PYTHON 
 undefined 
Visual Studio Code version undefined 
Visual Studio Code language undefined 
Visual Studio Code shell undefined 
ESP-IDF Extension version undefined 
Workspace folder c:<HOMEPATH>\Documents\Projekte\EPT\1MBN_Euro NCAP Targets (EPT)\EPT-Software\EPT Master 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) undefined
ESP-IDF Path (idf.espIdfPath) undefined
ESP-MDF Path (idf.espMdfPath) undefined
ESP-Matter Path (idf.espMatterPath) undefined
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) undefined
Custom extra paths undefined
System python Path (idf.pythonInstallPath) undefined
Virtual environment Python path (computed) undefined
Serial port (idf.port) undefined
OpenOCD Configs (idf.openOcdConfigs) undefined
ESP-IDF Tools Path (idf.toolsPath) undefined
Git Path (idf.gitPath) undefined
Notification Mode (idf.notificationMode) undefined
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) undefined
Access to ESP-IDF Path (idf.espIdfPath) undefined
Access to ESP-MDF Path (idf.espMdfPath) undefined
Access to ESP-Matter Path (idf.espMatterPath) undefined
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) undefined
Access to ESP-IDF Custom extra paths
Access to System python Path (idf.pythonInstallPath) undefined
Access to Virtual environment Python path (computed) undefined
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) undefined
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path undefined
Spaces in ESP-ADF Path (idf.espAdfPath) undefined
Spaces in ESP-IDF Path (idf.espIdfPath) undefined
Spaces in ESP-MDF Path (idf.espMdfPath) undefined
Spaces in ESP-Matter Path (idf.espMatterPath) undefined
Spaces in ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) undefined
Spaces in ESP-IDF Custom extra paths
Spaces in System python Path (idf.pythonInstallPath) undefined
Spaces in Virtual environment Python path (computed) undefined
Spaces in ESP-IDF Tools Path (idf.toolsPath) undefined
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version undefined
ESP-IDF version undefined
Python version undefined
Python's pip version undefined
-------------------------------------------------- Project configuration settings ----------------------------------------------------------
-------------------------------------------------- Python packages in Virtual environment Python path (computed) ---------------------------
Python packages undefined
---------------------------------------------------- Check ESP-IDF python requirements.txt -------------------------------------------------
Check ESP-IDF Python packages undefined
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages undefined
----------------------------------------------------------- Latest error -----------------------------------------------------------------
{
    "errno": -4058,
    "code": "ENOENT",
    "syscall": "spawn C:\\Espressif\\tools\\idf-python\\3.11.2\\python.exe",
    "path": "C:\\Espressif\\tools\\idf-python\\3.11.2\\python.exe",
    "spawnargs": [
        "-c",
        "import sys; print('{}.{}'.format(sys.version_info.major, sys.version_info.minor))"
    ],
    "cmd": "C:\\Espressif\\tools\\idf-python\\3.11.2\\python.exe -c import sys; print('{}.{}'.format(sys.version_info.major, sys.version_info.minor))"
}

### Extension

Log is not existant

### Description

Upon configuring ESP-IDF, I get

> "C:\Espressif\tools\idf-python\3.11.2\python.exe -m pip" is not valid. (ERROR_INVALID_PIP)

The python interpreter is located at the mentioned path.
Calling pip from commandline gives
```
pip --version
pip 24.0 from C:\Python312\Lib\site-packages\pip (python 3.12)
```
I guess, that ESP-IDF comes without pip and grabs the wrong one from somewhere on my computer.
Probably ESP-IDF would better user my Python312 but I do not know how to change this.

### Debug Message

Done

### Other Steps to Reproduce

Platformio works with ESP32, I just want to use espsecure and esptool in addition to what platformio gives

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2025-04-07:
This could happen if `C:\Espressif\tools\idf-python\3.11.2\python.exe` was installed from Windows IDF Installer instead of being downloaded from the vscode extension. The IDF installer doesn't come with pip but the one downloaded from vscode extension setup wizard does.

I would recommend to use another location for IDF_TOOLS_PATH (currently C:\Espressif) to install ESP-IDF for vscode extension or delete C:\Espressif\tools\idf-python and idf-python from C:\Espressif\dist to make sure the extension downloads idf-python.

brianignacio5, 2025-04-18:
Closing this issue for lack of response from user.

Comment here if you need more assistance.

closed 2025-04-18 by brianignacio5

crushonme, 2025-05-08:
Run below command in CMD resolved the error.
`C:\Espressif\tools\idf-python\3.11.2\python.exe -m ensurepip`

For more details, please refer to [Failed-to-configure-ESP-IDF-in-VSCode](https://crushonme.github.io/2025/05/08/Failed-to-configure-ESP-IDF-in-VSCode/)