espressif/vscode-esp-idf-extension #1501: extension configure can't get IDF_TOOLS_PATH and Python virtual environment in docker

shishirong, 2025-04-07:
### OS

Windows

### Operating System version

Windows 10 + wsl2 + docker

### ESP-IDF version

1.9.1

### Python version

3.10.12

### Doctor command output

none

### Description

It is fixed but It dosn't works in my work. [ref](https://github.com/espressif/vscode-esp-idf-extension/issues/1435)
devcontainer.json
```json
{
	"name": "ESP-IDF QEMU",
	"build": {
		"dockerfile": "Dockerfile",
		"args": {
			"IDF_CLONE_BRANCH_OR_TAG": "release/v5.3",
			"IDF_CLONE_SHALLOW": "true",
			"IDF_INSTALL_TARGETS": "esp32"
		}
	},
	"customizations": {
		"vscode": {
			"settings": {
				"terminal.integrated.defaultProfile.linux": "bash",
				"idf.espIdfPath": "/opt/esp/idf",
				"idf.toolsPath": "/opt/esp",
				"idf.gitPath": "/usr/bin/git"
			},
			"extensions": [
				"espressif.esp-idf-extension",
				"espressif.esp-idf-web"
			]
		}
	},
	"runArgs": [
		"--privileged"
	]
}
```

Dockerfile:
```dockerfile
ARG DOCKER_TAG=release-v5.3
FROM espressif/idf:${DOCKER_TAG}

ENV LC_ALL=C.UTF-8
ENV LANG=C.UTF-8

RUN apt-get update -y && apt-get install udev -y

RUN echo "source /opt/esp/idf/export.sh > /dev/null 2>&1" >> ~/.bashrc

ENTRYPOINT [ "/opt/esp/entrypoint.sh" ]

CMD ["/bin/bash", "-c"]
```

[image]

### Debug Message

none

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2025-04-07:
You don't need to use the setup wizard in the Docker container.

Why doesn't work ? Could you share the doctor command output ?

Do you have settings from Windows in the `.vscode/settings.json` ? These settings would override any settings defined in `.devcontainer.json` and that would make it not to work.

shishirong, 2025-04-07:
you are right, build successful when I don't to use the setup wizard.
but vscode-esp-idf-extension functions is not  integrity that excute command fail when I use "ESP-IDF:EXPLOER COMMANDS"  like "Show Examples" or "New Project Wizard" ...

tips:
```log
No ESP-IDF Setups found

or 

No ESP-IDF frameworks found
```

brianignacio5, 2025-04-07:
Thank you @shishirong  That is indeed an issue we didn't consider. I'll take a look and try to come up with a fix.

brianignacio5, 2025-04-16:
I've made a fix in #1513 Please take a look @shishirong 

Updating the extension docker is a bit weird, look at steps to reproduce in the PR to follow how to install the vsix installer.

You can test these changes by installing this VSIX by click menu **View** -> **Command Palette...**, type **Install from VSIX** and then select downloaded **esp-idf-extension.vsix** file to install the extension.
* [esp-idf-extension.vsix.zip](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/2954495143.zip)

shishirong, 2025-04-16:
yes it is works, thanks you . @brianignacio5 

but `select current esp-idf version`  tips "No ESP-IDF Setups found"

`show examples` just show esp-adf examples if I installed adf , absence esp-idf examples.
[image]

`new project wizard` list is empty
[image]

brianignacio5, 2025-04-17:
I've tested on docker for Show Examples and New Project Wizard and they seem to work.

I can think that if there are not shown, it means the setup might not be correct. Maybe some conflict settings in the project .vscode/settings.json ? Could you share the doctor command output ?

shishirong, 2025-04-17:
it is works when I tried in a new docker container.

logs in my old docker container:
.vscode/settings.json:
```json
{
    "idf.pythonInstallPath": "/usr/bin/python"
}
```

doctor command output:
```log
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 6.6.75.1-microsoft-standard-WSL2 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 /vscode/cursor-server/bin/linux-x64/61e99179e4080fecf9d8b92c6e2e3e00fbfb53f0/bin/remote-cli:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:<HOMEPATH>/.local/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code Remote name attached-container 
Visual Studio Code version 1.96.2 
Visual Studio Code language zh-cn 
Visual Studio Code shell /usr/bin/bash 
ESP-IDF Extension version 1.9.1 
Workspace folder /workspaces/RopeV2.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) /opt/esp/esp-adf
ESP-IDF Path (idf.espIdfPath) /opt/esp/idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths /opt/esp/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin:/opt/esp/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin:/opt/esp/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin:/opt/esp/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin:/opt/esp/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin:/opt/esp/tools/cmake/3.30.2/bin:/opt/esp/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/bin:/opt/esp/tools/ninja/1.12.1:/opt/esp/tools/esp-rom-elfs/20240305
ESP-IDF extra vars
    OPENOCD_SCRIPTS: /opt/esp/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/share/openocd/scripts
    ESP_ROM_ELF_DIR: /opt/esp/tools/esp-rom-elfs/20240305/
    IDF_PYTHON_ENV_PATH: /opt/esp/python_env/idf5.3_py3.12_env
    ESP_IDF_VERSION: 5.3
    PATH: /opt/esp/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin:/opt/esp/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin:/opt/esp/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin:/opt/esp/tools/cmake/3.30.2/bin:/opt/esp/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/bin:/opt/esp/tools/qemu-xtensa/esp_develop_9.0.0_20240606/qemu/bin:/opt/esp/python_env/idf5.3_py3.12_env/bin:/opt/esp/idf/tools:$PATH
    IDF_DEACTIVATE_FILE_PATH: /tmp/tmpbjq1bnsgidf_304
User extra vars (idf.customExtraVars)
System python Path (idf.pythonInstallPath) /usr/bin/python
Virtual environment Python path (computed) /opt/esp/python_env/idf5.3_py3.12_env/bin/python
Serial port (idf.port) /dev/ttyACM1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) /opt/esp
Git Path (idf.gitPath) /usr/bin/git
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) true
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to /opt/esp/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin: true
Access to /opt/esp/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin: false
Access to /opt/esp/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin: true
Access to /opt/esp/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin: false
Access to /opt/esp/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin: true
Access to /opt/esp/tools/cmake/3.30.2/bin: true
Access to /opt/esp/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/bin: true
Access to /opt/esp/tools/ninja/1.12.1: false
Access to /opt/esp/tools/esp-rom-elfs/20240305: true
Access to System python Path (idf.pythonInstallPath) true
Access to Virtual environment Python path (computed) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
[... 160 lines cut ...]
{"message":"Error checking riscv32-esp-elf version","stack":"Error: spawn riscv32-esp-elf-gcc ENOENT\n\tat ChildProcess._handle.onexit (node:internal/child_process:285:19)\n\tat onErrorNT (node:internal/child_process:483:16)\n\tat process.processTicksAndRejections (node:internal/process/task_queues:82:21)","category":"IdfToolsManager checkBinariesVersion","level":"error","timestamp":"2025-04-17T02:05:05.620Z"}
{"message":"Cannot access filePath: ${env:MDF_PATH}","stack":"Error: ENOENT: no such file or directory, access '${env:MDF_PATH}'\n\tat Object.accessSync (node:fs:246:11)\n\tat Fe (/home/vscode/.cursor-server/extensions/espressif.esp-idf-extension-1.9.1/dist/extension.js:2:1432907)\n\tat /home/vscode/.cursor-server/extensions/espressif.esp-idf-extension-1.9.1/dist/extension.js:2:1670363\n\tat Ia (/home/vscode/.cursor-server/extensions/espressif.esp-idf-extension-1.9.1/dist/extension.js:2:1671354 [... cut ...]
{"message":"Error checking riscv32-esp-elf version","stack":"Error: spawn riscv32-esp-elf-gcc ENOENT\n\tat ChildProcess._handle.onexit (node:internal/child_process:285:19)\n\tat onErrorNT (node:internal/child_process:483:16)\n\tat process.processTicksAndRejections (node:internal/process/task_queues:82:21)","category":"IdfToolsManager checkBinariesVersion","level":"error","timestamp":"2025-04-17T02:05:43.333Z"}

--------------------------------------------------------------------------------------------------------------------------------------------

```

brianignacio5, 2025-04-23:
It seems to me that the old docker maybe is still using the current extension from marketplace and not the one from the PR vsix. I would suggest to delete the `~/.vscode/extension/espressif-**` directory to make sure old extension is deleted before installing the vsix from PR in the container.

If it works in the new container, then after my colleagues review the PR and merge it will close this issue.

PR #1513 (2025-05-19): [VSC-1642] add current setup to list of setups in examples new project wizards
## Description

Fix `ESP-IDF: Show Examples` and `ESP-IDF: New Project` command to show current idf setup from Docker environment for example.

Fixes #1501

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Open an ESP-IDF project in a docker container with `Reopen in container` command.  Make sure that the project doesn't have idf.espIdfPath or other settings from host machine to avo [... cut]
closed 2025-04-23 by shishirong