espressif/vscode-esp-idf-extension #1504: Breakpoints created from Breakpoint Tab do not show in text editor

bryghtlabs-richard, 2025-04-11:
### OS

Windows

### Operating System version

Windows 11

### Visual Studio Code version

1.99.0

### ESP-IDF version

5.1.2

### Python version

3.11.2

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.26100 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v12.8\bin;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v12.8\libnvvp;C:\Program Files\Python311\Scripts\;C:\Program Files\Python311\;C:\Program Files\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\dotnet\;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files\Git\cmd;C:\Program Files\PuTTY\;C:\Program Files (x86)\Nordic Semiconductor\nrf5x\bin\;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:\Program Files\nodejs\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin\;C:\Program Files\IVI Foundation\VISA\Win64\Bin\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin;C:\Program Files (x86)\IVI Foundation\IVI\Bin\;C:\Program Files\IVI Foundation\IVI\Bin\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\NVIDIA Corporation\Nsight Compute 2025.1.1\;C:\Program Files\NVIDIA Corporation\NVIDIA app\NvDLISR;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\GitHubDesktop\bin;C:\Program Files\mitmproxy\bin;C:\Program Files\heroku\bin;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\AppData\Roaming\npm; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.99.0 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.9.1 
Workspace folder c:\Projects\ChessUpTwo\ESP32 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:/Espressif/frameworks/esp-idf-v5.1.2/
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths C:\Espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin;C:\Espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin;C:\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin;C:\Espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin;C:\Espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin;C:\Espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\Espressif\tools\cmake\3.24.0\bin;C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin;C:\Espressif\tools\ninja\1.10.2;C:\Espressif\tools\idf-exe\1.0.3;C:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:\Espressif\tools\esp-rom-elfs\20230320
ESP-IDF extra vars
    OPENOCD_SCRIPTS: C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\share\openocd\scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:\Espressif\tools\esp-rom-elfs\20230320\
    IDF_PYTHON_ENV_PATH: C:\Espressif\python_env\idf5.1_py3.11_env
    ESP_IDF_VERSION: 5.1
    PATH: C:\Espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin;C:\Espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin;C:\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin;C:\Espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin;C:\Espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin;C:\Espressif\tools\esp-clang\15.0.0-23786128ae\esp-clang\bin;C:\Espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\Espressif\tools\cmake\3.24.0\bin;C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin;C:\Espressif\tools\ninja\1.10.2\;C:\Espressif\tools\idf-exe\1.0.3\;C:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:\Espressif\python_env\idf5.1_py3.11_env\Scripts;C:/Espressif/frameworks/esp-idf-v5.1.2/tools;%PATH%
    IDF_DEACTIVATE_FILE_PATH: C:\Users\RICHAR~1\AppData\Local\Temp\tmpjorp8pscidf_2880
User extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230208/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:\Espressif\tools\esp-rom-elfs\20230320/
System python Path (idf.pythonInstallPath) C:\Espressif\tools\idf-python\3.11.2\python.exe
Virtual environment Python path (computed) C:\Espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM5
OpenOCD Configs (idf.openOcdConfigs) board/esp32s3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Espressif
Git Path (idf.gitPath) C:\Espressif\tools\idf-git\2.39.2\cmd\git.exe
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to C:\Espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin: true
Access to C:\Espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin: true
[... 259 lines cut ...]
{"message":"Cannot access filePath: ${env:MDF_PATH}","stack":"Error: ENOENT: no such file or directory, access 'C:\\Users\\RichardAllen\\AppData\\Local\\Programs\\Microsoft VS Code\\${env:MDF_PATH}'\n\tat Object.accessSync (node:fs:247:11)\n\tat t.accessSync (node:electron/js2c/node_init:2:9332)\n\tat Fe (c:\\Users\\RichardAllen\\.vscode\\extensions\\espressif.esp-idf-extension-1.9.1\\dist\\extension.js:2:1432827)\n\tat c:\\Users\\RichardAllen\\.vscode\\extensions\\espressif.esp-idf-extension-1.9.1\\dist\\extension.js:2:1668796\n\tat Pa (c:\\Users\\RichardAllen\\.vscode\\extensions\\espressif.esp-idf-extension-1.9.1\\dist\\extension.js:2:1669787)\n\tat process.processTicksAndRejections (node:internal/process/task_queues:95:5)\n\tat async c:\\Users\\RichardAllen\\.vscode\\extensions\\espressif.esp-idf-extension-1.9.1\\dist\\extension.js:2:1893256","level":"error","timestamp":"2025-04-11T19:16:37.335Z"}
{"message":"Cannot access filePath: ${env:ESP_MATTER_PATH}","stack":"Error: ENOENT: no such file or directory, access 'C:\\Users\\RichardAllen\\AppData\\Local\\Programs\\Microsoft VS Code\\${env:ESP_MATTER_PATH}'\n\tat Object.accessSync (node:fs:247:11)\n\tat t.accessSync (node:electron/js2c/node_init:2:9332)\n\tat Fe (c:\\Users\\RichardAllen\\.vscode\\extensions\\espressif.esp-idf-extension-1.9.1\\dist\\extension.js:2:1432827)\n\tat c:\\Users\\RichardAllen\\.vscode\\extensions\\espressif.esp-idf-extension-1.9.1\\dist\\extension.js:2:1668888\n\tat Pa (c:\\Users\\RichardAllen\\.vscode\\extensions\\espressif.esp-idf-extension-1.9.1\\dist\\extension.js:2:1669787)\n\tat process.processTicksAndRejections (node:internal/process/task_queues:95:5)\n\tat async c:\\Users\\RichardAllen\\.vscode\\extensions\\espressif.esp-idf-extension-1.9.1\\dist\\extension.js:2:1893256","level":"error","timestamp":"2025-04-11T19:16:37.335Z"}
{"message":"Cannot access filePath: ${env:HOMEKIT_PATH}","stack":"Error: ENOENT: no such file or directory, access 'C:\\Users\\RichardAllen\\AppData\\Local\\Programs\\Microsoft VS Code\\${env:HOMEKIT_PATH}'\n\tat Object.accessSync (node:fs:247:11)\n\tat t.accessSync (node:electron/js2c/node_init:2:9332)\n\tat Fe (c:\\Users\\RichardAllen\\.vscode\\extensions\\espressif.esp-idf-extension-1.9.1\\dist\\extension.js:2:1432827)\n\tat c:\\Users\\RichardAllen\\.vscode\\extensions\\espressif.esp-idf [... cut ...] or","timestamp":"2025-04-11T19:16:37.336Z"}

### Description

Breakpoints created by "add function breakpoint" do not show in the text editor. Ex: there is a breakpoint on processPlacement(),

[image]

But there is no breakpoint shown here:

[image]

It should look like this instead:

[image]

### Debug Message

.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2025-04-14:
Share some example code to reproduce the issue ?

bryghtlabs-richard, 2025-04-14:
Sure @brianignacio5 , does this help?

1) In VSCode, open c:\Espressif\frameworks\esp-idf-v5.1.2\examples\get-started\blink
2) Configure for ESP32-S3
3) Build the project
4) flash to ESP32-S3 devkit
5) Start Debugging
6) Go to breakpoint tab
7) click + button with hover text 'add breakpoint'
8) type or paste function name `blink_led`
9) press F5/Continue
10) my screen looks like this
11) there is no breakpoint indicator within `blink_led` function

[image]

brianignacio5, 2025-04-24:
Hi @bryghtlabs-richard sorry for late reply. 

I do manage to reproduce the issue you mention but the breakpoint is set and the program does stop at blink_led function.

The missing part is the icon part next to the line number right? I believe this is designed this way.

In the debug adapter protocol, the key difference is that are red-dot breakpoint are defined as [SourceBreakpoint](https://microsoft.github.io/debug-adapter-protocol/specification#Types_SourceBreakpoint) and + are defined as [FunctionBreakpoint](https://microsoft.github.io/debug-adapter-protocol/specification#Types_FunctionBreakpoint). You can see SourceBreakpoint does have the line as argument passed down to gdb and the function does not, since gdb uses the function name to determine where to stop, which can be in multiple places. 

In GDB please refer [here](https://ftp.gnu.org/old-gnu/Manuals/gdb/html_node/gdb_28.html) where there is `break filename:linenum`and `break filename:function`.

bryghtlabs-richard, 2025-04-24:
> Hi @bryghtlabs-richard sorry for late reply.

No worries! Thank you for your work on ESP-IDF VSCode plugin.

>The missing part is the icon part next to the line number right?

That's correct. 

>I believe this is designed this way.

Respectfully, I believe this is a poor design choice. Generally, anytime we can, we should communicate what happened to the operator without asking to read the backend GDB or OpenOCD protocol text.

Normal flow: program stops at breakpoint. This is immediately clear because the program counter points to a breakpoint.

Confusing flow: program stops. Open up the GDB output, scroll through GDB/OCD text, find it was a breakpoint I forgot to clear from the last run. Open breakpoint menu. Read each item. Delete the right one.

> You can see SourceBreakpoint does have the line as argument passed down to gdb and the function does not, since gdb uses the function name to determine where to stop, which can be in multiple places.

Especially in this case, we should indicate the impacted function breakpoints in the IDE.

brianignacio5, 2025-04-24:
I understand your opinion regarding this, I just trying to convey that is done by design from GDB and DAP themselves. Since the functionBreakpoint doesn't include the line number (because a function could exist in multiple file and multiple lines) the red dot doesn't get added and the program just stops. I agree that this should be improved but unfortunately it might be necessary for support from the vscode devs themselves to improve this flow, in both dap and vscode UI.

bryghtlabs-richard, 2025-04-24:
Thanks - I understand now.

Do you think it could work if the GDB module returned these locations through the [BreakpointLocations](https://microsoft.github.io/debug-adapter-protocol/specification#Types_BreakpointLocation) request?

brianignacio5, 2025-04-24:
I was testing this behavior and I found that the GDB MI command does return the line for our example blink_led here.

In https://github.com/espressif/vscode-esp-idf-extension/blob/master/src/cdtDebugAdapter/adapter/GDBDebugSession.ts#L736 you can see that gdbbp.bkpt does contain a `line` field which is returned. 

Since the `setFunctionBreakpointRequest` does return [Breakpoint](https://microsoft.github.io/debug-adapter-protocol/specification#Types_Breakpoint) array it makes sense that maybe the issue was the missing line argument.

I've tried doing 

```
const createActual = (
        breakpoint: mi.MIBreakpointInfo
      ): DebugProtocol.Breakpoint => ({
        id: parseInt(breakpoint.number, 10),
        verified: true,
        line: parseInt(breakpoint.line, 10),
      });
```

but the UI result is the same and the red dot is not applied even after doing the `createActual` change above.

This leads me to believe this is a vscode design choice of only using red dot for user defined source breakpoints added here.

bryghtlabs-richard, 2025-04-24:
>This leads me to believe this is a vscode design choice of only using red dot for user defined source code breakpoints and not for other breakpoints like data breakpoints or function breakpoints.

It's possible.

    const createActual = (
        breakpoint: mi.MIBreakpointInfo
      ): DebugProtocol.Breakpoint => ({
        id: parseInt(breakpoint.number, 10),
        verified: true,
        line: parseInt(breakpoint.line, 10),
      });

Do you think it also need the `source:` specified along with line?

brianignacio5, 2025-04-24:
> It's possible.
>
> ```
> const createActual = (
>     breakpoint: mi.MIBreakpointInfo
>   ): DebugProtocol.Breakpoint => ({
>     id: parseInt(breakpoint.number, 10),
>     verified: true,
>     line: [...]

I also tested it, like this:

```ts
const createActual = (
        breakpoint: mi.MIBreakpointInfo
      ): DebugProtocol.Breakpoint => ({
        id: parseInt(breakpoint.number, 10),
        verified: true,
        line: parseInt(breakpoint.line, 10),
        source: {
          name: breakpoint.file,
          path: breakpoint.fullname,
        } as DebugProtocol.Source,
      });
```

with same result.

Considering that the setBreakpointRequest return just line `https://github.com/espressif/vscode-esp-idf-extension/blob/master/src/cdtDebugAdapter/adapter/GDBDebugSession.ts#L544`

```ts
 const createState = (
        vsbp: DebugProtocol.SourceBreakpoint,
        gdbbp: mi.MIBreakpointInfo
      ): DebugProtocol.Breakpoint => {
        if (vsbp.logMessage) {
          this.logPointMessages[gdbbp.number] = vsbp.logMessage;
        }

        let line = 0;
        if (gdbbp.line) {
          line = parseInt(gdbbp.line, 10);
        } else if (vsbp.line) {
          line = vsbp.line;
        }

        return {
          id: parseInt(gdbbp.number, 10),
          line,
          verified: true,
        };
      };
```

I can't seem to produce desired result.