espressif/vscode-esp-idf-extension #1505: Unable to set breakpoint on inline asm function call

bryghtlabs-richard, 2025-04-11:
### OS

Windows

### Operating System version

Windows 11

### Visual Studio Code version

1.99.2

### ESP-IDF version

5.1.2

### Python version

3.11.2

### Doctor command output

.

### Extension

{"message":"OpenOCD server failed to start because of Error: Unknown param: -alt-work-area-phys, try one of: -type, -event, -work-area-virt, -work-area-phys, -work-area-size, -work-area-backup, -endian, -coreid, -chain-position, -dbgbase, -rtos, -defer-examine, -gdb-port, or -gdb-max-connections\n❌ WARNING!!! OpenOCD binary version (Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)) doesn't match the .cfg file version (v0.12.0-esp32-20230419)\r\nC:\\Espressif\\tools\\openocd-esp32\\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/target/esp_common.cfg:34: Error: Unknown param: -alt-work-area-phys, try one of: -type, -event, -work-area-virt, -work-area-phys, -work-area-size, -work-area-backup, -endian, -coreid, -chain-position, -dbgbase, -rtos, -defer-examine, -gdb-port, or -gdb-max-connections\r\nin procedure 'script' \r\nat file \"embedded:startup.tcl\", line 28\r\nat file \"C:\\Espressif\\tools\\openocd-esp32\\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/board/esp32s3-builtin.cfg\", line 13\r\nin procedure 'configure_esp_workarea' called at file \"C:\\Espressif\\tools\\openocd-esp32\\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/target/esp32s3.cfg\", line 102\r\nat file \"C:\\Espressif\\tools\\openocd-esp32\\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/target/esp_common.cfg\", line 34\r\n","stack":"Error: OpenOCD server failed to start because of Error: Unknown param: -alt-work-area-phys, try one of: -type, -event, -work-area-virt, -work-area-phys, -work-area-size, -work-area-backup, -endian, -coreid, -chain-position, -dbgbase, -rtos, -defer-examine, -gdb-port, or -gdb-max-connections\n\tat Socket.<anonymous> (\\dist\\extension.js:2:1566337)\n\tat Socket.emit (node:events:518:28)\n\tat addChunk (node:internal/streams/readable:561:12)\n\tat readableAddChunkPushByteMode (node:internal/streams/readable:512:3)\n\tat Readable.push (node:internal/streams/readable:392:5)\n\tat Pipe.onStreamRead (node:internal/stream_base_commons:191:23)","level":"error","timestamp":"2025-02-19T20:41:57.274Z"}
{"message":"OpenOCD Exit with non-zero error code 1","stack":"Error: Spawn exit with non-zero1\n\tat ChildProcess.<anonymous> (\\dist\\extension.js:2:1567030)\n\tat ChildProcess.emit (node:events:518:28)\n\tat maybeClose (node:internal/child_process:1104:16)\n\tat Socket.<anonymous> (node:internal/child_process:456:11)\n\tat Socket.emit (node:events:518:28)\n\tat Pipe.<anonymous> (node:net:343:12)","level":"error","timestamp":"2025-02-19T20:41:57.275Z"}
{"message":"OpenOCD server failed to start because of Error: Unknown param: -alt-work-area-phys, try one of: -type, -event, -work-area-virt, -work-area-phys, -work-area-size, -work-area-backup, -endian, -coreid, -chain-position, -dbgbase, -rtos, -defer-examine, -gdb-port, or -gdb-max-connections\n❌ -esp32-20230419/openocd-esp32/share/openocd/scripts/target/esp_common.cfg:34: Error: Unknown param: -alt-work-area-phys, try one of: -type, -event, -work-area-virt, -work-area-phys, -work-area-size, -work-area-backup, -endian, -coreid, -chain-position, -dbgbase, -rtos, -defer-examine, -gdb-port, or -gdb-max-connections\r\nin procedure 'script' \r\nat file \"embedded:startup.tcl\", line 28\r\nat file \"C:\\Espressif\\tools\\openocd-esp32\\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/board/esp32s3-builtin.cfg\", line 13\r\nin procedure 'configure_esp_workarea' called at file \"C:\\Espressif\\tools\\openocd-esp32\\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/target/esp32s3.cfg\", line 102\r\nat file \"C:\\Espressif\\tools\\openocd-esp32\\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/target/esp_common.cfg\", line 34\r\n","stack":"Error: OpenOCD server failed to start because of Error: Unknown param: -alt-work-area-phys, try one of: -type, -event, -work-area-virt, -work-area-phys, -work-area-size, -work-area-backup, -endian, -coreid, -chain-position, -dbgbase, -rtos, -defer-examine, -gdb-port, or -gdb-max-connections\n\tat Socket.<anonymous> (\\dist\\extension.js:2:1566337)\n\tat Socket.emit (node:events:518:28)\n\tat addChunk (node:internal/streams/readable:561:12)\n\tat readableAddChunkPushByteMode (node:internal/streams/readable:512:3)\n\tat Readable.push (node:internal/streams/readable:392:5)\n\tat Pipe.onStreamRead (node:internal/stream_base_commons:191:23)","level":"error","timestamp":"2025-02-19T20:44:34.357Z"}
{"message":"OpenOCD Exit with non-zero error code 1","stack":"Error: Spawn exit with non-zero1\n\tat ChildProcess.<anonymous> (\\dist\\extension.js:2:1567030)\n\tat ChildProcess.emit (node:events:518:28)\n\tat maybeClose (node:internal/child_process:1104:16)\n\tat Socket.<anonymous> (node:internal/child_process:456:11)\n\tat Socket.emit (node:events:518:28)\n\tat Pipe.<anonymous> (node:net:343:12)","level":"error","timestamp":"2025-02-19T20:44:34.367Z"}
{"message":"OpenOCD server failed to start because of Error: Unknown param: -alt-work-area-phys, try one of: -type, -event, -work\n❌ p32\\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/target/esp_common.cfg:34: Error: Unknown param: -alt-work-area-phys, try one of: -type, -event, -work","stack":"Error: OpenOCD server failed to start because of Error: Unknown param: -alt-work-area-phys, try one of: -type, -event, -work\n\tat Socket.<anonymous> (\\dist\\extension.js:2:1566337)\n\tat Socket.emit (node:events:518:28)\n\tat addChunk (node:internal/streams/readable:561:12)\n\tat readableAddChunkPushByteMode (node:internal/streams/readable:512:3)\n\tat Readable.push (node:internal/streams/readable:392:5)\n\tat Pipe.onStreamRead (node:internal/stream_base_commons:191:23)","level":"error","timestamp":"2025-02-19T20:50:20.204Z"}
{"message":"OpenOCD Exit with non-zero error code 1","stack":"Error: Spawn exit with non-zero1\n\tat ChildProcess.<anonymous> (\\dist\\extension.js:2:1567030)\n\tat ChildProcess.emit (node:events:518:28)\n\tat maybeClose (node:internal/child_process:1104:16)\n\tat Socket.<anonymous> (node:internal/child_process:456:11)\n\tat Socket.emit (node:events:518:28)\n\tat Pipe.<anonymous> (node:net:343:12)","level":"error","timestamp":"2025-02-19T20:50:20.207Z"}
{"message":"OpenOCD server failed to start because of Error: Unknown param: -alt-work-area-phys, try one of: -type, -event, -work-area-virt, -work-area-phys, -work-area-size, -work-area-backup, -endian, -coreid, -chain-position, -dbgbase, -rtos, -defer-examine, -gdb-port, or -gdb-max-connections\n❌ enocd/scripts/target/esp_common.cfg:34: Error: Unknown param: -alt-work-area-phys, try one of: -type, -event, -work-area-virt, -work-area-phys, -work-area-size, -work-area-backup, -endian, -coreid, -chain-position, -dbgbase, -rtos, -defer-examine, -gdb-port, or -gdb-max-connections\r\nin procedure 'script' \r\nat file \"embedded:startup.tcl\", line 28\r\nat file \"C:\\Espressif\\tools\\op","stack":"Error: OpenOCD server failed to start because of Error: Unknown param: -alt-work-area-phys, try one of: -type, -event, -work-area-virt, -work-area-phys, -work-area-size, -work-area-backup, -endian, -coreid, -chain-position, -dbgbase, -rtos, -defer-examine, -gdb-port, or -gdb-max-connections\n\tat Socket.<anonymous> (\\dist\\extension.js:2:1566337)\n\tat Socket.emit (node:events:518:28)\n\tat addChunk (node:internal/streams/readable:561:12)\n\tat readableAddChunkPushByteMode (node:internal/streams/readable:512:3)\n\tat Readable.push (node:internal/streams/readable:392:5)\n\tat Pipe.onStreamRead (node:internal/stream_base_commons:191:23)","level":"error","timestamp":"2025-03-26T15:20:44.606Z"}
{"message":"OpenOCD Exit with non-zero error code 1","stack":"Error: Spawn exit with non-zero1\n\tat ChildProcess.<anonymous> (\\dist\\extension.js:2:1567030)\n\tat ChildProcess.emit (node:events:518:28)\n\tat maybeClose (node:internal/child_process:1104:16)\n\tat ChildProcess._handle.onexit (node:internal/child_process:304:5)","level":"error","timestamp":"2025-03-26T15:20:44.609Z"}
{"message":"OpenOCD server failed to start because of Error: Unknown param: -alt-work-area-phys, try one of: -type, -event, -work-area-virt, -work-area-phys, -work-area-size, -work-area-backup, -endian, -coreid, -chain-position, -dbgbase, -rtos, -defer-examine, -gdb-port, or -gdb-max-connections\n❌ -esp32/share/openocd/scrip [... cut ...]
I understand that execution breakpoints can only be set on instructions, and that the mapping between lines of C/C++ and instructions is tenuous at best. But in this case, the above code should be compiled with `-Og` so I think the IDE is at fault. 

If attempting to place a breakpoint on a line without an instruction, a more useful approach would be to place the breakpoint on the source line where the next instruction occurs.

### Debug Message

No logs, it just silently fails.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2025-04-24:
In the debug console, do you see any output related to set breakpoint ?

Also in the debug console, could you try in insert a breakpoint from the console terminal and see what is the result of that.

gerekon, 2025-05-13:
@bryghtlabs-richard I can reproduce this with ESP32 GDB from command line. So it is not the problem of VsCode.
```
(gdb) b cpu.c:108
Breakpoint 2 at 0x40084630: file /home/alexey/projects/esp/esp-idf/components/esp_hw_support/cpu.c, line 109.
(gdb) disass 0x40084630
(gdb) b cpu.c:110
Breakpoint 3 at 0x400e3cb8: file /home/alexey/projects/esp/esp-idf/components/esp_hw_support/cpu.c, line 272.
Note: automatically using hardware breakpoints for read-only addresses.
Dump of assembler code for function esp_cpu_wait_for_intr:
   0x40084630 <+0>:	entry	a1, 32
   0x40084633 <+3>:	waiti	0
   0x40084636 <+6>:	retw.n
End of assembler dump.
```
It can be partially worked around if you enable `Compiler options → Optimization Level → Debug without optimization (-O0)` instead of default `Debug (-Og)`.
In that case compiler adds two additional nops before return making `retw.n` aligned at 4 bytes boundary.
```
(gdb) b cpu.c:108
Breakpoint 2 at 0x400869d5: file /home/alexey/projects/esp/esp-idf/components/esp_hw_support/cpu.c, line 111.
(gdb) disass 0x400869d5
Dump of assembler code for function esp_cpu_wait_for_intr:
   0x400869d0 <+0>:	entry	a1, 32
   0x400869d3 <+3>:	mov.n	a7, a1
   0x400869d5 <+5>:	waiti	0
   0x400869d8 <+8>:	nop.n
   0x400869da <+10>:	nop.n
   0x400869dc <+12>:	retw.n
End of assembler dump.
(gdb) b cpu.c:110
Breakpoint 3 at 0x400869da: file /home/alexey/projects/esp/esp-idf/components/esp_hw_support/cpu.c, line 121.
(gdb) b cpu.c:111
Note: breakpoint 3 also set at pc 0x400869da.
Breakpoint 4 at 0x400869da: file /home/alexey/projects/esp/esp-idf/components/esp_hw_support/cpu.c, line 121.
```
But as you can see the actual line is 121 (not 110/111) and breakpoint is set just before return instead of `0x400869d5` (on `waiti	0`). So I will move this issue to our toolchain repo.