espressif/vscode-esp-idf-extension #1507: IDE flashes between multiple text files

bryghtlabs-richard, 2025-04-12:
### OS

Windows

### Operating System version

Windows 11

### Visual Studio Code version

1.99.2

### ESP-IDF version

5.1.2

### Python version

3.11.2

### Doctor command output

text too large makes browser slow

### Extension

text too large makes browswer slow

### Description

With too many breakpoints(around 3 or 4) set in the debugger, each time I click 'step over', the editor flickers to a different source code file (usually `esp_cpu_wait_for_intr()`) before going back to the correct line. This is very distracting, since I can no longer keep my eyes on the relevant code in the debugger.

Disabling the extra breakpoints appears to fix it.

### Debug Message

.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

bryghtlabs-richard, 2025-04-15:
Today, with 3 breakpoints, this isn't happening. I'll try to narrow down the conditions that cause it.

brianignacio5, 2025-04-24:
It might be related to the limited number of HW breakpoints and how openOCD does software breakpoints in flash:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/tips-and-quirks.html#breakpoints-and-watchpoints-available

@erhankur @gerekon  Opinions about this ?

bryghtlabs-richard, 2025-05-29:
Still a problem in 1.10, still intermittent.

bryghtlabs-richard, 2025-06-16:
Still a problem, still intermittent.

brianignacio5, 2025-07-01:
Could you provide some steps to reproduce in a small example project? Need to understand the issue better.

bryghtlabs-richard, 2025-07-01:
I'm afraid I can't. This morning I tried with `esp_http_client` example and with my product code, but it didn't happen. Seems like it only happens when trying to reproduce a real fault.

closed 2025-07-22 by github-actions[bot]