espressif/vscode-esp-idf-extension #1508: flash device fail when use `rfc2217://host.docker.internal:4000?ign_set_control` as port

shishirong, 2025-04-13:
### OS

Windows

### Operating System version

Window + wsl2+ docker

### ESP-IDF version

5.3.2

### Python version

3.12.3

### Doctor command output

```log
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 6.6.75.1-microsoft-standard-WSL2 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 /vscode/vscode-server/bin/linux-x64/4949701c880d4bdb949e3c0e6b400288da7f474b/bin/remote-cli:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:<HOMEPATH>/.local/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.99.2 
Visual Studio Code language zh-cn 
Visual Studio Code shell /usr/bin/bash 
ESP-IDF Extension version 1.9.1 
Workspace folder /workspaces/RopeV2.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) /opt/esp/esp-adf
ESP-IDF Path (idf.espIdfPath) /opt/esp/idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths /opt/esp/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin:/opt/esp/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin:/opt/esp/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin:/opt/esp/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin:/opt/esp/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin:/opt/esp/tools/cmake/3.30.2/bin:/opt/esp/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/bin:/opt/esp/tools/ninja/1.12.1:/opt/esp/tools/esp-rom-elfs/20240305
ESP-IDF extra vars
    OPENOCD_SCRIPTS: /opt/esp/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/share/openocd/scripts
    ESP_ROM_ELF_DIR: /opt/esp/tools/esp-rom-elfs/20240305/
    IDF_PYTHON_ENV_PATH: /opt/esp/python_env/idf5.3_py3.12_env
    ESP_IDF_VERSION: 5.3
    PATH: /opt/esp/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin:/opt/esp/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin:/opt/esp/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin:/opt/esp/tools/cmake/3.30.2/bin:/opt/esp/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/bin:/opt/esp/tools/qemu-xtensa/esp_develop_9.0.0_20240606/qemu/bin:/opt/esp/python_env/idf5.3_py3.12_env/bin:/opt/esp/idf/tools:$PATH
    IDF_DEACTIVATE_FILE_PATH: /tmp/tmptg8ty3u6idf_7895
User extra vars (idf.customExtraVars)
System python Path (idf.pythonInstallPath) /usr/bin/python
Virtual environment Python path (computed) /opt/esp/python_env/idf5.3_py3.12_env/bin/python
Serial port (idf.port) rfc2217://host.docker.internal:4000?ign_set_control
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) /opt/esp
Git Path (idf.gitPath) /usr/bin/git
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) true
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to /opt/esp/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin: true
Access to /opt/esp/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin: false
Access to /opt/esp/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin: true
Access to /opt/esp/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin: false
Access to /opt/esp/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin: true
Access to /opt/esp/tools/cmake/3.30.2/bin: true
Access to /opt/esp/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/bin: true
Access to /opt/esp/tools/ninja/1.12.1: false
Access to /opt/esp/tools/esp-rom-elfs/20240305: true
[... 90 lines cut ...]
Device PID identification is only supported on COM and /dev/ serial ports.
...................................

A fatal error occurred: Failed to connect to ESP32: Wrong boot mode detected (0x1e)! The chip needs to be in download mode.
```

extension config 
```json
{
    "idf.port": "rfc2217://host.docker.internal:4000?ign_set_control",
    "idf.flashType": "UART",
}
```
ref: https://github.com/espressif/vscode-esp-idf-extension/issues/471

docker is note stablize when use new version kernal what added usb module, so use rfc2217 remote uart is better for me.
fix this pls!

### Debug Message

similar as Description

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2025-04-14:
I think this is an issue with esptool.py itself not the vscode extension.

You can also experiment using usbipd as described in [here](https://docs.espressif.com/projects/vscode-esp-idf-extension/en/latest/additionalfeatures/docker-container.html)

shishirong, 2025-04-14:
My fault, 
It's a lot of latency time, because I'm also using virtualhere remote usb , which can cause a lot of latency

closed 2025-04-14 by shishirong