espressif/vscode-esp-idf-extension #1509: Invalid VSCode json files generated for new project

loehnertj, 2025-04-14:
### OS

Linux

### Operating System version

Ubuntu 24.04.2 LTS

### Visual Studio Code version

1.99.0 (2025-04-02)

### ESP-IDF version

5.5.0 (Github master branch)

### Python version

3.12.3

### Doctor command output

See attached report.txt

[report.txt](https://github.com/user-attachments/files/19741538/report.txt)

### Extension

Found no related info in the log (except followup errors)

### Description

When setting up a project from any example (e.g. blink, hello_world, file_serving), the files ``.vscode/settings.json`` and ``.vscode/c_cpp_properties.json`` are invalid. Namely, each line is prepended by the literal number ``4`` one or more times.

Example c_cpp_properties.json:

```
{
4"configurations": [
44{
444"name": "ESP-IDF",
444"compilerPath": "${config:idf.toolsPath}/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc",
444"compileCommands": "${config:idf.buildPath}/compile_commands.json",
444"includePath": [
(...)
```

It strongly looks like the json writer is supposed to put 4 spaces of indent, but instead uses the literal number "4" as indent string. Being a noob, I couldn't pinpoint the location of the fault.

The json files can be fixed by hand but it is mildly inconvenient. ;-)

### Debug Message

No relevant debug messages seen. Supposedly, json dumping works fine from the perspective of the generator.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2025-04-15:
This happens because you have defined `"editor.tabSize": "4"` in your settings.json, either in `Preferences: Open User Settings (JSON)` or in the project you had opened when you created the new project

The value should be `"editor.tabSize": 4`

brianignacio5, 2025-04-15:
Here is a bugfix PR to use fixed value 2 for spaces = tab in #1510 

You can test these changes by installing this VSIX by click menu **View** -> **Command Palette...**, type **Install from VSIX** and then select downloaded **esp-idf-extension.vsix** file to install the extension.
* [esp-idf-extension.vsix.zip](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/2945046641.zip)

loehnertj, 2025-04-15:
Indeed, you were right. This seems in turn to be a bug in the VsCode-Vim extension. The ``"editor.tabSize": "4" line is created by the Vim command ``:set ts=4``. Too lazy to file another issue now :-/

Anyway, I tested the bugfixed vsix file and indeed, JSON files are okay now even with the bad tabSize setting. Many thanks!

PR #1510 (2025-04-16): remove use of editor tab size set static 2 spaces
## Description

Remove use of `editor.tabSize` to determine spaces equivalent for tabs in generated json files such as c_cpp_properties.json, settings.json, etc.

Fixes #1503
Fixes #1506 
Fixes #1509

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Click on "ESP-IDF: New Project" to create a new project. All json files should have a fixed 2 spaces tab size.

- Expected behaviour:
All json files should have a fi [... cut]
closed 2025-04-16 by brianignacio5