espressif/vscode-esp-idf-extension #1511: [Feature Request]: Debugger watch-panel: format variables

bryghtlabs-richard, 2025-04-15:
**Is your feature request related to a problem? Please describe.**

When debugging ESP32 projects, I'm often frustrated trying to interpret variables.

**Describe the solution you'd like**
In the watch panel, I'd like the ability to decode variables a few different ways.

1. For C integral types, I'd like to be able to display them as integer, or as hexadecimal.
2. For `char *` and `char[]` types, I'd like to be able to display them as an ASCII or UTF8 string(until the end of buffer, or first NULL-char, or end of memory bank).

**Describe alternatives you've considered**

Currently for char[], I scroll vertically trying to read the string, like early pictogram cuneiform. For integral types, I often copy and paste them into Windows-calculator. These are not good use of debugging time.

**Additional context**
[I found some hacks here](https://stackoverflow.com/questions/39973214/how-to-display-the-hex-value-in-the-watch-panel-of-vs-code), but they didn't work.

Here's an example of where I'd like to be able to toggle the display mode. This number in decimal means nothing, but in hex is quite clear to me:

[image]

My apologies if there's a way to do this already, but I couldn't figure it out.

jdbaptista, 2025-04-22:
@bryghtlabs-richard, you can cast the variable to a different type such as (char *) or (char) within the watch expression as shown below.

[image]

Alternatively, you can click the 'View Binary Data' button, which will open the memory viewer. 

[image]

This will allow you to see decoded text beside the RAM and view data in different formats, such as octal. I'm not sure how to directly change the format in the WATCH panel directly, though.

[image]

bryghtlabs-richard, 2025-04-22:
Thanks @jdbaptista , that's helpful for strings for sure.

It looks like VSCode added the ability to format strings towards the end of this issue(I had to expand everything to find it): https://github.com/microsoft/vscode/issues/28025 . The debug plugin will need extended to [support this though](https://code.visualstudio.com/updates/v1_49#_contributable-context-menu-for-variables-view):

[image]

PR #1521 (2025-05-09): [VSC-1654 | 1584 | 1583] add evaluate on hover, view vars as hex, set data breakpoints
## Description

1. Add the ability to show variables values on hover (it will run a evaluateRequest to the debug adapter)
2. Add a right click option to show the integer value parsed as Hex and binary. Values will be  show in `ESP-IDF: Hex View` in debug tab. Variables values are updated during debug session if it changes.
3. Add Data breakpoints request. Just right click a variable and select `Break on Value Read`, `Break on Value Read` or `Break on Value Read`.

[image]

Fixes #1511 
Fixes #1420
Fixes #141 [... cut]
closed 2025-05-09 by brianignacio5