espressif/vscode-esp-idf-extension #1512: [Feature Request]: distinct name status-bar icons

bryghtlabs-richard, 2025-04-15:
**Is your feature request related to a problem? Please describe.**
When configuring the status-bar, it's confusing when all ESP-IDF status icons share the name "ESP-IDF (extension)"

**Describe the solution you'd like**
I'd like each status-bar icon to have a distinct name in the right-click menu

**Describe alternatives you've considered**
An alternative is to configure the IDE one icon at a time.

**Additional context**
Here's what I mean:

[image]

brianignacio5, 2025-04-16:
You can enable disable status bar icons from here:

[image]

bryghtlabs-richard, 2025-04-16:
> Where do you see this menu? I'm not aware how the status bar icon name is described here by the vscode api. Need to take a look but I assume is not using the field from the api.

Oops, the screenshot above is from right-clicking on the status menu itself:

[image]

brianignacio5, 2025-04-17:
It doesn't seem to be possible to modify the name shown in this menu you've shown. VSCODE seems to report the status bar item just from source extension and doesn't use the status bar item name or field as described in https://code.visualstudio.com/api/references/vscode-api#StatusBarItem

Seems this is more of a vscode feature request itself. Consider this is true for other extensions too.

bryghtlabs-richard, 2025-04-17:
> Seems this is more of a vscode feature request itself. Consider this is true for other extensions too.

My other extensions do not do this, so I looked into it.

>It doesn't seem to be possible to modify the name shown in this menu you've shown. VSCODE seems to report the status bar item just from source extension and doesn't use the status bar item name or field as described in https://code.visualstudio.com/api/references/vscode-api#StatusBarItem

I haven't setup ESP-VSCode extension to build from source yet, so I hacked the minimized extension.js file to set `statusBarItem.name = statusBarItem.text`, and I get this:

[image]

So, it looks like setting the .name field worked for me (except for icons without text, which are now clear in the menu), and we need useful names for the .name property.

brianignacio5, 2025-04-18:
Apologies you are right!

I've added the PR https://github.com/espressif/vscode-esp-idf-extension/pull/1515 to fix the issue !

bryghtlabs-richard, 2025-05-09:
Thanks Brian!

PR #1515 (2025-05-09): add status bar item names
## Description

Status bar items when you click right them should show the name of the command they will execute. Before was only as described in tooltip.

Fixes #1512

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

1. Right click an status bar item from this extension
2. The tooltip text should be shown in the list of status bar items shown.

[image]

- Expected behaviour:

Status bar item should show meaningful text for right click (to enable disable status bar item).

- Expected output:

## How has this  [... cut]