espressif/vscode-esp-idf-extension #1520: [Feature Request]: Add multiple configuration support for idf.py argument files

jdbaptista, 2025-04-21:
**Is your feature request related to a problem? Please describe.**
I make heavy use of the extension's multiple configuration feature, however it does not allow me to conditionally set the CMake TEST_COMPONENTS variable. Something like below is what would solve my issue, but it is impossible to do this way because `idf_build_get_config()` is not usable before `project()`:
```
idf_build_get_config(COMPONENT_UNDER_TEST CONFIG_TEST_COMPONENT)
message(STATUS "CONFIG_TEST_COMPONENT = ${COMPONENT_UNDER_TEST}")

if (COMPONENT_UNDER_TEST STREQUAL "test_ota")
    set(TEST_COMPONENTS "ota")
elseif(COMPONENT_UNDER_TEST STREQUAL "test_indicators")
   set(TEST_COMPONENTS "indicators")
endif()

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(esp-firmware)
```
Ultimately I would like to have multiple testing configurations that test different components, which is necessary in order to mock particular component dependencies in some test builds and not others. If using the command line directly, I can achieve this goal by creating an [idf.py arguments file](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/tools/idf-py.html#passing-arguments-via-a-file) and passing it in with `idf.py @ test_ota build flash monitor`. I plan on creating these files soon, however I would much rather use the IDE extension. I am under the assumption that the multiple configuration extension feature uses idf.py argument files in the background, however I do not have a direct way to add the -T option to them to choose which tests are run.

**Describe the solution you'd like**
The esp_idf_project_configuration.json file defines the following options for use:
```
"test_ota": {
    "build": {
      "compileArgs": [],
      "ninjaArgs": [],
      "buildDirectoryPath": "${workspaceFolder}\\builds\\test_ota",
      "sdkconfigDefaults": [
        ".\\configurations\\sdkconfig.settings",
        ".\\configurations\\secrets\\sdkconfig.test_secrets",
        ".\\configurations\\sdkconfig.test",
        ".\\components\\ota\\test\\sdkconfig.test_ota",
        ".\\configurations\\sdkconfig.V2_0_default"
      ],
      "sdkconfigFilePath": "${workspaceFolder}\\configurations\\current\\sdkconfig.test_ota"
    },
    "env": {},
    "flashBaudRate": "",
    "monitorBaudRate": "",
    "openOCD": {
      "debugLevel": 0,
      "configs": [],
      "args": []
    },
    "tasks": {
      "preBuild": "",
      "preFlash": "",
      "postBuild": "",
      "postFlash": ""
    }
  }
```
I would like to be able to link an idf.py arguments file to a project configuration in order to set other flags such as -T. Alternatively, I would like to be able to list components for test with an option in the json file. I believe that adding the ability to link an arguments file would be a simpler and more extensible solution. Something like:
```
"test_ota": {
    ...
    "tasks": {
      "preBuild": "",
      "preFlash": "",
      "postBuild": "",
      "postFlash": ""
    },
    "extraIDFArgumentsFiles": [
         ".\\components\\ota\\test\\idf_arguments"
     ] 
  }
```
Where `components\ota\test\idf_arguments` would contain "-T ota".

**Describe alternatives you've considered**
Adding a direct option to configure the -T option is another way to add this feature.

closed 2025-04-22 by jdbaptista

brianignacio5, 2025-04-22:
Hi @jdbaptista did you find a solution for the issue you were facing ?

jdbaptista, 2025-04-22:
Brian,

Yes, I was able to solve my issue by defining an environment variable in the "env" portion of each configuration. I can then use this to conditionally choose options in the CMake files with $ENV{var}, which usually needs quotations around it.

I would like to say that this is not well-documented in the extension, but was fairly intuitive. The -T idf.py option is not actually a standard option for projects, so I don't think being able to modify this will be an issue for others.

For future reference on mocking your own components, here is my setup:

1. Each project configuration option defines the environment variable.
```
"env": {
  "executable": "test_ota"
},
```
2. The TEST_COMPONENTS CMake variable is set depending on the value of "$ENV{executable}".
```
message(STATUS "executable: $ENV{executable}")
if ("$ENV{executable}" STREQUAL "test_ota")
    set(TEST_COMPONENTS "ota")
endif()
```
3. The main component conditionally compiles the correct main source file based on the variable.
```
if ("$ENV{executable}" STREQUAL "main")
    set(srcs "src/main.c"
             "src/initialize.c")
    set(includes "include")
elseif ("$ENV{executable}" STREQUAL "test_ota")
    set(srcs "src/test_ota_main.c"
             "src/initialize.c")
    set(includes "include")
else()
    message(FATAL_ERROR "Value of executable environment variable $ENV{executable} is not known!")
endif()

set(requires ...)

idf_component_register(SRCS ${srcs}
                        INCLUDE_DIRS ${includes}
                        REQUIRES ${requires})
```
4. The ota testing component contains a file that defines configuration options that control mocking, which is included as a default file in the test_ota configuration.
```
# list of mocked components for ota unit tests
CONFIG_MOCK_INDICATORS=y
```
5. Other components use the configuration option to conditionally compile mocking source files instead of regular sources and to add mocking control header files.
```
set(srcs "src/indicators.c")
set(include_dirs "include")
set(requires common app_errors animations)

if (CONFIG_MOCK_INDICATORS)
    set(srcs "mock/src/mock_indicators.c")
    list(APPEND include_dirs "mock/include")
endif ()

idf_component_register(SRCS ${srcs}
                    INCLUDE_DIRS ${include_dirs}
                    REQUIRES ${requires})
```

brianignacio5, 2025-04-23:
Thanks @jdbaptista for the reply!

You are right, we should update https://docs.espressif.com/projects/vscode-esp-idf-extension/en/latest/additionalfeatures/project-configuration.html to have better description.

If you don't mind, could I use your example code in the project configuration documentation as env tutorial ? The project configuration editor can be very useful but I have yet to provide such good examples as yours.