espressif/vscode-esp-idf-extension #1524: [Feature Request]: show currently selected OpenOCD target

bryghtlabs-richard, 2025-04-24:
**Is your feature request related to a problem? Please describe.**
This is a very small issue, but I was having some trouble configuring OpenOCD. I pressed F1, `Select OpenOCD Board Config`, but it seems to always highlights the top board config, even after I've chosen another board config.

**Describe the solution you'd like**
I'd like the currently selected board-config to be highlighted if we can. Or moved to the top?

**Additional context**

I believe I selected `via builtin USB-JTAG`, but it seems to always show via ESP-PROG. Is ESP-PROG actually selected? Or, does it just always show 'via ESP-PROG'?

[image]

brianignacio5, 2025-04-28:
Made a fix in #1527

bryghtlabs-richard, 2025-05-01:
Thanks @brianignacio5 , VSIX works great for me!

PR #1527 (2025-05-12): [VSC-1661] use createQuickPick for openocd board to show currently active item.
## Description

Add quick pick functionality with currently selected option.

Fixes #1524
Fixes #1543

## Type of change

- New feature (non-breaking change which adds functionality)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Click on "ESP-IDF: Select OpenOCD Configuration"
2. Execute action. The currently selected OpenOCD board should be highlighted.
3. Observe results.

- Expected behaviour:
The currently selected OpenOCD board should be highlig [... cut]
closed 2025-05-12 by brianignacio5