espressif/vscode-esp-idf-extension #1526: [Feature Request]: esp32s3 can not lanuch qemu in esp-idf-vscode

razor7788, 2025-04-27:
ERROR:
esp32s3 is not supported by Espressif QEMU. Check ESP-IDF and QEMU version installed.

brianignacio5, 2025-04-28:
I think we should open a ticket in esp-idf directly because we don't maintain supported targets for tool.

Please refer to https://github.com/espressif/esp-idf/blob/master/tools/tools.json#L955

This line should be updated for all supported targets.

razor7788, 2025-04-28:
> I think we should open a ticket in esp-idf directly because we don't maintain supported targets for QEMU tool.
> 
> Please refer to https://github.com/espressif/esp-idf/blob/master/tools/tools.json#L955
> 
> This line should be updated for all supported targets.

ok  i should report this issue on esp-idf project

closed 2025-04-28 by razor7788