espressif/vscode-esp-idf-extension #1528: Project component missing from menuconfig on Linux

jdbaptista, 2025-04-27:
### OS

Linux

### Operating System version

Ubuntu 24.04.2 LTS

### Visual Studio Code version

1.99.3 (17baf8)

### ESP-IDF version

5.3.1 & 5.3.3 & 5.4.1

### Python version

3.12.3

### Doctor command output

[report.txt](https://github.com/user-attachments/files/19931431/report.txt)

### Extension

[esp_idf_vsc_ext.log](https://github.com/user-attachments/files/19931436/esp_idf_vsc_ext.log)

### Description

I have project configurations which pull from multiple sdkconfigDefaults, with "sdkconfig.V2_0_default" containing
```
CONFIG_IDF_TARGET="esp32s3"
CONFIG_HARDWARE_VERSION=15
CONFIG_HARDWARE_REVISION=0

CONFIG_SUPPORT_STROBING=y
...
```
CONFIG_SUPPORT_STROBING is a KConfig option  in my  `animations` component under the `Animation Menu` configuration menu. In the sdkconfig file that is generated when setting the device target, I expect to see both `CONFIG_HARDWARE_VERSION=15` and `CONFIG_SUPPORT_STROBING=y`. The debug output of the command is:
```
Loading defaults file /home/jdbaptista/Documents/repos/traffic-pcb-sb/software/esp-firmware/configurations/sdkconfig.settings...
Loading defaults file /home/jdbaptista/Documents/repos/traffic-pcb-sb/software/esp-firmware/configurations/sdkconfig.develop...
Loading defaults file /home/jdbaptista/Documents/repos/traffic-pcb-sb/software/esp-firmware/configurations/sdkconfig.V2_0_default...
warning: unknown kconfig symbol 'SUPPORT_STROBING' assigned to 'y' in /home/jdbaptista/Documents/repos/traffic-pcb-sb/software/esp-firmware/configurations/sdkconfig.V2_0_default
```
I see `CONFIG_HARDWARE_VERSION=15`, however `CONFIG_SUPPORT_STROBING=y` is not in the file and `#CONFIG_SUPPORT_STROBING is not set` is also not in the file. When opening menuconfig under this project configuration, the "Animations Menu" is not present even though the debug output states that "animations" is a component and has the correct component path.
```
-- Components: actions animations api_connect app_errors app_nvs app_trace app_update bootloader bootloader_support bt circular_buffer cmock common console cxx driver efuse esp-tls esp_adc esp_app_format esp_bootloader_format esp_coex esp_common esp_driver_ana_cmpr esp_driver_cam esp_driver_dac esp_driver_gpio esp_driver_gptimer esp_driver_i2c esp_driver_i2s esp_driver_isp esp_driver_jpeg esp_driver_ledc esp_driver_mcpwm esp_driver_parlio esp_driver_pcnt esp_driver_ppa esp_driver_rmt esp_driver_sdio esp_driver_sdm esp_driver_sdmmc esp_driver_sdspi esp_driver_spi esp_driver_touch_sens esp_driver_tsens esp_driver_uart esp_driver_usb_serial_jtag esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_vfs_console esp_wifi espcoredump esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 indicators json led_matrix log lwip main mbedtls mqtt newlib nvs_flash nvs_sec_provider openthread ota partition_table perfmon protobuf-c protocomm pthread refresh routines sdmmc soc spi_flash spiffs tcp_transport test_config test_manual touch_element ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /home/jdbaptista/Documents/repos/traffic-pcb-sb/software/esp-firmware/components/actions /home/jdbaptista/Documents/repos/traffic-pcb-sb/software/esp-firmware/components/animations
...
```
The same setup is working on Windows 10, with the only difference being that I have changed the filepaths in `esp_idf_project_configuration.json` from 
```
"buildDirectoryPath": "${workspaceFolder}\\builds\\develop_V2_0",
      "sdkconfigDefaults": [
        ".\\configurations\\sdkconfig.settings",
        ".\\configurations\\sdkconfig.develop",
        ".\\configurations\\sdkconfig.V2_0_default"
      ],
      "sdkconfigFilePath": "${workspaceFolder}\\configurations\\current\\sdkconfig.dev_V2_0"
```
to
```
"buildDirectoryPath": "${workspaceFolder}/builds/develop_V2_0",
      "sdkconfigDefaults": [
        "./configurations/sdkconfig.settings",
        "./configurations/sdkconfig.develop",
        "./configurations/sdkconfig.V2_0_default"
      ],
      "sdkconfigFilePath": "${workspaceFolder}/configurations/current/sdkconfig.dev_V2_0"
```
I would also like to say that it would be a nice feature for the backslashes and forward slashes here to be switched automatically depending on what the operating system expects.

I have tested that this doesn't work with ESP-IDF v5.3.1, v5.3.3, and v5.4.1. For reference, my main component requires the animations component:

```
idf_component_register(SRCS ${srcs}
                        INCLUDE_DIRS ${includes}
                        REQUIRES unity
                                common
                                test_manual
                                led_matrix
                                actions
                                animations
                                api_connect
                                app_errors
                                app_nvs
                                ota
                                routines
                                refresh
                                indicators
                                esp_driver_gpio
                                esp_driver_uart
                                esp_driver_usb_serial_jtag
                                esp_netif
                                esp-tls
                                vfs
                                esp_wifi
                                nvs_flash
                                esp_timer)
```

### Debug Message

[debug.txt](https://github.com/user-attachments/files/19931516/debug.txt)

### Other Steps to Reproduce

This is the repo I am having issues with: https://github.com/zinkwazi/traffic-pcb-sb. The project is located in software/esp-firmware. V2_0 project configurations are meant to be built for an ESP32S3 and V1_0 for an ESP32. If necessary, I can make a smaller example project that has a similar structure to see if the issue occurs there; please let me know if that would be helpful!

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2025-04-28:
Hi @jdbaptista 

I'm not very sure what could be the issue. I would recommend reaching out to https://github.com/espressif/esp-idf-kconfig to see if there an issue about this.

From what I can see maybe the confserver.py has not added animations as part of the project build. On the build directory, there should be generated a `build/config/kconfig_menus.json` containing kconfig items from the whole project.

Does this happens if you try using `idf.py menuconfig` ?

jdbaptista, 2025-05-05:
The component menu is not present in `build/config/kconfig_menus.json` nor does it show up in `idf.py menuconfig`. I will reach out at the other repo and update this issue if a solution is found there.

jdbaptista, 2025-05-06:
The issue was that my configuration files were named `KConfig.projbuild` instead of `Kconfig.projbuild`. Another issue that was brought up is that `true`/`false` default values are unrecognized and should be `y`/`n` instead.

closed 2025-05-06 by jdbaptista