espressif/vscode-esp-idf-extension #1530: [Feature Request]: "Install ESP-Matter" command should prompt for version

olavt, 2025-04-28:
When you issue the "Install ESP-Matter" command it looks like the current snapshot is pulled from the esp-matter repo.

I think it would be better if the "Install ESP-Matter" command could prompt for a specific version. Ideally it should also just present the esp-matter version, which would be compatible with the esp-idf version installed.

brianignacio5, 2025-04-30:
Apologies @olavt We are most likely going to remove the feature altogether because ESP-IDF Matter is released now as a ESP-IDF component in the [Component Registry](https://components.espressif.com/components/espressif/esp_matter)

Please take a look at the component and consider  using it from there as moving forward would the preferred way.

olavt, 2025-05-05:
I think using a component should be fine.

What would typically be the release cycle of updating the component? I see that it seems quite slow. It would be nice to be able to select a stable version, or be able to pick up recent enhancements / bug fixes too.

brianignacio5, 2025-05-19:
You can use any specific version you want from the component registry like `idf.py add-dependency "espressif/esp_matter^<version>"`

For the release cycle information, I would refer to you to the [ESP-Matter repository](https://github.com/espressif/esp-matter)

closed 2025-06-09 by github-actions[bot]