espressif/vscode-esp-idf-extension #1531: [Feature Request]: Cannot build /debug project-chip/connectedhomeip (MATTER) under VSCODE ESP-IDF

jamesharrow, 2025-04-30:
**Is your feature request related to a problem? Please describe.**
I cannot build connectedhomeip under ESP-IDF VSCode due to a missing setup step.

**Describe the solution you'd like**
A mechanism where the connectedhomeip 'source ./scrtips/activate.sh' step is run.

**Describe alternatives you've considered**
Building under command line

**Additional context**
In the connectedhomeip repo: https://github.com/project-chip/connectedhomeip
The instructions to build for ESP32 https://github.com/project-chip/connectedhomeip/blob/master/docs/platforms/esp32/build_app_and_commission.md state that you can build from the command line as follows:

```
ESP-IDF

Tools path SHALL be added to the PATH environment variable to make the tools usable from the command line. ESP-IDF provides another script which does that.

$ cd path/to/esp-idf
$ source export.sh

Activate the Matter environment. Below command needs to be executed after sourcing esp-idf/export.sh.

$ cd path/to/connectedhomeip
$ source scripts/activate.sh

Ccache

Enable Ccache for faster IDF builds. It is recommended to have Ccache installed for faster builds.

$ export IDF_CCACHE_ENABLE=1

Change to example application directory

$ cd examples/<app-name>/esp32
Set the Matter target to build

$ idf.py set-target (target chip)

...

Build the application

$ idf.py build

```

However trying this in vscode with ESP-IDF I hit a problem that the ability to 'source ./scripts/activate.sh' cannot be done.
Also it is not clear if when opening VSCode if one should open the top level connectedhomeip repo, or the specific example folder:
e.g. examples/energy-management-app/esp32 
[This includes a thirdparty subfolder which links back to the top level so technically it should be okay to open VS Code at this level)]

I believe I just need a pre-build step?

brianignacio5, 2025-05-06:
This is already done by the `ESP-IDF: Install ESP-Matter` extension command.

brianignacio5, 2025-05-19:
Closing this issue due to lack of response from user.

Comment here if you need more assistance.

closed 2025-05-19 by brianignacio5

jamesharrow, 2025-05-19:
Apologies

I tried this and it only works for specific applications. At least I tried to use Master and there is no energy-gateway-app or energy-management-app example:

I tried to work out what the magic formula was to extend the ESP-Matter repo to get it to build but had to focus on other aspects.  At least I can now debug in the IDE but have to build it outside of vs code.

brianignacio5, 2025-05-19:
An alternative is to use esp-matter from ESP Component registry https://components.espressif.com/components/espressif/esp_matter

jamesharrow, 2025-05-19:
I think the issue is that the esp_matter does not cover the Project Chip (Matter) example applications I want to use.
So it only covers lighting app and a few other examples, whereas I would like to use the WaterHeater/EVSE which is in the energy-management-app (1.4) and Price/GHG which is on master.

How do I build this under vscode/ESP-IDF: 
https://github.com/project-chip/connectedhomeip/tree/master/examples/energy-gateway-app/esp32

If there were instructions on how to extend the basic lighting to cover these other example apps then that would help.

So the issue is more in the esp-matter repo not on vscode ESP-IDF, although if the vscode ESP-IDF had better examples on how to resolve my original post (being able to run a 2nd setup script) then that would also do.

brianignacio5, 2025-05-19:
If that command doesn't work it is properly either connectedhomeip supported by ESP-Matter is not the same from your example project.

When you run the `ESP-IDF: Install ESP-Matter` the activation script does get executed.  connectedhomeip pigweed and zap is added to vscode build process from the esp-matter path ${ESP_MATTER_PATH} to env PATH and as ZAP_INSTALL_PATH respectively.

I think the question now goes to the [ESP-Matter repository](https://github.com/espressif/esp-matter) and why the example is not available in their releases.