espressif/vscode-esp-idf-extension #1533: [Feature Request]: "OpenOCD Board configuration files are updated" -> would be good to link to configuration files

bryghtlabs-richard, 2025-05-01:
**Is your feature request related to a problem? Please describe.**

I'm having some OpenOCD config troubles. When I change settings in the IDE, it reports that the config files are changed, but which ones? When I run 'git status' I can't find any changes.

**Describe the solution you'd like**
I'd like either a link, or the ability to click on this popup and open the relevant config files that were updated.

**Additional context**
[image]

brianignacio5, 2025-05-06:
What has changed is the `idf.openOcdConfigs` configuration setting, the list of files that openOCD uses to connect when using vscode extension. Basically the file given when running `openocd -f file`

We can update the message itself. But I don't think showing a link to open the file itself is a good idea, usually the file should not modified for most users only those who know how to tweak things in openOCD.

bryghtlabs-richard, 2025-05-07:
>But I don't think showing a link to open the file itself is a good idea, usually the file should not modified for most users only those who know how to tweak things in openOCD.

That's reasonable. I think I'm just confused because often when it says, "OpenOCD Board configuration files are updated", it often doesn't seem to update them, but also doesn't indicate an error. It turns out `idf.openOcdConfigs` was being set in `%APPDATA%\Code\User\settings.json`, which does make sense for some settings like UART port name. However, after updating to VSCode ESP-IDF plugin 1.10, as has my coworker, we cannot debug in VSCode. It seems `idf.openOcdConfigs` is not being passed to OpenOCD, which fails immediately with:

```
C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin\openocd.exe

❌ Error: Debug Adapter has to be specified, see "adapter driver" command
```

It seems like it's being set by VSCode, but not read back in or passed to OpenOCD.

closed 2025-05-20 by brianignacio5