espressif/vscode-esp-idf-extension #1535: Configuration files and C/C++ extension problem

shiinakaze, 2025-05-06:
### OS

Windows

### Operating System version

Windows 11

### Visual Studio Code version

1.99.3

### ESP-IDF version

5.4

### Python version

3.11.2

### Doctor command output

none

### Description

IDF-generated configuration files can sometimes cause issues with C/C++ extensions

[image]

Moreover, in the c_cpp_properties.json, where are the variables `config:idf.espIdfPath` and `config:idf.espIdfPathWin` located ?

```json
{
  "configurations": [
    {
      "name": "ESP-IDF",
      "compilerPath": "${config:idf.toolsPathWin}\\tools\\xtensa-esp-elf\\esp-14.2.0_20241119\\xtensa-esp-elf\\bin\\xtensa-esp32s3-elf-gcc.exe",
      "compileCommands": "${config:idf.buildPath}/compile_commands.json",
      "includePath": [
        "${config:idf.espIdfPath}/components/**",
        "${config:idf.espIdfPathWin}/components/**",
        "${workspaceFolder}/**"
      ],
      "browse": {
        "path": [
          "${config:idf.espIdfPath}/components",
          "${config:idf.espIdfPathWin}/components",
          "${workspaceFolder}"
        ],
        "limitSymbolsToIncludedHeaders": true
      }
    }
  ],
  "version": 4
}

```

### Debug Message

none

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2025-05-06:
Please change `"compileCommands": "${config:idf.buildPath}/compile_commands.json",` to `"compileCommands": "${config:idf.buildPathWin}/compile_commands.json",`

config:idf.espIdfPath and config:idf.espIdfPathWin are based on the extension current settings defined either in .vscode/settings.jsonor `Preferences: Open Workspace Settings (JSON)` or `Preferences: Open User Settings (JSON)`

Based in the error you are showing it is trying to use gcc from mingw64 instead of user the Xtensa toolchain compiler defined in the c_cpp_properties.json path.

shiinakaze, 2025-05-06:
> Please change `"compileCommands": "${config:idf.buildPath}/compile_commands.json",` to `"compileCommands": "${config:idf.buildPathWin}/compile_commands.json",`
> 
> config:idf.espIdfPath and config:idf.espIdfPathWin are based on the extension current settings defined either in .vscode/settings.jsonor `Preferences: Open Workspace Settings (JSON)` or `Preferences: Open User Settings (JSON)`
> 
> Based in the error you are showing it is trying to use gcc from mingw64 instead of user the Xtensa toolchain compiler defined in given c_cpp_properties.json content. Is the json file provided in `.vscode/c_cpp_properties.json` ?

Yes, I configured it, it was generated by the IDF extension.

[image]

[image]

shiinakaze, 2025-05-06:
> Please change `"compileCommands": "${config:idf.buildPath}/compile_commands.json",` to `"compileCommands": "${config:idf.buildPathWin}/compile_commands.json",`
> 
> `config:idf.espIdfPath` and `config:idf.espIdfPathWin` are based on the extension current settings (idf.espIdfPath for example) defined either in `.vscode/settings.json` or `Preferences: Open Workspace Settings (JSON)` or `Preferences: Open User Settings (JSON)`.
> 
> Based in the error you are showing it is trying to use gcc from mingw64 instead of user the Xtensa toolchain compiler defined in given c_cpp_properties.json content. Is the json file provided in `.vscode/c_cpp_properties.json` ?

I might have found the issue. There are multiple projects in my workspace, and one of them has the compiler configured as x86 gcc in the c_cpp_properties.json, causing the LSP to keep searching and getting stuck.

[image]

brianignacio5, 2025-05-07:
Another thing is that `${config:idf.buildPath}` is relative to the currently selected workspace folder of the ESP-IDF extension. You can see which one by hovering the folders next to ESP-IDF v5.4.0 version in the status bar icons.

You need to select the current workspace folder for the commands to work with `ESP-IDF: Current Project` as described in https://docs.espressif.com/projects/vscode-esp-idf-extension/en/latest/additionalfeatures/multiple-projects.html

shiinakaze, 2025-05-07:
> Another thing is that `${config:idf.buildPath}` is relative to the currently selected workspace folder of the ESP-IDF extension. You can see which one by hovering the folders next to ESP-IDF v5.4.0 [...]

ok, thanks

shiinakaze, 2025-05-09:
> Can we close this issue [@shiinakaze](https://github.com/shiinakaze) ?

Oh, sorry, of course.

closed 2025-05-09 by brianignacio5