espressif/vscode-esp-idf-extension #1543: AppData settings.json cross-contamination between projects

bryghtlabs-richard, 2025-05-07:
### OS

Windows

### Operating System version

Windows 11

### Visual Studio Code version

1.99.3 (user install)

### ESP-IDF version

5.1.2

### Python version

3.11.2

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.26100 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v12.8\bin;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v12.8\libnvvp;C:\Program Files\Python311\Scripts\;C:\Program Files\Python311\;C:\Program Files\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\dotnet\;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files\Git\cmd;C:\Program Files\PuTTY\;C:\Program Files (x86)\Nordic Semiconductor\nrf5x\bin\;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:\Program Files\nodejs\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin\;C:\Program Files\IVI Foundation\VISA\Win64\Bin\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin;C:\Program Files (x86)\IVI Foundation\IVI\Bin\;C:\Program Files\IVI Foundation\IVI\Bin\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\NVIDIA Corporation\Nsight Compute 2025.1.1\;C:\Program Files\NVIDIA Corporation\NVIDIA app\NvDLISR;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\GitHubDesktop\bin;C:\Program Files\mitmproxy\bin;C:\Program Files\heroku\bin;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\AppData\Roaming\npm; 
System environment variable PYTHON 
 undefined 
Visual Studio Code Remote name undefined 
Visual Studio Code version 1.99.3 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.10.0 
Workspace folder c:\Projects\ChessUpTwo\ESP32 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:/Espressif/frameworks/esp-idf-v5.1.2/
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths C:\Espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin;C:\Espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin;C:\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin;C:\Espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin;C:\Espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin;C:\Espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\Espressif\tools\cmake\3.24.0\bin;C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin;C:\Espressif\tools\ninja\1.10.2;C:\Espressif\tools\idf-exe\1.0.3;C:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:\Espressif\tools\esp-rom-elfs\20230320
ESP-IDF extra vars
    OPENOCD_SCRIPTS: C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\share\openocd\scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:\Espressif\tools\esp-rom-elfs\20230320\
    IDF_PYTHON_ENV_PATH: C:\Espressif\python_env\idf5.1_py3.11_env
    ESP_IDF_VERSION: 5.1
    PATH: C:\Espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin;C:\Espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin;C:\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin;C:\Espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin;C:\Espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin;C:\Espressif\tools\esp-clang\15.0.0-23786128ae\esp-clang\bin;C:\Espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\Espressif\tools\cmake\3.24.0\bin;C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin;C:\Espressif\tools\ninja\1.10.2\;C:\Espressif\tools\idf-exe\1.0.3\;C:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:\Espressif\python_env\idf5.1_py3.11_env\Scripts;C:/Espressif/frameworks/esp-idf-v5.1.2/tools;%PATH%
    IDF_DEACTIVATE_FILE_PATH: C:\Users\RICHAR~1\AppData\Local\Temp\tmpsl4_utn0idf_836
User extra vars (idf.customExtraVars)
System python Path (idf.pythonInstallPath) C:\Espressif\tools\idf-python\3.11.2\python.exe
Virtual environment Python path (computed) C:\Espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) board/esp32s3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Espressif
Git Path (idf.gitPath) C:/Espressif/tools/idf-git/2.44.0/cmd/git.exe
Notification Mode (idf.notificationMode) All
Flash type (idf.flashType) UART
Flash partition to use (idf.flashPartitionToUse) all
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to C:\Espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin: true
Access to C:\Espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin: true
[... 292 lines cut ...]
User : 5 5 option
s.c:52 configuration_output_handler(): debug_level: 3User : 6 5 options.c:52 configuration_output_handler(): 

.........Richard omitted a lot of text so it would fit in Github form..........

❌ Error: 941 204 mpsse.c:350 
❌ mpsse_open(): unable to open ftdi device with description '*', serial '*' at bus location '*'
Debug: 942 205 command.c:541 run_command(): Command 'init' failed with error code -100
User : 943 205 command.c:612 command_run_line(): C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\share\openocd\scripts/target/esp_common.cfg:9: Error: 
at file "C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\share\openocd\scripts/target/esp_common.cfg", line 9
Debug: 944 206 esp_xtensa.c:198 esp_xtensa_target_deinit(): start
Debug: 945 206 xtensa.c:3423 xtensa_target_deinit(): start
Debug: 946 206 target.c:2166 target_free_all_working_areas_restore(): freeing all working areas
Debug: 947 206 esp_xtensa.c:198 esp_xtensa_target_deinit(): start
Debug: 948 206 xtensa.c:3423 xtensa_target_deinit(): start

Debug: 949 207 target.c:2166 target_free_all_working_areas_restore(): freeing all working areas

For assistance with OpenOCD errors, please refer to our Troubleshooting FAQ: https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
OpenOCD Exit with non-zero error code 1
[Stopped] : OpenOCD Server

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

bryghtlabs-richard, 2025-05-08:
Thank you for mentioning 'idf.saveScope' - I hadn't seen that, but it has the same bug. Default should probably be stored in project-scope, then opt-in to global-scope.

brianignacio5, 2025-05-12:
In https://github.com/espressif/vscode-esp-idf-extension/pull/1527 I fix the issue of saving openOCD boards in current workspace folder.

closed 2025-05-12 by brianignacio5

bryghtlabs-richard, 2025-05-12:
Thanks @brianignacio5 !

PR #1527 (2025-05-12): [VSC-1661] use createQuickPick for openocd board to show currently active item.
## Description

Add quick pick functionality with currently selected option.

Fixes #1524
Fixes #1543

## Type of change

- New feature (non-breaking change which adds functionality)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Click on "ESP-IDF: Select OpenOCD Configuration"
2. Execute action. The currently selected OpenOCD board should be highlighted.
3. Observe results.

- Expected behaviour:
The currently selected OpenOCD board should be highlig [... cut]