espressif/vscode-esp-idf-extension #1546: Launching OpenOCD fails with "Cannot read properties of undefined (reading 'length')" for projects with configurations (aka with a valid 'esp_idf_project_configuration.json' file)

GillesZunino, 2025-05-08:
### OS

Windows

### Operating System version

Window 11 24H2

### Visual Studio Code version

1.100

### ESP-IDF version

5.4.1

### Python version

3.11

### Doctor command output

Error message `Cannot read properties of undefined (reading 'length')` at src\idfConfiguration.ts line 95

```
    case "idf.openOcdConfigs":
      return currentProjectConf.openOCD.configs.length     // <-- currentProjectConf.openOCD.configs is undefined here
        ? currentProjectConf.openOCD.configs
        : "";
```

This error makes it very difficult to debug any project with configuration

### Description

1 - Update to VS Code ESP IDF Extension version 1.10. I have also tried the latest from master (#47fb8b9d2b0b7493b71ad86d9a627228fbabcf5b) and it has the same failure

2 - Open a project with at least one project configuration. I am using this project [https://github.com/GillesZunino/esp32-holiday-tree](https://github.com/GillesZunino/esp32-holiday-tree). Notice the relevant section of the project configuration: 

```
    "openOCD": {
      "debugLevel": 0,
      "configs": [], // <-- OBSERVE: empty array
      "args": []
    },
```

3 - In VS Code, start OpenOCD by clicking on the status bar "ESP-IDF: OpenOCD Server" and clicking "Start OpenOCD"

**EXPECTED**: OpenOCD starts and emits some logs about its activity

**ACTUAL**: OpenOCD fails to start with error message `Cannot read properties of undefined (reading 'length')` in the VS Code output console. No debug log is emitted

**COMPARE**: Everything works as expected with projects without configurations (aka **no** `esp_idf_project_configuration.json` file)

### Debug Message

The problem seems to be in src\project-conf\index.ts, near line 289. There is the following code:

```
      // --- Process OpenOCD Configuration ---
      const openOCDConfigs = resolvePaths
        ? resolveConfigPaths(workspaceFolder, openOCDConfig?.configs, resolvePaths)
        : openOCDConfig?.configs;
```

My configuration file has an empty array for ` openOCDConfig?.configs`. The boolean `resolvePaths` is set to `true` (as it should) and `resolveConfigPaths(workspaceFolder, openOCDConfig?.configs, resolvePaths)` returns `undefined`.

Later on near line 327, the content of `openOCDConfigs` (still `undefined`) is assigned to `projectConfElements[confName].openOCD.configs`:

```
      projectConfElements[confName] = {
         // ..... code removed for brievity
        openOCD: {
          debugLevel: openOCDConfig?.debugLevel,
          configs: openOCDConfigs,
          args: openOCDArgs ?? [],
        },
```

The code which starts OpenOCD (specifically in OpenOcdManager.ts, line 171) attempts to retrieve the current project configuration settings and fails accessing undefined from `projectConfElements[confName].openOCD.configs`.

**Possible Solution**
It seems most of the code does not expect `projectConfElements[confName].openOCD.configs` to be `null` or  `undefined`. It might be best to enhance src\project-conf\index.ts, near line 289 to never emit `undefined` or `null` in `openOCDConfigs`. That said, I am not an expert and there could be better solutions.

### Other Steps to Reproduce

``

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

GillesZunino, 2025-05-09:
**Workaround**

This can be worked around as follows:

1 - Create an empty text file named `openocd-workaround` at the root of the workspace
2 - Change the OpenOCD configuration for all configurations in your `esp_idf_project_configuration.json` to load this empty file

```
    "openOCD": {
      "debugLevel": 0,
      "configs": [ "openocd-workaround" ],
      "args": []
    },
```

3 - IMPORTANT: Close the folder in VS Code and reopen it. The change did not apply fully until I closed the project

jdbaptista, 2025-05-12:
This workaround doesn't seem to be working for me, unfortunately. First, I typically use the `>ESP-IDF: Select OpenOCD Board Configuration` and set it to `ESP32-S3 Chip (via builtin USB-JTAG)`. I don't usually need to have an OpenOCD configuration file in my project, so setting `openOCD.configs` to `[ "board/esp32s3-builtin.cfg" ]` provides an output of:

```
[OpenOCD]
❌ Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:25)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
embedded:startup.tcl:28: Error: Can't find /home/jdbaptista/Documents/repos/traffic-pcb-sb/software/esp-firmware/board/esp32s3-builtin.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 28

For assistance with OpenOCD errors, please refer to our Troubleshooting FAQ: https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
OpenOCD Exit with non-zero error code 1
[Stopped] : OpenOCD Server
```

I found [this](https://github.com/espressif/openocd-esp32/blob/master/tcl/board/esp32s3-builtin.cfg) file, which I assume is the correct configuration, and placed it at `board/esp32s3-builtin.cfg` in my project. After this, I can start openOCD and get the output:

```
[OpenOCD]
Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:25)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

Info : only one transport option; autoselecting 'jtag'

Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections

Info : esp_usb_jtag: serial (F0:9E:9E:28:99:38)

Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz

Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)

Info : [esp32s3.cpu0] Examination succeed

Info : [esp32s3.cpu1] Examination succeed

Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections

Info : [esp32s3.cpu0] Target halted, PC=0x420A285E, debug_reason=00000001

Info : [esp32s3.cpu0] Reset cause (12) - (Software CPU0 reset)

Info : Set GDB target to 'esp32s3.cpu0'

Info : [esp32s3.cpu1] Target halted, PC=0x4037B07A, debug_reason=00000000

Info : [esp32s3.cpu1] Reset cause (12) - (Software CPU1 reset)
```

Unforunately, launching debugging with the `Eclipse CDT GDB Adapter` doesn't seem to do anything and provides no output in either the debug console or output window. Does anyone know what is happening here? Typically I do not even need to start OpenOCD directly, as launching debugging will start it automatically.

GillesZunino, 2025-05-13:
In the configuration files, paths are relative to the workspace. Try to use the full path to the built in board file.
Alternatively, you can probably use ${idf.toolsPath} (or equivalent} to construct the path the built in board file
Another option is to create new file in your workspace and include the built in configuration files from there.

None of these are ideal, but these should get us going until our friends at espressif can get to this one !

closed 2025-05-20 by brianignacio5