espressif/vscode-esp-idf-extension #1547: Version 1.10.0 plugin is broken and does not work - command 'isdIdf.X' not found

larryk3, 2025-05-09:
### OS

Windows

### Operating System version

Windows 10, Windows 11

### Visual Studio Code version

1.100.0

### ESP-IDF version

5.4.0

### Python version

3.11

### Doctor command output

[image]

The doctor command does not run.

### Description

After upgrading to `1.10.0`, the plugin does not work. None of the commands work, and it shows the error message `command 'espIdf.<command>' not found`.

All of the buttons at the bottom to select esp version, and select com port, etc are also gone.

When I downgrade to the previous version `1.9.1`, everything works and is back to normal.

### Debug Message

[image]

This same error happens when I press any of the commands.

### Other Steps to Reproduce

Tried on both windows 10 and 11. 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2025-05-12:
Could you provide the Developer Tools output and the extension activation output?

In Visual Studio Code, go to `Help` > `Toggle Developer Tools` and copy any errors from the Console tab related to this extension.

In Visual Studio Code, go to menu `View` > `Output` > Choose `Extension Host` from the dropdown. There should be some error related to the espressif extension printed there.

Also make sure to delete all idf.* settings in your settings.json:

1. Workspace folder configuration settings in `${workspaceFolder}/.vscode/settings.json`
2. Workspace configuration settings in the workspace’s `<name>.code-workspace` file
3. User settings defined in settings.json
  - Windows: `%APPDATA%\Code\User\settings.json`
  - MacOS: `$HOME/Library/Application Support/Code/User/settings.json`
  - Linux: `$HOME/.config/Code/User/settings.json`

larryk3, 2025-05-12:
[image]

I tried removing the `idf.*` settings, but that didn't seem to help.

Here is the output from the `Extension Host`:

```
2025-05-11 21:09:59.552 [info] Extension host terminating: renderer closed the MessagePort
2025-05-11 21:09:59.574 [error] An error occurred when disposing the subscriptions for extension 'eamodio.gitlens':
2025-05-11 21:09:59.576 [error] TypeError: Cannot read properties of undefined (reading 'dispose')
	at Object.dispose (file:///c:/Users/larry/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:145:120034)
	at m0.<anonymous> (file:///c:/Users/larry/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:115:8921)
	at m0.dispose (file:///c:/Users/larry/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:115:9017)
	at Object.dispose (c:\Users\larry\.vscode\extensions\eamodio.gitlens-17.0.3\dist\gitlens.js:1283:24001)
	at c:\Users\larry\.vscode\extensions\eamodio.gitlens-17.0.3\dist\gitlens.js:1863:15285
	at Array.forEach (<anonymous>)
	at Object.dispose (c:\Users\larry\.vscode\extensions\eamodio.gitlens-17.0.3\dist\gitlens.js:1863:15267)
	at Er (file:///c:/Users/larry/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:25:746)
	at file:///c:/Users/larry/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:123:15046
	at file:///c:/Users/larry/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:25:1056
	at Object.dispose (file:///c:/Users/larry/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:8:705)
	at PJ.eb (file:///c:/Users/larry/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:123:11919)
	at file:///c:/Users/larry/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:123:9874
	at Array.map (<anonymous>)
	at PJ.$ (file:///c:/Users/larry/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:123:9862)
	at PJ.terminate (file:///c:/Users/larry/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:123:10105)
	at v2.terminate (file:///c:/Users/larry/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:124:1434)
	at us (file:///c:/Users/larry/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:225:5702)
	at MessagePortMain.<anonymous> (file:///c:/Users/larry/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:225:2249)
	at MessagePortMain.emit (node:events:524:28)
	at MessagePortMain._internalPort.emit (node:electron/js2c/utility_init:2:2949)
2025-05-11 21:09:59.607 [error] Error: Channel has been closed
	at s (file:///c:/Users/larry/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:125:2431)
	at Object.appendLine (file:///c:/Users/larry/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:125:2570)
	at R.logOutputMessage (c:\Users\larry\AppData\Local\Programs\Microsoft VS Code\resources\app\extensions\json-language-features\client\dist\node\jsonClientMain.js:2:290791)
	at R.info (c:\Users\larry\AppData\Local\Programs\Microsoft VS Code\resources\app\extensions\json-language-features\client\dist\node\jsonClientMain.js:2:290524)
	at ChildProcess.<anonymous> (c:\Users\larry\AppData\Local\Programs\Microsoft VS Code\resources\app\extensions\json-language-features\client\dist\node\jsonClientMain.js:2:514797)
	at ChildProcess.emit (node:events:524:28)
	at ChildProcess._handle.onexit (node:internal/child_process:293:12)
2025-05-11 21:09:59.618 [error] Error: Channel has been closed
	at s (file:///c:/Users/larry/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:125:2431)
	at Object.appendLine (file:///c:/Users/larry/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:125:2570)
	at _0x57de3e.<computed>.<computed> (c:\Users\larry\.vscode\extensions\ms-python.vscode-pylance-2025.4.1\dist\extension.bundle.js:1:140622)
	at _0x57de3e.<computed>.<computed> [as appendLine] (c:\Users\larry\.vscode\extensions\ms-python.vscode-pylance-2025.4.1\dist\extension.bundle.js:1:140622)
	at P.logOutputMessage (c:\Users\larry\.vscode\extensions\ms-python.python-2025.6.0-win32-x64\out\client\extension.js:2:2018199)
	at P.info (c:\Users\larry\.vscode\extensions\ms-python.python-2025.6.0-win32-x64\out\client\extension.js:2:2017932)
	at ChildProcess.<anonymous> (c:\Users\larry\.vscode\extensions\ms-python.python-2025.6.0-win32-x64\out\client\extension.js:2:2236275)
	at ChildProcess.emit (node:events:524:28)
	at ChildProcess._handle.onexit (node:internal/child_process:293:12)
2025-05-11 21:09:59.651 [info] Extension host with pid 21804 exiting with code 0
2025-05-11 21:10:00.525 [info] Extension host with pid 11548 started
[... cut ...]
2025-05-11 21:10:19.735 [error] TypeError: Cannot convert undefined or null to object
	at Function.keys (<anonymous>)
	at c:\Users\larry\.vscode\extensions\espressif.esp-idf-extension-1.10.0\dist\extension.js:2:1371355
	at l (c:\Users\larry\.vscode\extensions\espressif.esp-idf-extension-1.10.0\dist\extension.js:2:1372008)
	at Ze (c:\Users\larry\.vscode\extensions\espressif.esp-idf-extension-1.10.0\dist\extension.js:2:1451242)
	at async zt.configureWithDefaultValues (c:\Users\larry\.vscode\extensions\espressif.esp-idf-extension-1.10.0\dist\extension.js:2:1469429)

```

brianignacio5, 2025-05-12:
Do you have a project configuration being used at the moment ? It seems the env variable is undefined somehow.

Check the `esp_idf_project_configuration.json` file for the currently selected profile

larryk3, 2025-05-12:
This is the `esp_idf_project_configuration.json` file I have: https://github.com/K3Engineering/dynamite-sampler-firmware/blob/main/esp_idf_project_configuration.json

I tried removing the file, and then downgrading and upgrading the extension - it still fails with the same error.

Can you replicate the error if you clone my repository?

larryk3, 2025-05-12:
I removed the entire `%appdata%\Code\User\workspaceStorage\` folder, and it seems like it works now. I looked through the cached files, and I couldn't find anything offensive in there. So I'm not sure what's up.

brianignacio5, 2025-05-13:
In `esp_idf_project_configuration.json` any profile can define env variables (the `env: {}` field). We added some fixes regarding the env variable in project configuration but we didn't have the correct validation when the object is undefined. We will fix this in a PR very soon.

For your use case it seems that an older profile was crashing the extension. Apologies, adding new features usually introduces bugs to unexpected areas ;(

closed 2025-05-20 by brianignacio5